MarlinFirmware/Marlin #16801: [BUG] (short description)Use G28, Error, Homing Failed

jackshi618, 2020-02-07:
Run 2.0.3
Use G28, Error, Homing Failed
X=min, Y=min, Z=max
use AUTO_BED_LEVELING_LINEAR
Run G28, 1.Z-axis reset normal, 2.X-axis normal, 3.Y-axis run to half-reported error, Homing Failed
But if you are set to
X=min, Y=min, Z=min
Then everything's fine.
Run 1.8 is OK

jackshi618, 2020-02-07:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4171821/Marlin.zip)

jackshi618, 2020-02-07:
Only when you reset Y makes a mistake.

jackshi618, 2020-02-07:
[pins_3DYMY.zip](https://github.com/MarlinFirmware/Marlin/files/4171964/pins_3DYMY.zip)

boelle, 2020-02-08:
@jackshi618 tried bugfix 2.0.x ?

boelle, 2020-02-08:
where did you get marlin from? marlin config files do not have chinesse writing in it

jackshi618, 2020-02-08:
2.0.3 
Chinese i added it myself.
Marlin download from GitHub

jackshi618, 2020-02-08:
@boelle 2.0.3

jackshi618, 2020-02-08:
@boelle 
Chinese i added it myself.
Marlin download from GitHub

boelle, 2020-02-08:
there are many places on github where you can get marlin

please link to where you got it

jackshi618, 2020-02-09:
@boelle 
https://github.com/MarlinFirmware/Marlin

jackshi618, 2020-02-09:
Didn't try 2.0.x.

boelle, 2020-02-09:
please try

jackshi618, 2020-02-09:
ok,Test it later.

jackshi618, 2020-02-09:
@boelle try 2.0.x,The same problem.

boelle, 2020-02-09:
do you use normal endstops with a microswitch?

what does M119 say if no nedstop is activated?

and what happens if you touch the endstop with a finger and send M119?

do one endstop and then move to the next, just to check if they are bad

jackshi618, 2020-02-09:
run 1.1.8 ,no bug,no err
1.1.9-2.0.x,same bug

Papipapito, 2020-02-26:
Hello I'm same problem, with 2.0.4.3
Delta with Microswitch into zxy max , Board SKR mini v1.1
M119 ALL ENDSTOP OPEN
Send G28, the printer up to endstop
Move arms first X, and when clik into endstop, the console show:

Pinter Halted kill Called

Sometimes the error show when move Y to endstop

And sometimes everything works fine and I can launch a G33, it is a very random failure to do G28

thinkyhead, 2020-02-26:
Please test with the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) so we're all on the same page here. That branch contains all the latest patches.

Papipapito, 2020-02-26:
Hello

Yes, with bugfix-2.0.x are solved .

when does bugfix come together in branch stable 2.0.x?

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle