MarlinFirmware/Marlin #16802: [BUG] (short description)

murdock62, 2020-02-07:
### Bug Description

Version 2.0.3.

With my delta 3d printer (Anycubic Kossel), at the end of a print, the screen stay in "print mode" 
i need to stop the job in the menu. 

No problem with version 2.0.1

### Steps to Reproduce

Just print a part.

My config :
[Cfg.zip](https://github.com/MarlinFirmware/Marlin/files/4171977/Cfg.zip)

Many thanks

AbuMaia, 2020-02-07:
I'm having the same issue on a prusa-style printer. When a print finishes or is cancelled, the printer does not idle down, and the steppers do not disengage. If I need to change the filament between prints, I have to power cycle the printer first to get the extruder to let go.

thisiskeithb, 2020-02-07:
This was fixed in #16749, so you’ll have to wait for the next tagged release or grab the latest [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip).

AbuMaia, 2020-02-07:
Thanks Keith. Since the fix was only flipping one character, I found it quicker and easier to just change it in my local copy.

murdock62, 2020-02-08:
Thanks for the information !

AbuMaia, 2020-02-08:
I'm afraid I must report that changing the < to > did not solve my problem. I edited, saved, and flashed the firmware to my printer, and just completed a print. The LCD screen options were still in-print options, and while the X and Y axes were free (and presumably the Z axis, though I didn't want to try it and have to re-level), the extruder motor was still powered and locked up.

PM3Delta, 2020-02-09:
Since Feb 4th the 2.0.x-bugfix fixed that issue for me. I had presumed that #16753 had taken care of it. I might be mistaken though.

boelle, 2020-02-09:
@PM3Delta yes use the latest bugfix 2.0.x

@murdock62 why the reopen?

murdock62, 2020-02-10:
@boelle Oops !! wrong manipulation sorry !

closed 2020-02-10 by murdock62

AbuMaia, 2020-02-14:
Using the latest bugfix version does not fix the issue for me. I still have to reset the printer to access the axis movement menu after a print finishes.

thinkyhead, 2020-02-14:
See if the today's patch helps. Due to a bug, turning off the bed was not counting towards print job termination.

AbuMaia, 2020-02-15:
Thanks @thinkyhead. I think that worked.