MarlinFirmware/Marlin #16803: G29 S-0 or G29 S-2 will not print M421 data for mesh.gcode

DroneMang, 2020-02-07:
Hi,
Why/Purpose of this: have a mesh.gcode file for PLA and ABS that can be loaded without probing the bed after firmware updates.

I have 2 mesh's loaded in slot 0 and slot 1, for PLA at 70 deg and ABS at 110.

If I run the command G29 S-0 it will save the currently loaded mesh into slot 0.  

If I run G29 S-2:
Recv: ?Invalid storage slot.
Recv: ?Use 0 to 2

I have run G29 A and M500 after each slot is loaded or saved so all the data is in eeprom and I have verified that it is there by running G29 L0 or L1 and printing it with G29 T1.

So the workaround is to load/save whatever mesh into slot 1 so that it can be printed out with G29 S-1, and then saved into a mesh.gcode file.

Desired behaviour:
G29 S-0
G29 S-1
G29 S-2
will each print out savable data in the M421 I0 J0 Z0.0106 ..... etc format.

Very good work on here to all who check these and fix things!  Marlin is great!
--

ManuelMcLure, 2020-02-07:
Only ``G29 S-1`` will print the _currently loaded_ mesh, i.e. whichever mesh was last loaded with ``G29 L<slot>``. If you need to print  the mesh stored in slot 2:
```
G29 L2
G29 S-1
```

DroneMang, 2020-02-08:
Ok I guess that works. Also it would be good to add how to save that to
mesh.gcode and how to load into firmware to the UBL and G29 documentation.
Thanks!

ManuelMcLure, 2020-02-08:
If this has answered your question please close the issue. If you'd like to improve the documentation in this area, we could always use more volunteers! Documentation is a great way to dip your toes into the Marlin waters. Take a look at the [Marlin Documentation Github](https://github.com/MarlinFirmware/MarlinDocumentation).

DroneMang, 2020-02-08:
ok I'll close it.  Yea maybe I'll look into documentation...

One question, does the currently loaded mesh survive a reboot?  How is that handled in EEPROM?  

Do I need to load the mesh I want to use at each restart, or only load it when changing between saved mesh's?

ManuelMcLure, 2020-02-08:
In general you want to load the mesh in your start gcode, specially if you have different meshes for different filaments. The current mesh will not survive a reboot- I’m not sure if it will load mesh 0 on reboot or if the mesh will just be empty. Any meshes you explicitly saved with G29 S<slot> will survive even a firmware upgrade.

closed 2020-02-08 by boelle

DroneMang, 2020-02-08:
> mesh will just be empty. Any meshes you explicitly saved with G29 S will

So what would your start Gcode be as far as the mesh?

G29 L0

Or,

G29 L0
G29 F10 (is this necessary each time? What does it default to if not
included?)
G29 A
M500

Thanks.

DroneMang, 2020-02-08:
Also,

With a firmware update if I run

M502
M500
M501

Then there is nothing in the mesh slots, so what command sequence do you
run after an update to keep the mesh's but also clear and reset the EEPROM?
Thanks.

ManuelMcLure, 2020-02-08:
What leveling method are you using? UBL works the way I describe. After saving meshes with ``G29 S0``/``G29 S1``, etc. those meshes will survive a firmware update and can be loaded with ``G29 L0``/``G29 L1``, etc.

My start gcode looks like:
```
G28 ; home all axes
M420 L0 ; Load UBL matrix
M420 S1 ; Activate UBL
M420 Z10.0 ; 10mm fade height
```

thinkyhead, 2020-02-10:
> …to keep the [meshes]…

Copy the output of `M503 S0` and store it away for later.
Run that g-code to restore settings and meshes.