MarlinFirmware/Marlin #16804: [BUG] M701/M702 doesn't ask for temperature

Zemistr, 2020-02-07:
### Bug Description

M701/M702 doesn't ask for temperature

### My Configurations
- Uncommented NOZZLE_PARK_FEATURE
- Uncommented ADVANCED_PAUSE_FEATURE
- Uncommented FILAMENT_LOAD_UNLOAD_GCODES

### Steps to Reproduce

1. Send M701 or M702 with cold nozzle

**Expected behavior:** [What you expect to happen]
1. Opens filament load/unload menu
1. Asks for material if not heating/heated up
1. Loads the filament
1. ...
1. Profit

**Actual behavior:** [What actually happens]
1. Opens filament load/unload menu
1. ...
1. Tells **Hotend too cold**
1. Returns to main screen

boelle, 2020-02-08:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Zemistr, 2020-02-08:
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

I saw this in the Issue template but honestly... I tried 4 different printers, and it's the same.
Marlin 2.0.3 and uncommented NOZZLE_PARK_FEATURE, ADVANCED_PAUSE_FEATURE and FILAMENT_LOAD_UNLOAD_GCODES.

You really don't need these config files... but as you wish...

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4173483/Marlin.zip)

InsanityAutomation, 2020-02-08:
Try not to be too hard on him, he's not a developer but his role is to make sure developers have everything they need when they make it to an issue to save us time. In this case you're right, config is irrelevant but it's a matter of process really.

In any case, I've got about 4 issues lined up in this area to work on now, so I may bump it a bit higher on my list ... That'll make it 3rd atm.

ellensp, 2020-02-08:
using M701/M702 is different from using the menus. M701/M702 should not ask for temperature.

But the filament load/unload menu system should check and ask for a temperature before calling M701/M702  

so one of the first questions a developer would ask is what screen do you have? as different screen use different systems for their menus.. 
this information is in the configuration files you grudgingly provided. 
Turns out its a very common REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER vs anything with a more menu system.

So I disagree that config file is irrelevant in this case.

ellensp, 2020-02-08:
using current bugfix-2.0.x It does as it should. tests for temperature and inserts various preheat options before allowing a unload or load.   On both extruders,

Zemistr, 2020-02-08:
> using current bugfix-2.0.x It does as it should. tests for temperature and inserts various preheat options before allowing a unload or load. On both extruders,

So if you send M702 after you switched on the printer, it asks for the filament temperature?

ellensp, 2020-02-08:
No, it wont. It is not meant to.  It just loads and unloads. you are meant to set the prerequisites with other gcodes.

Zemistr, 2020-02-08:
How can I invoke the menu to choose the filament temperature?
I expected that behavior before load or unload.

ellensp, 2020-02-08:
The LCD works as expected. You tell it to change filament or upload or load and if the hotend is cold it will prompt you to set a temperature. 
You get three options. two presets Preheat PLA or Preheat PETG and the Preheat Custom option where you set the temperature on the lcd.

The predefined temperatures is defined in Configuration.h 
#define PREHEAT_1_LABEL       "PLA"
#define PREHEAT_1_TEMP_HOTEND 215
#define PREHEAT_2_LABEL       "PETG"
#define PREHEAT_2_TEMP_HOTEND 240

raw gcode you need to tell it a temperature before calling load/unload

Zemistr, 2020-02-08:
Via the menu it does exactly this.
Buuuut... If I send command via command line, it just opens the menu about "waiting for zero temperature" saying "loading/unloading filament" and in one second (+-) it disappears :|

If this works in the bugfix version... ok. I will try it with the another release.

ellensp, 2020-02-08:
Your still missing the point.

If your controlling the machine with gcode, it expects you to control the machine with gcode.
It wont prompt for a temperature on the LCD as are you using gcodes....   It will however tell you via console "Hotend too cold" so you can give a different  gcode to set the temperature. 

The only error here in my view is it shouldn't change to the load filament screen until after it has done sanity checks that you have set a temperature.

ellensp, 2020-02-08:
ie changing Marlin/src/gcode/feature/pause/M701_M702.cpp  
```
  #if HAS_LCD_MENU
    lcd_pause_show_message(PAUSE_MESSAGE_LOAD, PAUSE_MODE_LOAD_FILAMENT, target_extruder);
  #endif
```
to   
```
#if HAS_LCD_MENU
     if (!thermalManager.targetTooColdToExtrude(active_extruder))
       lcd_pause_show_message(PAUSE_MESSAGE_LOAD, PAUSE_MODE_LOAD_FILAMENT, target_extruder);
#endif
```
and something very similar for M702

Zemistr, 2020-02-08:
> Your still missing the point.
>
> If your controlling the machine with gcode, it expects you to control the machine with gcode.
> It wont prompt for a temperature on the LCD as are you using gcodes.... [...]

I wanted to add custom menu item to invoke unload and load in one menu item (plus a lot of other code ;) )

thinkyhead, 2020-02-11:
> I wanted to add custom menu item to invoke unload and load in one menu item (plus a lot of other code ;) )

Something like this?
```cpp
GCODES_ITEM_P(PSTR("Heat/Change"), PSTR("M109 S200\nM701"));
```
Of course there are always the options under `CUSTOM_USER_MENUS`….

closed 2020-02-11 by thinkyhead

Zemistr, 2020-02-11:
@thinkyhead This is the thing... I created item in the **CUSTOM_USER_MENUS** but it calls just M701/M702. I don't want to set there the temperature. I don't know what kind the filament in the nozzle is.
My problem is not with the M701/M702. It's actually more that it didn't ask for the temperature.

thinkyhead, 2020-02-11:
We don't have G-codes invoke interaction with the LCD screen. But sure, that can be done with the proper coding. Marlin is meant to be very customizable. But I don't think we'll be adding that as a general feature, and I certainly don't have time to implement it at this time. The TODO list is huuuge.

Zemistr, 2020-02-11:
@thinkyhead Totally understand 😉 
I'm software developer too... so... 😄