MarlinFirmware/Marlin #16805: [BUG] "Has TMC2130" flag for X2 with DUAL_X_CARRIAGE

AlexandrZloy, 2020-02-08:
Seems that the issue exists in this commit. Leave comment with description. Please verify:
https://github.com/MarlinFirmware/Marlin/commit/1084068a3e6fd0ac0fb9853f33594d3b48e78ecb#commitcomment-37187289
Thanks you.

boelle, 2020-02-08:
seems to be related to 2.0.2 and 2.0.3 is that correct?

AlexandrZloy, 2020-02-08:
> seems to be related to 2.0.2 and 2.0.3 is that correct?

2.0.3 for sure. But I can't say about 2.0.2 - I does not use it. "Suspicious" code was commited on Jan 20, 2020.

boelle, 2020-02-08:
what about bugfix 2.0.x ? that one is updated daily so might been fixed

AlexandrZloy, 2020-02-08:
> what about bugfix 2.0.x ? that one is updated daily so might been fixed

No, the same code in bugfix-2.0.x now:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/core/drivers.h
```
#if EITHER(X_DUAL_STEPPER_DRIVERS, DUAL_X_CARRIAGE)
  #define AXIS_DRIVER_TYPE_X2(T) _AXIS_DRIVER_TYPE(X2,T)
#else
  #define AXIS_DRIVER_TYPE_X2(T) false
#endif
```

closed 2020-02-10 by thinkyhead