MarlinFirmware/Marlin #16808: [BUG] Printer pausing at end of print "Click to resume"

ArketypeDesign, 2020-02-08:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-02-08:
Somebody else just reported what sounds like the same thing when testing another issue I worked on (linked above).

I could have sworn I have already seen this behavior reported and supposedly fixed in the past week. Was that something you had looked at, @thinkyhead?

carl1961, 2020-02-08:
I had same issue with all marlin versions, I am running the latest 2.0 (downloaded yesterday)  stops at last couple lines in code ( when LED statis turns Green ,I think )it has to do with the 5050 LED's I have led enabled.  work around is doing this:

@shitcreek commented on Aug 25, 2019 • 
Since you have #define PCA9533 enabled I think this has more to do with PRINTER_EVENT_LEDS #14774.
The current work around is to comment out
C:\BIGTREETECH\Marlin-2.0.x\Marlin\src\gcode\queue.cpp    starts at Line: 536 - 544
#if HAS_RESUME_CONTINUE
  inject_P(PSTR("M0 S"
    #if HAS_LCD_MENU
      "1800"
    #else
      "60"
    #endif
  ));
#endif
in queue.cpp

https://github.com/MarlinFirmware/Marlin/issues/15054

[image]
[image]
[image]

PM3Delta, 2020-02-09:
Since Feb 4th the 2.0.x-bugfix fixed that issue for me. I had presumed that #16753  had taken care of it. I might be mistaken though.

thisiskeithb, 2020-02-09:
Using f0cbc85 and printing via SD, I am also still seeing this behavior before the last couple of layers. At no time did I cancel the print, but I did click resume:

```prolog
Recv: echo:Unknown command: "M73 P99 R0"
[...]
Recv: ok P0 B0
[...]
Recv: X:124.69 Y:124.52 Z:39.80 E:0.00 Count X:13148 Y:12436 Z:16000
[...]
Recv: ok P0 B0
[...]
Recv: ok P28 B1
[...]
Recv: echo:busy: processing
[...]
Recv: ok P31 B21
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:Unknown command: "M73 P100 R0"
Recv: //action:notification 59m 22s
Recv: echo:Print time: 59m 22s
[...]
Changing monitoring state from "Printing from SD" to "Cancelling"
Recv: ok P31 B30
[...]
Recv: ok P31 B31
Send: M108
[...]
Recv: ok P31 B31
Recv: ok P31 B31
Send: G1 E-1.5 F1000
Send: G1 Z20 F200
Recv: ok P30 B31
Send: M108
Recv: ok P29 B31
Send: M84
Recv: ok P29 B31
Send: M104 T0 S0
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: ok P31 B30
Send: M108
Recv: //action:notification MK32-bit Ready.
Recv: ok P31 B31
Recv: ok P31 B31
Send: M140 S0
Recv: ok P31 B31
Send: M106 S0
Changing monitoring state from "Cancelling" to "Operational"
Recv: ok P31 B31
Send: G1 E-1.5 F1000
Send: G1 Z20 F200
Recv: ok P30 B31
Recv: ok P30 B31
Send: M84
[...]
Recv: ok P31 B30
[...]
Send: M104 T0 S0
Recv: ok P31 B31
Recv: //action:notification MK32-bit Ready.
Recv: ok P31 B31
Send: M140 S0
Recv: ok P31 B31
Send: M106 S0
Send: G1 E-1.5 F1000
Recv: ok P31 B31
Recv: ok P30 B31
Send: G1 Z20 F200
Recv: ok P30 B31
Send: M84
Send: M104 T0 S0
Recv: ok P31 B30
Recv: //action:notification MK32-bit Ready.
```

thinkyhead, 2020-02-09:
What are we looking for in that log?

thisiskeithb, 2020-02-09:
> What are we looking for in that log?

I have no idea because I can't figure out why it's pausing.

thinkyhead, 2020-02-09:
Well, there is no indication where the pause occurred. Use `M111 S3` to enable logging of every command that arrives, wait for the pause to occur and take note of the log at that point.

thisiskeithb, 2020-02-09:
Will do. I'm about to start another test print.

thinkyhead, 2020-02-09:
As for the original issue, if you have `PRINTER_EVENT_LEDS` enabled, this is a feature. The print pauses at the end to let the light stay at the "print completed" color for 30 minutes (or just 1 minute if you have no LCD controller).

thinkyhead, 2020-02-09:
Can you please describe the strange behavior you are seeing in a concise manner to save time for the weary reader?

thisiskeithb, 2020-02-09:
I immediately get a resume print prompt when starting a print & it also pauses near the end. I don't have `PRINTER_EVENT_LEDS` enabled.

thisiskeithb, 2020-02-09:
I'll fire up some more printers and flash to the latest commit see if I can collect more data.

thinkyhead, 2020-02-09:
I understand that your issue is different. This is why we prefer that people do not piggyback on similar but most likely unrelated issues.

thisiskeithb, 2020-02-09:
Sorry about that. I didn't realize that were different.

closed 2020-02-09 by thinkyhead

carl1961, 2020-02-09:
@thinkyhead  The op and me have  #define RGB_LED enabled, at the last layer or maybe two layers the marlin firmware will stop ( with hot end on print ) until you press the encoder to resume. it happens when the led turns green on my 3d printer.  only workaround was commenting those lines out in queue.cpp . i know that is not the correct fix but it gets us going. If I comment out   #define RGB_LED where the led stop working, the issue goes away so it must be in the event led lights that turn different colors for different stages of the print job.

thinkyhead, 2020-02-09:
I'm adding an (undocumented) `Q` parameter to M0/M1 so that the screen won't read "Click to Resume…" but will show the total print time at the end of the print. However, it will still be waiting for a click. That will be merged before too long…

thisiskeithb, 2020-02-09:
> If I comment out #define RGB_LED where the led stop working, the issue goes away

That was the source of the pause I was experiencing. Thanks for finding it!

carl1961, 2020-02-09:
@thisiskeithb  been studding this gode since thinkhead mentioned M0/M1

is might be more related to onPrintCompleted  , thats when the green led come on.

file    printer_event_leds.h

 #if ENABLED(SDSUPPORT)

    static inline void onPrintCompleted() {
      leds.set_green();
      #if HAS_LEDS_OFF_FLAG
        leds_off_after_print = true;
      #else
        safe_delay(2000);
        set_done();
      #endif
    }

ArketypeDesign, 2020-02-09:
OK... my mistake- I had RGB enabled (I do not have RGB lights on this printer).
So this is no longer a problem for me.

Still- for those with lights it looks like the "pause" occurs before all of the ending g-code is excecuted, which I think is still a problem.

thinkyhead, 2020-02-10:
> Still- for those with lights it looks like the "pause" occurs before all of the ending g-code is excecuted, which I think is still a problem.

It is only used for SD printing, and only after queueing the last command from the card.
So… it's mysterious.

thinkyhead, 2020-02-10:
There was a bug in the command injection at the end of SD printing, so `M0` was being executed ahead of the final commands in the G-code file. This has now been patched. Since this is a safety issue, I recommend you update your copy of Marlin soon, and for extra safety on machines that might not have this update, be sure to put an `M400` at the end of all your G-code files and in your slicer custom end g-code.

thisiskeithb, 2020-02-11:
We still saw this bug with `M400` in our end gcodes, but I rebased my configs on the latest commit, so we should be covered.

Thanks for the patch!

thinkyhead, 2020-02-11:
Of course, the `M400` also doesn't get reached. 🤦‍♂

Moyster, 2020-02-16:
@thinkyhead : 
> I'm adding an (undocumented) Q parameter to M0/M1 so that the screen won't read "Click to Resume…" but will show the total print time at the end of the print. However, it will still be waiting for a click. That will be merged before too long… A more complete solution will skip using M0/M1 altogether.

This breaks the lcd menu on printers with status led enabled, there never is a "click to resume" nor does it listen to any click after that, led stays green, can't confirm/use any action that relies on sending Gcode (homing / etc...) but I can navigate my SD and start a print (not sure if it starts the print as i didn't try), rebooting the printer works, using a bugfix tree from before the coolstep imp/revert doesn't show this bug (this commit in particular : https://github.com/MarlinFirmware/Marlin/commit/0ce83dee9560fc30f27fdddc450d3861725d9a22 ).

thinkyhead, 2020-02-17:
> nor does it listen to any click after that

I'm messing around with this and should have it sorted soon.

jwebber23, 2020-02-17:
I have the same issue running 2.0.3 on a BTT SKR MINI E3 V1.2.  If i enable my NEOPIXEl's i get the green lights and Click to Resume a few lines before the end of the print (before any of my end g-code runs).  If is disable them it run to the end as expected.

I have created a short G Code run without the Bed and Extruder heating and this gives the same results, 
NEOPIXEL's on => stops early
NEOPIXEL's off => works fine.

How does the code determine the "end of the print" from the SD card?

Moyster, 2020-02-17:
> I'm messing around with this and should have it sorted soon.  

My bad, EU time here, saw that as I started reflashing / running a debug gcode.. oh well, glad you found something, I'll still add some debug logs to my opened issue.  

@jwebber23 : Use the bugfix branch, It was supposed to be fixed a 5-6 days ago.

jwebber23, 2020-02-17:
> > I'm messing around with this and should have it sorted soon.
> 
> My bad, EU time here, saw that as I started reflashing / running a debug gcode.. oh well, glad you found something, I'll still add some debug logs to my opened issue.
> 
> @jwebber23 : Use the bugfix branch, It was supposed to be fixed 5-6 days ago.
> 
> Also, I never had the bug this issue refers to, RGB led and PRINTER_EVENT_LEDS always worked since 1.x for my setup, so... what kind of slicer do y'all use ?

I am currently using Prusa Slicer but will transfer to Simplify 3D and see if there is any change. I will try a build tomorrow with the Bug Fix. I don't have enough pins for RGB so will have to stick with Neopixels.

Moyster, 2020-02-17:
I use Cura here and the gcode from SD works and triggers the correct screen.  
Neopixel or RGB should behave the same, also worth checking if you didn't scuff :  

>   #define EVENT_GCODE_SD_STOP "G28XY"       // G-code to run on Stop Print (e.g., "G28XY" or "G27")

If you changed this line, make sure you use "\n" to separate each gcodes (like "G28XY\n M150 R0 U0 B0")
If you're using leds anyway, I guess the current bugfix might not help (as the "click to resume" post-print screen is gone for us, led users).
Wait for https://github.com/MarlinFirmware/Marlin/issues/16878 to be solved 📦

jwebber23, 2020-02-22:
I have now tried 2.0.4 (downloaded today) and latest Bugfix (downloaded today) and both now stop near the end and i no longer get the click to continue.  I have tried with Simply 3D and self generated test G-Code all with the same result.

Not sure if a fix has been put in any of the versions of code but it appears worse as without the click to continue you need to turn off the printer to do anything.....

Moyster, 2020-02-22:
@jwebber23 : https://github.com/MarlinFirmware/Marlin/issues/16908 
Didn't try myself, but apparently ellensp found the problem and posted a fix, too bad it missed the release mark.

thinkyhead, 2020-02-22:
I agree that `onPrintCompleted` has an issue. An SD print is considered completed as soon as the last command is read from the SD card and (presumably) stored in the g-code queue. The queue is typically about 4 items long. Some configs make it 10. And recently I've seen issues connected to a queue that was 64 commands long (though usually not full).

The `onPrintCompleted` method is called before the items in the queue are processed.

The injection of `M0` should be okay in itself, because this goes to the end of the queue at the point where the last command from SD has been added. You can enable echo logging with `M111 S3` and connect a host during a test print to see the exact order that commands are getting processed. I'll take a look at this over the weekend, as it may require some extra logic to make sure the print is only flagged as done when the last command has actually been processed.

Moyster, 2020-02-22:
Don't sweat it, according to this : https://github.com/MarlinFirmware/Marlin/pull/16921/files  
It looks like you inverted some operators when tidying up the loop, that might explain why it didn't show the "click to resume" (or shows it for a split second before moving on) and didn't register any clicks either.

jwebber23, 2020-02-22:
That would explain it once the last line is read from the SD card rather than executed. Was the bugfix fix added to 2.0.4 as well?

thinkyhead, 2020-02-23:
> It looks like you inverted some operators…

Yeah, that typo is one of many I let fall in this week due to lack of sleep while pushing Auto Build through. I'm taking it much slower this week.

So, please give the latest code a test. There may still be an issue with `CardReader::printingHasFinished` that could manifest under some circumstances. Specifically, what will happen is as soon as the end of the file is reached, `CardReader::printingHasFinished` calls `planner.synchronize` so if there's a move on the last line of the g-code file there would be a pause in the motion before it executed. That's not too serious, but `printingHasFinished` does a bit more besides…

- Calls `finish_and_disable` which disables all steppers (if `SD_FINISHED_RELEASECOMMAND`)
- Enqueues `M31` or `M117` (waiting for queue space to open up)
- May re-select the file in the UI.

All these tasks are fine if done at the right time, but these closing steps could come in ahead of the last line in the G-code file. And if the next-to-last move is very long, other oddities can arise. The out-of-order or delayed execution of the last line in the g-code file is what you should look for with `M111 S3` and I will be confirming this on my setup as well.

Moyster, 2020-02-23:
@jwebber23 : the fix is in bugfix and 2.0.4.2 (today's maintenance release).  

> due to lack of sleep  
Don't force-push yourself too much ;)

jwebber23, 2020-02-25:
Hi,

I just tried 2.0.4.3 and it still is stopping before the end of the SD card and not displaying the Click to resume.  Once you click it doesn't continue and you cant release the media either.  I have uploaded the platformio and my 2 config files in case there is something wrong with them which is causing my issue.  I will give bugfix a go tonight.
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/4248964/Config.zip)

Moyster, 2020-02-25:
I didn't run any print since this weekend, but as far as I know, the original issue you posted here was fixed.  
Related to sdprint and since then, this happened : https://github.com/MarlinFirmware/Marlin/issues/16936 .  
Can you do a tiny print and send "M111 S3" to see what happens at the end of it ?

jwebber23, 2020-02-25:
[M111.zip](https://github.com/MarlinFirmware/Marlin/files/4249603/M111.zip)
I have run it with the latest bug fix.

Attached is the text from Pronterface and the g code i ran.  It stops at line 122, a long way short of the end of the code.  This code runs fine in 2.0.4 when i comment out the PRINTER_EVENT_LEDS or earlier version when i don't enable the Neopixels

`#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)`
`//#define PRINTER_EVENT_LEDS`
`#endif`

thinkyhead, 2020-02-25:
How big is your `BUFSIZE`?

Moyster, 2020-02-25:
from the config.zip posted a few post higher, he has  BLOCK_BUFFER_SIZE 32

jwebber23, 2020-02-25:
I'm not sure. I have attached the configuration files a few posts back. It would be the default bigtreetech size for SKR mini E3. I am not in front of the machine until tomorrow.

thinkyhead, 2020-02-25:
The current bug appears to be caused by mis-use (or, over-use) of the `queue.inject_now_P` method from places in the Marlin program flow where it can lead to a freeze. Please give #16967 a test.

thinkyhead, 2020-02-26:
I've merged the patch with additional assurances to prevent a blocked queue, and yesterday patched a click-handling 🤦‍♂ bug in M0/M1.

jwebber23, 2020-02-26:
What version is the fix in?

thinkyhead, 2020-02-26:
All such changes go into `bugfix-2.0.x` of course.

jwebber23, 2020-02-26:
[M111 S3.txt](https://github.com/MarlinFirmware/Marlin/files/4253550/M111.S3.txt)
Hi,

It now gets through the file before the Click to Resume which is great.  The Green LED's come on a good few lines before it is finished running the g code.  Attached is the M111 S3 for this job.

thinkyhead, 2020-02-26:
> The Green LED's come on a good few lines before it is finished running the g code.

That's because they aren't turned on by adding to the queue, but immediately upon reaching the end of the file. We don't have a G-code for "call the LED onPrintCompleted method" but it's possible to add one and force the event into the queue.

jwebber23, 2020-02-26:
Thanks for that. Will this fix make it into 2.0.4?

thinkyhead, 2020-02-26:
Yes, the fix has already been merged into `bugfix-2.0.x`.

jwebber23, 2020-02-26:
Sorry, I meant the non bugfix version 2.0.4?

carl1961, 2020-02-27:
I verify , the fix  merged into bugfix-2.0.x fixes the issue. Thanks @thinkyhead and all who helped!

thinkyhead, 2020-02-27:
@jwebber23 — Now that 2.0.4.4 is out, yerp!