MarlinFirmware/Marlin #16809: [BUG] Getting intermittent "probing failed" what exactly is failing?

ForceConstant, 2020-02-08:
### Bug Description
I have a z probe and do a mesh leveling, and in the last few days I have been getting probing failed on some attempts. I can't tell from source what actually causes this error so am not sure what to check.

### My Configurations

TBP

### Steps to Reproduce
1. G28
1. G29

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

ellensp, 2020-02-09:
This is only used on one place.the code says
```
  if (isnan(measured_z)) {
    stow();
    LCD_MESSAGEPGM(MSG_LCD_PROBING_FAILED);
    SERIAL_ERROR_MSG(MSG_ERR_PROBING_FAILED);
  }
```
So the measured_z height is failing the isnan test. 
So the code could not measure the Z height.  

you could try 
set M111 S32  should give you more details during probing.
Requires DEBUG_LEVELING_FEATURE

CptanPanic, 2020-02-09:
Ok I got debug enabled, and here is the output below. Looks like NaN is coming from failure "SLOW Probe fail!" I don't really understand code to see what that actually means. Ideas?

```
...
Recv:   current_position= X196.25 Y112.50 Z0.17 : sync_plan_position
Recv:   current_position= X196.25 Y112.50 Z0.17 : <<< do_probe_move
Recv:   current_position= X196.25 Y112.50 Z0.17 : <<< run_z_probe
Recv: >>> do_blocking_move_to X196.25 Y112.50 Z3.18
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv:  T:22.30 /0.00 B:22.11 /0.00 @:0 B@:0
Recv: <<< probe_at_point
Recv: >>> probe_at_point(195.00, 117.50, raise, 0, probe_relative)
Recv:   current_position= X196.25 Y112.50 Z3.18 :
Recv: >>> do_blocking_move_to X235.00 Y112.50 Z3.18
Recv: <<< do_blocking_move_to
Recv:   current_position= X235.00 Y112.50 Z3.18 : set_probe_deployed
Recv: deploy: 1
Recv:   current_position= X235.00 Y112.50 Z3.18 : >>> run_z_probe
Recv:   current_position= X235.00 Y112.50 Z3.18 : >>> do_probe_move
Recv: >>> do_blocking_move_to X235.00 Y112.50 Z-0.50
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv:  T:22.27 /0.00 B:22.07 /0.00 @:0 B@:0
Recv:   current_position= X235.00 Y112.50 Z0.15 : sync_plan_position
Recv:   current_position= X235.00 Y112.50 Z0.15 : <<< do_probe_move
Recv:   current_position= X235.00 Y112.50 Z0.15 : <<< run_z_probe
Recv: >>> do_blocking_move_to X235.00 Y112.50 Z3.15
Recv: <<< do_blocking_move_to
Recv: <<< probe_at_point
Recv: >>> probe_at_point(195.00, 156.25, raise, 0, probe_relative)
Recv:   current_position= X235.00 Y112.50 Z3.15 :
Recv: >>> do_blocking_move_to X235.00 Y151.25 Z3.15
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv:  T:22.38 /0.00 B:22.34 /0.00 @:0 B@:0
Recv:   current_position= X235.00 Y151.25 Z3.15 : set_probe_deployed
Recv: deploy: 1
Recv:   current_position= X235.00 Y151.25 Z3.15 : >>> run_z_probe
Recv:   current_position= X235.00 Y151.25 Z3.15 : >>> do_probe_move
Recv: >>> do_blocking_move_to X235.00 Y151.25 Z-0.50
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv:   current_position= X235.00 Y151.25 Z0.08 : sync_plan_position
Recv:   current_position= X235.00 Y151.25 Z0.08 : <<< do_probe_move
Recv:   current_position= X235.00 Y151.25 Z0.08 : <<< run_z_probe
Recv: >>> do_blocking_move_to X235.00 Y151.25 Z3.08
Recv: <<< do_blocking_move_to
Recv:  T:22.34 /0.00 B:21.95 /0.00 @:0 B@:0
Recv: <<< probe_at_point
Recv: >>> probe_at_point(156.25, 156.25, raise, 0, probe_relative)
Recv:   current_position= X235.00 Y151.25 Z3.08 :
Recv: >>> do_blocking_move_to X196.25 Y151.25 Z3.08
Recv: <<< do_blocking_move_to
Recv:   current_position= X196.25 Y151.25 Z3.08 : set_probe_deployed
Recv: deploy: 1
Recv:   current_position= X196.25 Y151.25 Z3.08 : >>> run_z_probe
Recv:   current_position= X196.25 Y151.25 Z3.08 : >>> do_probe_move
Recv: >>> do_blocking_move_to X196.25 Y151.25 Z-0.50
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv:  T:22.23 /0.00 B:22.15 /0.00 @:0 B@:0
Recv:   current_position= X196.25 Y151.25 Z0.01 : sync_plan_position
Recv:   current_position= X196.25 Y151.25 Z0.01 : <<< do_probe_move
Recv:   current_position= X196.25 Y151.25 Z0.01 : <<< run_z_probe
Recv: >>> do_blocking_move_to X196.25 Y151.25 Z3.02
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv: <<< probe_at_point
Recv: >>> probe_at_point(117.50, 156.25, raise, 0, probe_relative)
Recv:   current_position= X196.25 Y151.25 Z3.02 :
Recv: >>> do_blocking_move_to X157.50 Y151.25 Z3.02
Recv: <<< do_blocking_move_to
Recv:  T:22.30 /0.00 B:22.11 /0.00 @:0 B@:0
Recv:   current_position= X157.50 Y151.25 Z3.02 : set_probe_deployed
Recv: deploy: 1
Recv:   current_position= X157.50 Y151.25 Z3.02 : >>> run_z_probe
Recv:   current_position= X157.50 Y151.25 Z3.02 : >>> do_probe_move
Recv: >>> do_blocking_move_to X157.50 Y151.25 Z-0.50
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv:  T:22.19 /0.00 B:21.95 /0.00 @:0 B@:0
Recv:   current_position= X157.50 Y151.25 Z-0.50 : sync_plan_position
Recv:   current_position= X157.50 Y151.25 Z-0.50 : <<< do_probe_move
Recv: SLOW Probe fail!
Recv:   current_position= X157.50 Y151.25 Z-0.50 : <<< run_z_probe
Recv: >>> do_blocking_move_to X157.50 Y151.25 Z2.50
Recv: <<< do_blocking_move_to
Recv:   current_position= X157.50 Y151.25 Z2.50 : set_probe_deployed
Recv: deploy: 0
Recv: >>> do_blocking_move_to X157.50 Y151.25 Z2.50
Recv: <<< do_blocking_move_to
Recv: Error:Probing Failed
Changing monitoring state from "Operational" to "Error: Probing Failed"
Send: M112
Send: N7 M112*38
Send: N8 M104 T0 S0*41
Send: N9 M140 S0*108
Changing monitoring state from "Error: Probing Failed" to "Offline (Error: Probing Failed)"
Connection closed, closing down monitor

```

GMagician, 2020-02-09:
Then probe din't triggered while moving slowly down (target pos -0.5)

Bob-the-Kuhn, 2020-02-09:
Try increasing the range of motion before Marlin declares a probing failure.  

In Configuration.h change **Z_PROBE_LOW_POINT**

ForceConstant, 2020-02-09:
Thanks that did it. Since the low point was very close to my probe offset,
probing didn't have much tolerance.

Bob-the-Kuhn, 2020-02-10:
Good news!

Please close this issue if there are no further problems.

closed 2020-02-10 by boelle