MarlinFirmware/Marlin #16815: [BUG] Marlin 2.0.3 TMC5160 connection error.

SFRSKZ, 2020-02-09:
hi. im using tmc5160s from  bigtree, their version v1.0. i was keeping them for a project and today i tried install them on my skr v1.3, i got connection error. i tried them while marlin 2 still in beta phase and they were working with spi just fine.  

what i did;
i cut the clk pin and solder a wire to gnd. 
all spi jumpers are connected to board. 
also i tried without clk>gnd cable, nothing different.  
tmc library is updated to latest.
TMC_USE_SW_SPI is defined. 

im not sure if i count as a bug, but its a dead-end for me right now. 

here is my config files. 
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4176878/config.zip)

chai-md, 2020-02-10:
Did you remove CLK pin?
https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/issues/2

SFRSKZ, 2020-02-10:
> Did you remove CLK pin?
> [bigtreetech/BIGTREETECH-TMC5160-V1.0#2](https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/issues/2)

yes, i did that. it was the first thing i did.

thinkyhead, 2020-02-11:
> im not sure if i count as a bug, but its a dead-end for me right now.

It doesn't, and we really don't have the resources to handle support in this issue queue. For best results getting help with configuration and troubleshooting, please use the following resources, where you can find someone else with your configuration:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-02-11 by thinkyhead