MarlinFirmware/Marlin #16819: [BUG] Dual Z, very bad sync after "Coolstep" implementation

PGrave, 2020-02-09:
-After flashing my SKR Pro with Marlin bugfix version from today, aprox. 15:00, my Z dual drive setup has a very rough motion and noise, ultimately halting the system. Had some problems compiling before, Z2 and Coolstep related, as Sanity check would require Z2 to be setup but Z2 drive configuration was removed from Configuration. h, on the Coolstep implementation.

### My Configurations
My configurations are included in the zip file, hardware is:
Ender 3 Pro; BTT SKR Pro V1.1; 5x BTT TMC2209 V1.0; dual Z steppers; 
[MyConfigs.zip](https://github.com/MarlinFirmware/Marlin/files/4177337/MyConfigs.zip)
### Steps to Reproduce
1. Compile Marlin bugfix;
2. Flash firmware;
3. Move Z axis and notice rough, unsynched movement between Z steppers

**Expected behavior:** [What you expect to happen]
-Sync between Z steppers

**Actual behavior:** [What actually happens]
-Unsynched movement between Z steppers

daleckystepan, 2020-02-09:
I have tried to disable Coolstep using
#define Z_COOLSTEP_LOWER_LOAD_THRESHOLD   0
without any success. But you can try it on your own.

PGrave, 2020-02-09:
I did try to disable Coolstep for Z, with no success.
I think that something has happened with the steps/mm assignment for the 2nd Z stepper: it seems it stays with the ones configured for E, instead of those for Z. In my case, that means almost half of the movement, which could explain the grinding and the machine halting.

thisiskeithb, 2020-02-10:
CoolStep was removed in 18a7276909a3e2aa1761ba0c46577a8f6e7baedb. It wasn't ready for use after all.

Fabi0San, 2020-02-11:
@PGrave can you please test with the latest bits?
We need to disambiguate whether the problem was with CoolStep or with the new way to config Z2, etc.
As described in this comment by @thinkyhead: https://github.com/MarlinFirmware/Marlin/pull/16823#issuecomment-584397268

thinkyhead, 2020-02-11:
In future all this driver malarkey will be organized more consistently. I had hoped to get a jump on that by extending the auto-assignment stuff, but the codebase needs more massaging before it will be ready. So, please grab the latest Marlin, replace the old driver type options in your config, and we'll revisit this again when things are more in place.

Moyster, 2020-02-11:
@Fabi0San so wait, according to the comment, if one has a tmc2209 as Z2 plugged (on the board) in E1, does that mean "#define E1_DRIVER_TYPE" defines the correct driver for Z2 ?  
I had the same issue with coolstep (re-arm 2x tmc2209 for Z1 & 2) but thought "#define Z_DRIVER_TYPE" would define the driver for both Zx ?  
If so, maybe a sanitycheck is necessary, something like "if NUM_Z/X_STEPPER_DRIVERS >= 2 and E1_DRIVER_TYPE is undefined = error" ?  

With coolstep on, and a clean eeprom, Z2 seems to not load the Z2 configs and the TMC driver menu only shows X / Y / Z / E1 (Z2 is gone, in any submenu, hybrid threshold / current / etc).

Fabi0San, 2020-02-11:
Yes, @thinkyhead  will correct me if I'm wrong but the goal was to  Z_DRIVER_TYPE set the driver type for both, but that has since been reverted https://github.com/MarlinFirmware/Marlin/commit/71539bcbeb68284663d0627207ffd5a9a377e920

So sync to latest bugfix, revert to your old ways of defining types for z and z2  and hold on for CoolStep until we untangle this.

PGrave, 2020-02-12:
@Fabi0San , I didn't test further, but my thoughts were just how I described in OP: not a Coolstep problem in itself, but a drive assignment logic problem. Thank you all for the help, I'll proceed as suggested by @thinkyhead and close this issue.

closed 2020-02-12 by PGrave