MarlinFirmware/Marlin #16820: Skr 1.3, bltouch does not deploy when homing

ppetrov-dev, 2020-02-09:
### Bug Description

I have Sapphire S (CoreXY) and use marlin 2.0 on SKR 1.3 with TMC2208, also added a clone of bltouch (3dtouch). The problem is with the following setup (connected to Z Max plug): 
```
#define USE_XMIN_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG

//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN 
#define Z_MIN_PROBE_PIN P1_24 //to use Z_MAX plug

#define X_HOME_DIR -1
#define Y_HOME_DIR 1
#define Z_HOME_DIR 1
```
I have no idea anymore how to force BlTouch works properly with ZMAX or ZMIN (tried both) without that setting ```Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN``` is defined.

I've looked at all of the BLTouch documentation as well, online and what I could find from others using it with the SKR 1.3 but haven't found anything with someone using the Z_MAX_ENDSTOP as the probe pin. Does it have to be possible?
### My Configurations

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4177404/configs.zip)
 `Configuration.h` and `Configuration_adv.h` files are attached.

### Steps to Reproduce

1. Home Z

**Expected behavior:** [What you expect to happen]
The Z motor is going to Z MAX (to up), bltouch deploys and waits for a touch 

**Actual behavior:** [What actually happens]
The Z motor is going to Z MAX (to up), but bltouch does not deploy

#### Additional Information
What I see on the printer started.
[image]

Bltouch works properly with the following settings (connected to Z Min plug):
```
#define USE_XMIN_PLUG
#define USE_ZMIN_PLUG
#define USE_YMAX_PLUG

#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN 

#define X_HOME_DIR -1
#define Y_HOME_DIR 1
#define Z_HOME_DIR -1
```
....but Z Motor is going **down**!!11

Please, HELP!

FreestylePocker, 2020-02-09:
Encountered this bug after updated to recent bugfix commit `38bf5523e147e10f1454498d2260573c48a2aab4`, it occurs after M500 or sometimes after cold start, dunno the reason. When attempting to home for 2nd time after reset button it works fine.
SKR PRO 1.1 + RepRapDiscount Full Graphic Smart Controller

ellensp, 2020-02-10:
On your printer the Z endstop is at the top of travel. this is the Z-min position. not the z-max.
you need #define Z_HOME_DIR -1

You like everyone else, gets this wrong. 
You need to look at the distance between the nozzle and the bed.
When the bed is up, the distance in minimum, this is you min position.
When the bed is down, the distance is maximum. this is yoru max position.
When you click on z+ the distance between the hot end and teh bed must increase, so the bed must lower
When you click on z- the distance between the hot end and the bed must decrease, so the bed must raise

"....but Z Motor is going down!!11"
Set your #define INVERT_Z_DIR true.

ppetrov-dev, 2020-02-10:
> On your printer the Z endstop/probe is at the top of travel. this is the Z-min position. not the z-max.
> you need #define Z_HOME_DIR -1
> 
> You, like everyone else, gets this backwards.
> You need to look at the distance between the nozzle and the bed.
> When the bed is up, the distance is at minimum, this is your min position.
> When the bed is down, the distance is at maximum. this is your max position.
> When you click on z+ the distance between the hot end and the bed must increase, so the bed must lower
> When you click on z- the distance between the hot end and the bed must decrease, so the bed must raise
> 
> "....but Z Motor is going down!!11"
> Set your #define INVERT_Z_DIR true.

that makes sense, I'll try at the lunchtime=)

boelle, 2020-02-10:
remember to close if it works

ppetrov-dev, 2020-02-10:
It worked! Thanks, **ellensp**!

closed 2020-02-10 by ppetrov-dev