MarlinFirmware/Marlin #16821: [BUG] Babystepping, Z probe offset accuracy

FreestylePocker, 2020-02-09:
### Bug Description
Hardware:
- SKR PRO v1.1
- RepRapDiscount Full Graphic Smart Controller
Sofwtware:
- Marlin bugfix-2.0.x commit 38bf552

  
  
### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

joewmorrissey, 2020-02-16:
Thank god, i'm not the only one to notice. This issue is not only on just baby stepping but on any movements. 

display menu -> motion -> move axis -> z - move by 10mm  - same issue

thinkyhead, 2020-02-16:
Try setting `Z_PROBE_OFFSET_RANGE_MIN` to -5 and `Z_PROBE_OFFSET_RANGE_MAX` to 5.

```cpp
#if Z_PROBE_OFFSET_RANGE_MIN >= -9 && Z_PROBE_OFFSET_RANGE_MAX <= 9
  #define LCD_Z_OFFSET_FUNC(N) ftostr54sign(N)
  #define LCD_Z_OFFSET_TYPE float43
#else
  #define LCD_Z_OFFSET_FUNC(N) ftostr52sign(N)
  #define LCD_Z_OFFSET_TYPE float52
#endif
```

joewmorrissey, 2020-03-05:
> Try setting `Z_PROBE_OFFSET_RANGE_MIN` to -5 and `Z_PROBE_OFFSET_RANGE_MAX` to 5.
> 
> ```c++
> #if Z_PROBE_OFFSET_RANGE_MIN >= -9 && Z_PROBE_OFFSET_RANGE_MAX <= 9
>   #define LCD_Z_OFFSET_FUNC(N) ftostr54sign(N)
>   #define LCD_Z_OFFSET_TYPE float43
> #else
>   #define LCD_Z_OFFSET_FUNC(N) ftostr52sign(N)
>   #define LCD_Z_OFFSET_TYPE float52
> #endif
> ```

Made this change in menu.h and still have 2 digits after decimal.

thinkyhead, 2020-03-06:
This might be better:
```cpp
#if Z_PROBE_OFFSET_RANGE_MIN >= -9 && Z_PROBE_OFFSET_RANGE_MAX <= 9
  #define LCD_Z_OFFSET_FUNC(N) ftostr54sign(N)
  #define LCD_Z_OFFSET_TYPE float52
#else
  #define LCD_Z_OFFSET_FUNC(N) ftostr52sign(N)
  #define LCD_Z_OFFSET_TYPE float43
#endif
```

skipfie, 2020-03-23:
Even after changing that block in menu.h, and setting the min and max to +/- 5, I'm still only seeing 2 decimal places on the LCD. Marlin 2.0.5.1

Allan-N, 2020-04-18:
Would it be simpler to just use ftostr53_63() ?

closed 2020-07-29 by github-actions[bot]