MarlinFirmware/Marlin #16822: [QUESTION] Move extruder steppers separately from XYZ steppers?

JupiterSymphony, 2020-02-09:
### Description

How can I move the extruder separately from the XYZ steppers? I want to be able to have the extruder move  at certain times regardless of the XYZ moves going on. 

I'm not sure from the firmware how I can directly call the E stepper motor to a position or how to dynamically manage it in motion/planner/queue classes. Any direction and help is greatly appreciated and my apologies if this goes against your convention for issues, I haven't been able to get help in the mentioned forums.

JupiterSymphony, 2020-02-09:
Ideally I'd be able to do something simple like this while the XYZs are going about their business from Gcodes:

```
if (someFlag) {
  eStepper.move(1);
} else {
  eStepper.move(-1);
}
```

Or if it's more to do with an object needing updating dynamically, such as current destination, then something like this:
```
if (someFlag) {
  destination.e = 1;
} else {
  destination.e = 0;
}
```

FreestylePocker, 2020-02-09:
Make the axes unaware of each other positions... so they to just go out of sync? Dunno what this may be needed for.
In that case easy solution is to take 2nd board, and use it for extruder, lol.

JupiterSymphony, 2020-02-09:
It's for some project, adapting the extruder stepper to control stuff but need it controled on a schedule. If I go with a second board, what's the recommended way to synchronize instructions between main board and 2nd board?

FreestylePocker, 2020-02-09:
You need dedicated processor, computer or some microcontroller. Both borads connected to it (via serial interface). I would use some programming language (python  for example) to read gcode, process strings, split on two serial consoles and feed boards... yup, it's complicated if you wish to do the sync two separate serial consoles(boards)

boelle, 2020-02-10:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-02-10 by boelle

thinkyhead, 2020-02-10:
@JupiterSymphony — Jump on to the Discord to discuss the idea. At this time it is not possible because everything goes into a single motion planner queue. All moves are therefore coordinated among all axes. However, as suggested, you could add a separate processor and use some kind of communication, such as i2c or serial. We've played with [things like that in the past](https://github.com/MarlinFirmware/Marlin/issues/4776#issuecomment-246262879).

JupiterSymphony, 2020-02-10:
@FreestylePocker @thinkyhead Thanks for the responses and information. Seems like that is indeed the way to go.