MarlinFirmware/Marlin #16824: [BUG] auto home (G28) fails with SENSORLESS_PROBING only

redkuul, 2020-02-10:
### Bug Description

I recently upgraded my Anycubic Kossel Linear Plus with an SKR 1.3 board and TMC2209 drivers. 
Currently I am able to print as I did with the default board. 

When I change my settings to SENSORLESS_PROBING, autohoming (G28) fails. It seems that it is triggered instantaneously, while I expect it to use the endstops. 

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4182469/Configuration.zip)
### Steps to Reproduce

1. Have a working delta printer
2. Turn on SENSORLESS_PROBING only, thus not SENSORLESS_HOMING
3. Start autocalibration (G33)

**Expected behavior:** 
Autohoming (G28) using endstops, followed by delta calibration using sensorless probing (G33). 

**Actual behavior:**
Autohome (G28) fails. 

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
[videos.zip](https://github.com/MarlinFirmware/Marlin/files/4182539/videos.zip)
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thinkyhead, 2020-02-11:
I can't see `SENSORLESS_PROBING` having any effect, other than to enable the options for stall sensitivity in-general. How do you have your physical endstops wired up in relation to your DIAG pin connections on the board? These may not interact well when combined together.

redkuul, 2020-02-13:
I couldn't find it either. That's why I reported a bug and not a fix. I have connected the endstops to gnd and signal of X+, Y+ and Z+. Which works, as you can see in the video. Driver diag pins are connected to the default SKR1.3/ BTT TMC2209 pins.

mgautam98, 2020-02-15:
I think it is fixed in #16736. Could you test the bugfix-2.0.x branch, tell me if it fixes your issue

redkuul, 2020-02-25:
Hello both, 

I've tried both bugfix-2.0.x and 2.0.4.3.
I found that when STALL_SENSITIVITY is set to 0 homing does work with the endstops (otherwise false positive right away). However, this of course inhibits proper working of sensorless probing. 

Also, I found M914 is not working as expected. If I set STALL_SENSITIVITY to >0 in configuration_adv.h, and via M914 to 0, the machine also doesn't home using the endstops. 

I suspect that when I set STALL_SENSITIVITY to 0, I disable other functions as well. 

I have also tried to use STALL_SENSITIVITY of 1, 64, 255 and start at the top side of the printer, so homing is almost correct. This results also in no triggering of the probing. 

Is my understanding that the SKR1.3 board does not need additional cables for the diag pin correct?

thinkyhead, 2020-02-26:
> Should I commit these changes, or shall I share them via this post?

Please share. It sounds like some may not be vital.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle