MarlinFirmware/Marlin #16837: [Question] Auto_build_marlin

GMagician, 2020-02-11:
Is there any reason why auto build open a new console everytime it is started?
It's only me that has to close everytime a lot of unused terminal windows?

Bob-the-Kuhn, 2020-02-14:
Try re-using the window.  There is a right click "Repeat Build" option that will do a new buid and display the results in the current window.

The only times the "Repeat Build" doesn't work is:
- When the motherboard is changed.
- When switching branches.  Sometimes PlatformIO doesn't notice changes so I always do a  "Remove project folder, add project folder and select 'Add to this window and reset state'" sequence.

GMagician, 2020-02-14:
> Try re-using the window. There is a right click "Repeat Build" option that will do a new buid and display the results in the current window.

Can't see it... where in vscode windows does it respond to such right click?

thinkyhead, 2020-02-15:
> Is there any reason why auto build open a new console everytime it is started?
> It's only me that has to close everytime a lot of unused terminal windows?

I'm working on the next version. I intend to make it use the same terminal by default, but maybe have a checkbox option to launch a new one for the build.

https://www.youtube.com/watch?v=UW3uKxxC1Vk

closed 2020-02-15 by thinkyhead