MarlinFirmware/Marlin #16838: [BUG] #16818 seems to corrupt communication

GMagician, 2020-02-11:
It seems that above reported PR corrupt all communications. I'm investigating

GMagician, 2020-02-11:
Well well well well, it seems that after each gcode line is seens twice
so 'line number is not last line number+1' error is sent

Tomorrow I'll investigate what is generated by compiler...

vivian-ng, 2020-02-12:
Will this affect printing from SD card? I have been trying to start a print from the SD card after flashing with latest bugfix-2.0.x, but the printer seems to just do nothing.

vivian-ng, 2020-02-12:
I reset to before the commit for [PR#16818](https://github.com/MarlinFirmware/Marlin/pull/16818) and can now start a print from SD card. Seems like the commit from PR#16818 onward may have some code that affects how gcode is read. Hopefully someone with better knowledge of how the PR was implemented (like @thinkyhead ) can shed some light.

GMagician, 2020-02-12:
@vivian-ng is it may affet also sd printing. Common code. This evening I'm gonna check it

PR #16840 (2020-02-14): Fix line parsing
Fix #16838
closed 2020-02-14 by GMagician