MarlinFirmware/Marlin #16839: [BUG] Z homing using a probe should be allowed when using a custom probe pin

stantond, 2020-02-11:
### Bug Description

1. When using a motherboard with a dedicated probe bin, `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` must be disabled in `configuration.h`.
In `Marlin/src/inc/Conditionals_LCD.h`:
2. Having commented out the above, `HAS_CUSTOM_PROBE_PIN` is `true`. This makes sense.
3. However, `HOMING_Z_WITH_PROBE (Z_HOME_DIR < 0 &&!HAS_CUSTOM_PROBE_PIN)` means that, because we have a custom probe pin, `HOMING_Z_WITH_PROBE` will be false.
4. I had to change the above to ``HOMING_Z_WITH_PROBE (Z_HOME_DIR < 0)` to enable homing using my BL Touch plugged into the dedicated ports on my SKR 1.4 Turbo.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

Using a board with a dedicated Z probe pin, disable `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` (because this is the correct setting for this configuration, and try to home Z.

**Expected behavior:** [What you expect to happen]

Z Probe works

**Actual behavior:** [What actually happens]

Z Probe does not deploy during Z homing

#### Additional Information

* Reproduced by multiple people, see https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/207

emaayan, 2020-02-19:
seems like it was done here https://github.com/MarlinFirmware/Marlin/issues/15731

stantond, 2020-02-19:
> seems like it was done here #15731

Define "done"

emaayan, 2020-02-19:
> > seems like it was done here #15731
> 
> Define "done"

as in suggested, the guy tried several combinations,,  look at option 7 which seems to be doing the same code you suggested.
this guy went though several combinations, of trying to get his bl-touch AND z-min end stop to work together and failed,. and they closed it because he wouldn't split it into FR and a BUG, (which to me seems like a BUG)
i had a rather long discussion about this on facebook , he said in order to  use the custom connections on skr, you need to also use the z min which i tried, and that's how i arrived to that bug. 
seems that currently no matter how you look at it, marlin does not support using BL-touch in anything other then z-min 
https://www.facebook.com/groups/755135144882809/permalink/996056110790710/

stantond, 2020-02-19:
AFAIK, there is no logical reason why Marlin should not allow you to home Z with a probe when using a dedicated probe pin, given it happily lets you home when not using a dedicated probe pin, all because of the one line of code I've highlighted.

That's the only problem in scope for this issue, and it's quite separate (and much simpler) than #15731, which I've also read previously. That one seems to be about using sensorless homing + endstops. I don't agree with closing it for administrative nonsense, it's still a valid issue, just needs renaming to a FR, but again, out of scope for this issue.

stantond, 2020-02-19:
I've re-read #15731, got muddled with another issue - I've commented on it, I think this issue solves part of that one

InsanityAutomation, 2020-02-20:
So realistically that check should look at use_zmin_plug to determine if the z min endstop is being used aside from the probe for homing. Any checks ensuring z min plug should also check for the probe to be in use. If you have both in use, home with the sensor only. Seems straightforward enough anyway.

Releece, 2020-03-09:
This fix is much needed for everyone with SKR1.4's and other boards with custom probe pins that use Z probes. Fortunately this bug-report is nicely concise instead of #15731.

RC-Thoughts, 2020-03-30:
Adding a upvote for this, also on SKR1.4 + BLTouch 3.1.

albertogg, 2020-04-02:
Does anyone know what's the reasoning behind the deactivation of the Z homing with a probe when using a custom PIN? I've tracked and this was added in https://github.com/MarlinFirmware/Marlin/pull/14338. All boards with dedicated probe pins are affected by this.

libby2cm, 2020-04-15:
Maybe i'm just not thinking clearly, but having run into this issue (bltouch with skr mini e3 v1.2), i'm wondering if it's even physically possible to make the endstop switch and ABL sensor (2 binary sensors) work together reliably.

Getting the traditional zlimit switch and bed level dialed while factoring in temperature fluctuations is one of the biggest PITA's that ABL inherently accounts for.  But when using a simple binary switch with an ABL sensor, you have a very small (and dynamic) margin to get right, otherwise best case, your mesh will be virtually floating above the bed so to speak, or your nozzle is going to ram into the bed to some degree, as the zlimit switch effectively becomes a single point secondary bed sensor.  I can't wrap my head around how to properly calc the z offset reliably either without alot of tinkering.

Maybe instead of homing, the zlimit switch needs to be purely a estop 

am i missing something?  my head does hurt in my defense right now :)

libby2cm, 2020-04-15:
OK, i'm lost. Is the goal to make zhoming and ABL both work via probe, then just drop the zstop switch 5mm or so below what is realistic so it's effectively only a estop in case probe fails?

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle