MarlinFirmware/Marlin #16843: [BUG] Movement micropauses when serial connected (SKR Mini E3 1.2 + BTT TFT Screen + Octoprint)

gonzaloalbito, 2020-02-12:
### Bug Description
I noticed some of my prints are getting bubbles too oftenly. These started eventually at some random point during the print and then repeats constantly at random points of each layer or wall until I reboot (shut down and turn on) the printer (or, as I discovered later, disconnecting the serial connected devices like TFT screen and octoprint).
After taking a look this, I see some movement micropauses which match the bubbles location.

After monitoring and discarding issues with steppers, drivers and SOC temperatures, I noticed micropauses and bubbling dissapeared after I disconnect the Octoprint USB and TFT RS-232 screen cable from the motherboard during a live disassembly (while printing). It was bubbling for about 100 layers before I disconnected these two cables and bubbles and micropauses completely dissapeared after that.
I'm not sure the issue is caused by having both connected or only one of them, but bubbles are gone since I keep both disconnected from my printer. I will continue testing it when I get some free time.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4191457/Configuration.zip)

### Steps to Reproduce

1. Connect the Octoprint and/or the BigTreeTech TFT35 screen
2. Start a print
3. Wait
4. Micropauses will start randomly at some point

**Expected behavior:** Reasonably smooth print

**Actual behavior:** Bubbly print

#### Additional Information

Bubbles: [image]

3xMicropauses video (most noticeable at the end of second 2): https://youtu.be/z7HCfQwK81E

bumpaneer, 2020-02-12:
This is also being discussed on the BTT firmware page: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/282

gonzaloalbito, 2020-02-12:
So it seems the problem is having both connected at the same time... I will try the proposed solution (connecting Rpi to the TFT and then the TFT to the board)

vicwomg, 2020-02-20:
I am having this issue as well. So far I haven't seen confirmation that the serial wiring fixes it. Can anyone confirm?

dinamitemic, 2020-02-25:
Same problem here experienced both with vases or 3dbenchy.
I'm using only octoprint on a raspberry pi 3, no uart displays connected.
Now I've increased the baudrate from 115200 to 250000, I'll try it.  If the problem persist I'll try also printing from pc so we can exclude a marlin related problem and will be octoprint the responsible.

robstermn, 2020-03-12:
I am having this issue as well, with SKR Mini E3 1.2 + BTT TFT Screen V35 + Octoprint.
Printing from Octoprint, I get these small zits all over.
I've increased the baudrate from 115200 to 250000 with the same issue.

[image]

[image]

If I print from the TFT display via micro SD, prints fine???

[image]

[image]

gonzaloalbito, 2020-03-13:
It seems pretty randomly.
I'd been able to do some prints with both connected without noticeable problems and sometimes these starts at mid print, so maybe you have been lucky with your TFT prints... (i'm usually doing several hours prints, like 10-15h prints)
I'm always printing from motherboard microSD, only using BTT screen for aux functions like bed leveling, preheat, stepper movements and so, and octoprint for option tuning by gcode and fixed time timelapses...
My blobs seems smaller than yours so II guess the problem exists no matter where are you printing from but it gets bigger if you're printing from the serial interface. It may also be related to my printing speed (40-60mm/s general speed which is about 20-30mm/s for perimeters)

jetshine, 2020-03-25:
I tried connecting the RPI to the TFT as mentioned.  I'm using BTT SKR E3 Mini + BTT TFT E3 V3.0 + RPi 3B.  I'm connected from Serial0 on the RPi (pins 8/10) to UART3 on the TFT board.
I still notice the print movements stalling, resulting in small zits and blobs.

I am going through the code to see what can be done. loopBackEnd() in menu.c seems to be where all the magic is happening, but I see that the sending from the command queue, acks, and receiving new gcodes are all done serially (in that order).  Perhaps the queue is running out too fast?

gonzaloalbito, 2020-03-26:
I'd been printing for a whole week without any stop due to COVID-19 quarantine measurements and all stuff we're printing for hospital workers in my country, and I'd noticed the same issue when only the TFT screen is connected (latest FW version 3.0.25.2). I noticed it's happening without the raspberry but it seems less frequent (maybe once each 2 or 3 layers depending the print size, with raspberry it was about once or twice each layer).

I'd also noticed the BTT TFT screen stops communicationg properly after a few minutes. Screen menus are working but selected actions doesn't reach the printer.

It seems completely a BTT TFT firmware issue for me.

BastR, 2020-04-28:
https://github.com/MarlinFirmware/Marlin/issues/16884

odaruc, 2020-06-01:
Is anyone seeing line number errors with TFT attached at the same time as Octoprint? If the host is having to retry transmissions then this could explain the bubbles maybe. I submitted a PR today to solve my own issue with TFT and Repetier server, and wondering now if the two issues are related.

Here is the PR for Marlin 2.0 bugfix branch if anyone wants to try to see if it has any impact on the bubble issue or not. The changes are in queue.cpp -- you could simply disable line number checking error to see if it makes a difference or not without merging the rest of it.

https://github.com/MarlinFirmware/Marlin/pull/18165

boelle, 2020-06-21:
@gonzaloalbito still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle