MarlinFirmware/Marlin #16845: [BUG] Unable to build using both TMCStepper and U8glib

Silverpumax, 2020-02-12:
### Bug Description

If i using the libs TMCStepper and U8glib together the bulding process **fails at linktime**.
Bulding with TMCStepper alone is Ok
Bulding with U8glib alone is Ok

I tested this behaviour also with Arduino 1.9beta and eclipse arduino cdt
For testing i set the Arduino path , the TMP path and  the Marlin path short as possible to the route of the filesystem. I tested this  behaviour also with the last branch bugfix.2.0x.

> _By using a older bugfix.2.0x. with STRING_DISTRIBUTION_DATE "2019-08-08" and by using the new libs i can build and run the system with no problems !_

### My Configurations
The only change in `Configuration.h` and `Configuration_adv.h` 
#define MOTHERBOARD BOARD_MKS_GEN_L 
#define X_DRIVER_TYPE  TMC5160 
#define MKS_MINI_12864
#define X_CS_PIN            59          
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4192059/configuration.zip)

### Steps to Reproduce

Build with:
Last branch: 2.0x or last branch bugfix.2.0x.
Arduino: 1.8.11 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

**Actual behavior:** 
Error at:
`Linking everything together...
avr-gcc: error: CreateProcess: No such file or directory
Bibliothek U8glib in Version 1.19.1 im Ordner: C:\Users\Becker\Documents\Arduino\libraries\U8glib  wird verwendet
Bibliothek TMCStepper in Version 0.6.2 im Ordner: C:\Users\Becker\Documents\Arduino\libraries\TMCStepper  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: C:\Program Files wird verwendet
Bibliothek SoftwareSerial in Version 1.0 im Ordner: C:\Program Files wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Arduino Mega or Mega 2560.
`

I notice also errors at compile time with no consequence to the build if i use the libs alone

`Error while detecting libraries included by C:\Temp\arduino_build_763872\sketch\src\feature\tmc_util.cpp`
`Error while detecting libraries included by C:\Temp\arduino_build_763872\sketch\src\module\stepper\trinamic.cpp`

#### Additional Information

[ArduinoErrorOutput.zip](https://github.com/MarlinFirmware/Marlin/files/4192048/ArduinoErrorOutput.zip)

ellensp, 2020-02-12:
The error is avr-gcc: error: CreateProcess: No such file or directory
Your a windows user and your hitting the windows command line limit,
You need to remove unused src/HAL directories and if not using it remove Marlin/src/lcd/extensible_ui

Silverpumax, 2020-02-13:
Thanks for your answer.
I found some issues for that but no conclusion.
Evidently the toolchain must use sh.exe instead cmd.exe !??!  But I haven't a clue.

Therefore my solution has been prove satisfactory.
By installing Ubuntu Linux with Arduino in a Windows Hyper-V enviroment for the only benefit to build marlin.

Warbonic, 2020-02-16:
removing src/HAL does not compile. I agree that removing Marlin/src/lcd/extensible_ui does work.

Has anyone had any results on this compiling u8glib & tmc2130 together with boxed Marlin?
Out of the box compiling is preferable to searching the internet for days to find that removing Marlin/src/lcd/extensible_ui is the only viable option.

sjasonsmith, 2020-02-16:
Have you tried switching to Visual Studio code with Platform I/O? That doesn't have the same issue with command line length that Arduino has. I can build the posted configurations with no issue in VS Code.

sjasonsmith, 2020-02-16:
> removing src/HAL

You can't just remove all of src/HAL, you can only remove sub-folders that you do not need. Since you presumably build for AVR, you could remove all the `HAL_*` sub-folders except HAL_AVR.

Protean-Man, 2020-02-16:
I have not used Visual Studio yet but will now. Thank you.
So far it has been a Marlin issue in my head, it being a Arduino IDE problem is helpful.

Silverpumax, 2020-02-16:
Yes i can remove `HAL_*` sub-folders except HAL_AVR `and except HAL_DUE`.
But the problem maintains.

I'd like try switching to Platform I/O with **eclipse**....

Until now i build marlin with Arduino in a Ubuntu Linux which is installed in a Windows 10 Hyper-V environment.

thinkyhead, 2020-02-16:
**PlatformIO** in **Visual Studio Code** with the **Auto Build Marlin** extension will be the recommended build method going forward. Configuration tools will be integrated into the ABM extension bit by bit.

thinkyhead, 2020-02-16:
As pointed out, Arduino can't filter project files in the way that PlatformIO can. So to get Arduino IDE to complete the build, the solution is to delete the HAL sub-folders that you are not using. This is not something we can address without a big re-architecting of Marlin.

closed 2020-02-16 by thinkyhead