MarlinFirmware/Marlin #16846: [FR] Extend or retract safe the Bltouch

Multie, 2020-02-12:
hello, I have installed a BlTouch on my 3d printer. But this does not always extend or retract, so I wanted to catch this problem by checking the condition after the change. Since this is already partially contained in Marlin:

Marlin_main.cpp Line 2147:
```c
bool set_bltouch_deployed(const bool deploy) {
      if (deploy && TEST_BLTOUCH()) {      // If BL-Touch says it's triggered
        bltouch_command(BLTOUCH_RESET);    //  try to reset it.
        bltouch_command(BLTOUCH_DEPLOY);   // Also needs to deploy and stow to
        bltouch_command(BLTOUCH_STOW);     //  clear the triggered condition.
        safe_delay(1500);                  // Wait for internal self-test to complete.
                                           //  (Measured completion time was 0.65 seconds
                                           //   after reset, deploy, and stow sequence)
        if (TEST_BLTOUCH()) {              // If it still claims to be triggered...
          SERIAL_ERROR_START();
          SERIAL_ERRORLNPGM(MSG_STOP_BLTOUCH);
          stop();                          // punt!
          return true;
        }
      }
      #if ENABLED(BLTOUCH_V3)
        #if ENABLED(BLTOUCH_FORCE_5V_MODE)
         bltouch_command(BLTOUCH_5V_MODE);
        #elif
          bltouch_command(BLTOUCH_OD_MODE);
        #endif
      #endif
      bltouch_command(deploy ? BLTOUCH_DEPLOY : BLTOUCH_STOW);
      #if ENABLED(BLTOUCH_V3)
        if(deploy)
          bltouch_command(BLTOUCH_SW_MODE);
      #endif

      #if ENABLED(DEBUG_LEVELING_FEATURE)
        if (DEBUGGING(LEVELING)) {
          SERIAL_ECHOPAIR("set_bltouch_deployed(", deploy);
          SERIAL_CHAR(')');
          SERIAL_EOL();
        }
      #endif
      return false;
    }
```
  
It would be better that more attempts are made to achieve the desired condition. As this ensures that the deployed or stowed state is safely reached and the printer does not accidentally drive into the print bed or interfere with the measurement. Unfortunately I can't find my way in the code and I don't know how to get the state of the Bltouch.

Marlin_main.cpp Line 2173:
```c
bool set_bltouch_deployed(const bool deploy) { 
     do  {
      bltouch_command(deploy ? BLTOUCH_DEPLOY : BLTOUCH_STOW);
    } while (/* check for state with escape counter*/)
  }
```
  
  
Sincerely yours
Multie9000

FreestylePocker, 2020-02-13:
Hi, I have bug on my SKR PRO after M500 command sometimes probe not deploys. May be somehow related?
There were another problem with random probe triggers related to PWM fan signal, that were solved by shielded cable from old PS/2 keyboard.

ManuelMcLure, 2020-02-13:
First of all you should really move to 2.0 - we are no longer adding enhancements to 1.1.x and the BLTouch code has been greatly improved there.

Multie, 2020-03-25:
Thank you for your answer.  Unfortunately I can't update to Marlin 2.x because the printer's CPU doesn't have enough memory.  I can program the solution to the problem myself.  The only thing I'm missing is the variable or the function with which I can read the status of the BlTouch.

sjasonsmith, 2020-09-11:
I am closing this feature request for several reasons:

1. Marlin 1.x is no longer developed. Changes such as this will not be made there.
2. Recent changes have greatly improved BLTouch reliability on STM32 platforms, which hopefully addresses theissue @FreestylePocker mentioned.
3. I don't believe there is any way to tell if the pin is deployed or not. You control the BLTouch through PWM output only, there is no 2-way communication channel.
4. "Changing more times" has no effect, because Marlin always retains the PWM command. When you send the "stow" or "deploy" command, it sends that command continuously until a new command is selected.

For remaining BlTouch issues, I suggest using the following resources to obtain help:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-11 by sjasonsmith