MarlinFirmware/Marlin #16847: [BUG] UBL fails at point 32 of 49

Blaster1920, 2020-02-12:
### Bug Description

Following ther UBL standard procedure, the probe stops at probing point 32/49. EEPROM can reach up until 89 points. Fixed mount probe wired to z-stop. Inset margin 10. Ender 3, with SKR e3 mini 1.2. Marlin bugfix updated at 12.02.2020
### My Configurations

http://s000.tinyupload.com/index.php?file_id=70752647368761022508

### Steps to Reproduce

```
Flash FW
M502
M500
M501
G28
G29 P1
```

**Expected behavior:** [What you expect to happen]

Get a complete mesh of the bed by probing all the estimated points.

**Actual behavior:** [What actually happens]

Stops at 32th point

#### Additional Information

UBL VERBOSE REPORT
```
***edited out
16:49:25.546 : Probing mesh point 29/49.
16:49:25.546 : >>> Probe::probe_at_point(153.33, 189.17, raise, 0, probe_relative)
16:49:25.547 : current_position= X59.00 Y209.17 Z6.70 :
16:49:25.547 : >>> do_blocking_move_to X202.33 Y209.17 Z6.70
16:49:25.547 : <<< do_blocking_move_to
16:49:26.832 : current_position= X202.33 Y209.17 Z6.70 : Probe::set_deployed
16:49:26.832 : deploy: 1
16:49:26.832 : current_position= X202.33 Y209.17 Z6.70 : >>> Probe::run_z_probe
16:49:26.832 : current_position= X202.33 Y209.17 Z6.70 : >>> Probe::probe_down_to_z
16:49:26.833 : >>> do_blocking_move_to X202.33 Y209.17 Z-0.27
16:49:26.833 : <<< do_blocking_move_to
16:49:29.649 : current_position= X202.33 Y209.17 Z1.76 : sync_plan_position
16:49:29.649 : current_position= X202.33 Y209.17 Z1.76 : <<< Probe::probe_down_to_z
16:49:29.650 : current_position= X202.33 Y209.17 Z1.76 : <<< run_z_probe
16:49:29.650 : >>> do_blocking_move_to X202.33 Y209.17 Z6.76
16:49:29.650 : <<< do_blocking_move_to
16:49:31.149 : <<< Probe::probe_at_point
16:49:31.149 : Probing mesh point 30/49.
16:49:31.149 : >>> Probe::probe_at_point(153.33, 10.00, raise, 0, probe_relative)
16:49:31.150 : current_position= X202.33 Y209.17 Z6.76 :
16:49:31.150 : >>> do_blocking_move_to X202.33 Y30.00 Z6.76
16:49:31.150 : <<< do_blocking_move_to
16:49:32.768 : current_position= X202.33 Y30.00 Z6.76 : Probe::set_deployed
16:49:32.768 : deploy: 1
16:49:32.769 : current_position= X202.33 Y30.00 Z6.76 : >>> Probe::run_z_probe
16:49:32.769 : current_position= X202.33 Y30.00 Z6.76 : >>> Probe::probe_down_to_z
16:49:32.769 : >>> do_blocking_move_to X202.33 Y30.00 Z-0.27
16:49:32.769 : <<< do_blocking_move_to
16:49:35.651 : current_position= X202.33 Y30.00 Z1.70 : sync_plan_position
16:49:35.651 : current_position= X202.33 Y30.00 Z1.70 : <<< Probe::probe_down_to_z
16:49:35.652 : current_position= X202.33 Y30.00 Z1.70 : <<< run_z_probe
16:49:35.652 : >>> do_blocking_move_to X202.33 Y30.00 Z6.70
16:49:35.652 : <<< do_blocking_move_to
16:49:37.145 : <<< Probe::probe_at_point
16:49:37.151 : Probing mesh point 31/49.
16:49:37.151 : current_position= X202.33 Y30.00 Z6.70 : Probe::set_deployed
16:49:37.151 : deploy: 0
16:49:37.151 : >>> do_blocking_move_to X202.33 Y30.00 Z6.70
16:49:37.152 : <<< do_blocking_move_to
16:49:37.152 : >>> do_blocking_move_to X166.00 Y137.00 Z6.70
16:49:37.152 : <<< do_blocking_move_to
16:49:38.237 : X:166.00 Y:137.00 Z:6.70 E:0.00 Count X:13280 Y:10960 Z:2679
```

RuairiSpain, 2020-02-17:
Mesh inset is a constant distance from all 4 edges of the bed.  You've a probe offset of -49, -20 this will mean some mesh coordinates are outside the range of the mesh area (really needs an inset of at least 49 to allow for the left offset).  Instead of inset, I'd recommend changing these in Configuration_adv.h: 
#define MIN_PROBE_EDGE_LEFT 69
#define MIN_PROBE_EDGE_RIGHT 20
#define MIN_PROBE_EDGE_FRONT 40
#define MIN_PROBE_EDGE_BACK  20

This will give you a mesh area that takes account of the probe offset, I added a 20mm extra in case you have clips.  Change the mesh area to fit your needs.

Interplain, 2020-02-17:
I am experiencing a similar problem, do you think you could throw some light on where i am going wrong an how to fix it. Mine probes like 73 points and doesn't fill in the rest of the map with the below settings in PROBE_SELECTED && !IS_KINEMATIC disabled, by putting double slash in front. 

[image]

It probes inside the dark lines, outside i doesn't go.

Enabled without slashes and with the below script it just moves down and triggers and doesn't do anymore probing.

My Probe is -27 for  the X from the Nozzle and -40 for the Y. My bed is 220 x 220 mm square. 
I have put :

PROBE_SELECTED && !IS_KINEMATIC
  #define MIN_PROBE_EDGE_LEFT 10 //for edge of bed (*)
  #define MIN_PROBE_EDGE_RIGHT 237 //220 + 27 - 10 (*)
  #define MIN_PROBE_EDGE_FRONT 10 //for edge of bed (*)
  #define MIN_PROBE_EDGE_BACK 250 //220 + 40 - 10 (*)

With the above in Config_adv.h, am i doing it right, if i put the Values posted in the previous post obviously not for my bed size it probes more place.

#define MIN_PROBE_EDGE_LEFT 69
#define MIN_PROBE_EDGE_RIGHT 20
#define MIN_PROBE_EDGE_FRONT 40
#define MIN_PROBE_EDGE_BACK 20

Is this a Bug, because it works with PROBE_SELECTED && !IS_KINEMATIC settings enabled with  Bi-linear works with (*).

RuairiSpain, 2020-02-17:
Hi @Interplain,  my understanding is that the MIN_PROBE_EDGE_* are setting an inner margin from each of the borders edges, where the probe should not visit when calculating the bed mesh.  Each UBL mesh margin is relative to the L/R/F/B edge.   So, based on your x/y offsets, I'd try these settings:
#define MIN_PROBE_EDGE_LEFT 47
#define MIN_PROBE_EDGE_RIGHT 20
#define MIN_PROBE_EDGE_FRONT 60
#define MIN_PROBE_EDGE_BACK 20

Let us know if these coordinates work for you.  

FYI, I included an extra 20mm margin on each edge, in case you have clips holding the bed.

ManuelMcLure, 2020-02-17:
The points that are not being probed are not reachable by the probe (due to the probe offset) so there’s no way you can probe the whole bed. Instead, use “G29 P3” so that UBL fills the unprobed points with reasonable values and use G26 to fine tune them.

Interplain, 2020-02-18:
Thank you RuairiSpain, thank you for your post. but unfortunately it doesn't take it to the boundaries that it should.. With the settings you have given it still doesn't plot the whole bed.

[image]

the setting produce the points in the image.

Interplain, 2020-02-18:
> The points that are not being probed are not reachable by the probe (due to the probe offset) so there’s no way you can probe the whole bed. Instead, use “G29 P3” so that UBL fills the unprobed points with reasonable values and use G26 to fine tune them.

I Understand your logik, but it shouldnt work that way. What your implying is that the Nozzle should be doing the plotting of the bed not the Probe..

ManuelMcLure, 2020-02-18:
No. Your nozzle can reach X=210, which means that the probe can only reach X=183. So any points between X=184 and X=210 cannot be probed. If your nozzle can go past the right edge of the bed before hitting a physical limit you can try increasing X_MAX_POS to allow the probe to go further to the left.
Probing is always done according to the position of the probe and the measured value is used when the nozzle moves to that position when printing. For example, the center of the bed is measured with the probe at 105,105 and that measurement will be used when the nozzle is at 105,105. This is the way it has always worked.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle