MarlinFirmware/Marlin #16848: [FR] Advanced_Pause_Feature without LCD

stevereno30, 2020-02-12:
Since switching to Octoprint, I no longer need an LCD controller for operating my machine. However, when my filament runout sensor triggers the advance_pause_feature, there is no way to exit the wait loop without a physical button. For those that run their machines on Octoprint, PrintRun, or some other external host, it would be useful to be able to have a software-button to initiate filament reloading.

ManuelMcLure, 2020-02-12:
Enable ``HOST_ACTION_COMMANDS`` and ``HOST_PROMPT_SUPPORT`` in Marlin and enable the Action Command Support plugin in OctoPrint. That should display a dialog on OctoPrint to continue the print.

stevereno30, 2020-02-12:
It looks like this option does not exist in Marlin 1.1.x

I found it in Marlin 2, but I would rather not upgrade my entire machine
just to make the LCD unnecessary.

thisiskeithb, 2020-02-12:
> I found it in Marlin 2, but I would rather not upgrade my entire machine
just to make the LCD unnecessary.

Is your hardware incompatible with 2.0.x? You should only need to migrate your config since 2.0.x runs on the same hardware as previous versions.

stevereno30, 2020-02-12:
Ah. For some reason, I have been under the impression that Marlin 2.0.x was
32 bit only. Thanks for your help!

Roxy-3D, 2020-02-13:
> Ah. For some reason, I have been under the impression that Marlin 2.0.x was
32 bit only. Thanks for your help!

 No.   As it turns out...   There are a lot of nice things we can do for the 32-bit processors (like putting an RTOS into the mix) but we are waiting to do that until we have to fork the code.   We are trying to keep everything (8-bit and 32-bit) in the same branch as long as possible.

stevereno30, 2020-02-13:
So I know this is not intended to be a support thread, but with
host_action_commands and host_prompt_support enabled in Marlin 2.0.x, and
with the Action Command Prompt plugin enabled, I get prompts on Octoprint,
but they don't actually do anything on the printer. It will tell me via
Octoprint that the filament sensor is triggered, but when I press the
"purge more" button within Octoprint, Marlin ignores it. Am I missing
something?

thinkyhead, 2020-02-14:
> when I press the "purge more" button within Octoprint, Marlin ignores it

It should work, so we just need to check to make sure it's integrated properly.

thinkyhead, 2020-02-14:
@InsanityAutomation — Do both of these work for host integration, or does one need to be changed?
```cpp
host_action_prompt_button(PSTR("Purge More"));
//     ...or...
host_action_prompt_button(PSTR("PurgeMore"));
```

thisiskeithb, 2020-02-14:
With OctoPrint 1.3.12 & Action Command plugin, `HOST_ACTION_COMMANDS` and `HOST_PROMPT_SUPPORT` enabled in Marlin, filament change & "purge more" buttons work correctly with the latest bugfix-2.0.x (and did as well in previous bugfix versions).

stevereno30, 2020-02-14:
I just updated to the latest bugfix release and the action command plugin
only displays two buttons after loading filament: PurgeMore and
DisableRunout. There appears to be no way to resume a print from Octoprint
once advanced pause is triggered.

thinkyhead, 2020-02-14:
I figured these were probably button labels for display porpoises only.

thinkyhead, 2020-02-14:
> There appears to be no way to resume a print from Octoprint once advanced pause is triggered

It should show "Continue" on the next prompt if you "Purge More" and the filament sensor is detecting filament.

stevereno30, 2020-02-14:
So then I will need to purge more regardless of whether or not I need to in
order to resume my print? Is this a limitation of Marlin, Octoprint, or
Action Command Prompt?

stevereno30, 2020-02-14:
I just ran a filament runout test again, and on last night's bugfix 2.0.x,
Octoprint 1.3.12, and using Action Command Prompt, my machine serves
prompts to Octoprint, but none of them send commands that my machine does
anything with. The PurgeMore button does nothing, the "continue" that
appears when the machine says "Insert filament and press button" does
nothing, and the "Continue" that appears after Purging more with the
machine's button does nothing. What am I missing? Do I need to do something
else beyond uncommenting HOST_ACTION_COMMANDS and HOST_PROMPT_SUPPORT in
configuration_adv.h? Will the prompts not work as long as an LCD is
attached? Thanks for any thoughts you might have on the issue.

InsanityAutomation, 2020-02-14:
@thinkyhead yes, display string only as long as mode is set correctly response will be handled appropriately.

If resume is not shown, then the runout sensor is not currently made, Ensure material is in the sensor. If it is a jam sensor, there is a known caveat where its looking for a single state. I had planned to get into that this weekend, but got sick and get a bit behind. Probably next weekend thatll be resolved.

@stevereno30 if you want it to function as intended, please ensure emergency parser is enabled as well.

stevereno30, 2020-02-14:
@InsanityAutomation Thanks for the input, but it doesn't seem to make any difference. The Action Command Prompt buttons don't do anything that shows up in the Octoprint terminal, though with emergency_parser uncommented, I can manually send an m108 command. The problem comes after this is sent:
`Recv: echo:Insert filament and send M108
Recv: //action:prompt_end
Recv: //action:prompt_begin Nozzle Parked
Recv: //action:prompt_button Continue
Recv: //action:prompt_show
Recv: echo:busy: paused for user
Send: N52 M108*29
Recv: Error:Line Number is not Last Line Number+1, Last Line: 39
Recv: Resend: 40
Recv: ok
Recv: echo:busy: processing
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:45 B@:29
Recv: //action:prompt_end
Recv: //action:prompt_begin Paused
Recv: //action:prompt_button PurgeMore
Recv: //action:prompt_button DisableRunout
Recv: //action:prompt_show
Recv: //action:resumed
Printer signalled that it resumed, switching state...
Changing monitoring state from "Pausing" to "Resuming"
Send: N53 M876 S0*95
Recv: ok
Recv: ok
Recv: ok
Recv: //action:prompt_end
Recv: //action:prompt_begin FilamentRunout T0
Recv: //action:prompt_show
Recv: //action:paused filament_runout 0
Recv: //action:paused
Printer signalled that it paused, switching state...
Changing monitoring state from "Resuming" to "Pausing"
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:45 B@:29 W:?
Recv:  T:240.00 /240.00//action:prompt_end
Recv: //action:prompt_begin Paused
Recv: //action:prompt_button Purge More
Recv: //action:prompt_button DisableRunout
Recv: //action:prompt_show
Recv: M876 Responding PROMPT_FILAMENT_RUNOUT
Recv:  B:60.00 /60.00 @:45 B@:29
Recv: Error:Line Number is not Last Line Number+1, Last Line: 39
Recv: Resend: 40
Recv: ok
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:45 B@:29 W:19
Recv: echo:busy: processing`

This then continues in a cycle. However, if I have the LCD plugged in and enabled in Marlin, the on-screen commands work every time.

InsanityAutomation, 2020-02-14:
When I originally wrote it, it was tested headless. A few things have been refactored since so I'll get it fired up hopefully Sunday then. Day job has me tied up tomorrow.

stevereno30, 2020-02-14:
I think I must have had something messed up in my filament runout logic. It was triggering a runout no matter what. I went back to my non-bugfix Marlin with emergency_parser enabled and Action Command Prompt worked as expected. However, I noticed something else that seems odd... when the machine reaches the filament unload step, it loses communication with Octoprint temporarily and begins cooling the toolhead. 

> Recv:  T:240.00 /240.00 B:60.00 /60.00 @:17 B@:44
Recv: //action:notification Print Paused
Recv: echo:busy: processing
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Pausing" to "Paused"
Send: M105
Recv: echo:busy: processing
Recv:  T:236.20 /240.00 B:60.00 /60.00 @:61 B@:44
Recv: echo:busy: processing
Recv:  T:235.40 /240.00 B:60.01 /60.00 @:66 B@:40
Recv: echo:busy: processing
Recv:  T:234.60 /240.00 B:60.07 /60.00 @:72 B@:20
Recv: echo:busy: processing
Recv:  T:233.80 /240.00 B:60.00 /60.00 @:81 B@:41
Recv: echo:busy: processing
Recv:  T:233.80 /240.00 B:60.00 /60.00 @:79 B@:40
Recv: echo:busy: processing
Recv:  T:233.80 /240.00 B:60.00 /60.00 @:81 B@:39
Recv: echo:Insert filament and send M108
Recv: //action:prompt_end
Recv: //action:prompt_begin Nozzle Parked
Recv: //action:prompt_button Continue
Recv: //action:prompt_show
Recv: echo:busy: paused for user
Recv:  T:234.35 /240.00 B:60.00 /60.00 @:74 B@:39
Recv: echo:busy: paused for user
Recv:  T:235.00 /240.00 B:60.00 /60.00 @:70 B@:38
Send: M876 S0
Recv: M876 Responding PROMPT_FILAMENT_RUNOUT_CONTINUE
Recv:  T:235.40 /240.00 B:60.00 /60.00 @:69 B@:38 W:?
Recv: echo:busy: processing
Recv:  T:236.00 /240.00 B:60.00 /60.00 @:61 B@:38
Recv:  T:236.20 /240.00 B:60.00 /60.00 @:59 B@:38 W:?
Recv:  T:236.20 /240.00 B:60.00 /60.00 @:64 B@:38 W:?
Recv: echo:busy: processing
Recv:  T:236.95 /240.00 B:60.00 /60.00 @:54 B@:38
Recv:  T:237.00 /240.00 B:60.01 /60.00 @:55 B@:35 W:?
Recv:  T:237.56 /240.00 B:60.00 /60.00 @:50 B@:38 W:?
Recv: echo:busy: processing
Recv:  T:238.00 /240.00 B:60.01 /60.00 @:47 B@:35
Recv:  T:238.00 /240.00 B:60.00 /60.00 @:49 B@:38 W:?
Recv:  T:238.94 /240.00 B:60.00 /60.00 @:38 B@:38 W:?
Recv: echo:busy: processing
Recv:  T:239.00 /240.00 B:60.01 /60.00 @:41 B@:35
Recv:  T:239.00 /240.00 B:60.01 /60.00 @:43 B@:35 W:?
Recv:  T:239.38 /240.00 B:60.01 /60.00 @:41 B@:35 W:9
Recv: echo:busy: processing
Recv:  T:240.00 /240.00 B:60.01 /60.00 @:33 B@:35
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:35 B@:38 W:8
Recv:  T:240.00 /240.00 B:60.04 /60.00 @:39 B@:25 W:7
Recv: echo:busy: processing
Recv:  T:240.19 /240.00 B:60.02 /60.00 @:38 B@:31
Recv:  T:240.69 /240.00 B:60.01 /60.00 @:31 B@:34 W:6
Recv:  T:241.00 /240.00 B:60.00 /60.00 @:30 B@:37 W:5
Recv: echo:busy: processing
Recv:  T:241.00 /240.00 B:60.01 /60.00 @:32 B@:34
Recv:  T:241.00 /240.00 B:60.00 /60.00 @:33 B@:37 W:4
Recv:  T:241.00 /240.00 B:60.00 /60.00 @:36 B@:36 W:3
Recv: echo:busy: processing
Recv:  T:241.00 /240.00 B:60.00 /60.00 @:37 B@:36
Recv:  T:241.00 /240.00 B:60.00 /60.00 @:37 B@:36 W:2
Recv:  T:241.00 /240.00 B:60.01 /60.00 @:38 B@:33 W:1
Recv: echo:busy: processing
Recv:  T:241.00 /240.00 B:60.00 /60.00 @:39 B@:36
Recv:  T:241.00 /240.00 B:60.00 /60.00 @:39 B@:36 W:0
Recv: //action:notification Print Paused
Recv: echo:busy: processing
Recv:  T:241.00 /240.00 B:60.00 /60.00 @:39 B@:36
[... 44 lines cut ...]
Recv: //action:prompt_button Dismiss
Recv: //action:prompt_show
Recv: //action:notification Printing...
Recv: ok
Recv:  T:240.00 /240.00 B:60.01 /60.00 @:46 B@:29 W:?
Recv:  T:239.88 /240.00 B:60.00 /60.00 @:48 B@:33 W:19
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:46 B@:33
Recv: echo:busy: processing
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:46 B@:33 W:17
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:46 B@:33 W:16
Recv:  T:239.94 /240.00 B:60.00 /60.00 @:47 B@:32
Recv: echo:busy: processing
Recv:  T:240.00 /240.00 B:60.01 /60.00 @:46 B@:29 W:15
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:46 B@:32 W:14
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:46 B@:32
Recv: echo:busy: processing
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:46 B@:32 W:13
Recv:  T:240.00 /240.00 B:60.04 /60.00 @:46 B@:19 W:12
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:46 B@:32
Recv: echo:busy: processing
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:46 B@:32 W:11
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:46 B@:32 W:10
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:46 B@:32
Recv: echo:busy: processing
Recv:  T:240.00 /240.00 B:60.00 /60.00 @:46 B@:32 W:9

thinkyhead, 2020-02-15:
> ```
> Recv: echo:Insert filament and send M108
> ```

It should start to output "`echo:busy: paused for user`" very soon after the line above is output. Can you point out where in the log output the long pause occurs?

It is normal for Marlin to start cooling down if the printer is paused for a while. It begins to heat up again once you insert the filament into the extruder and click continue.

stevereno30, 2020-02-17:
It is actually not operating at all how I expect it to...
When runout is triggered, I typically don't get to the machine before it times out. So when I start working with it, I need to select the reheat prompt. Once it reheats, it sends a "Reheat Done" continue button. If I press this button, it will load the filament and **continue the print.** It will still show the "PurgeMore" and "Continue" options after loading, but the machine will already be printing, and those buttons will do nothing (see the attached screenshot to see the prompt while the terminal shows print moves in the background). If you look at the terminal output below, you will see the point at which the temperature drops after prompting filament insert (which is never displayed as an Action Prompt. It only appears in the terminal). If you look after the temperature rises back up, there is an action prompt menu that is pushed to Octoprint, but there is an action:resume that was never sent from Octoprint. I also attached the log to show where the temperature starts to drop. I labeled it with a double asterisk.  

> Recv: echo:busy: paused for userRecv:  T:241.94 /240.00 B:60.00 /60.00 @:0 B@:33Recv:  T:241.06 /240.00 B:60.00 /60.00 @:1 B@:33 W:1Recv:  T:241.00 /240.00 B:60.00 /60.00 @:1 B@:33 W:0Recv: echo:busy: paused for userRecv:  T:241.00 /240.00 B:60.00 /60.00 @:1 B@:33Recv: echo:Insert filament and send M108Recv: //action:prompt_endRecv: //action:prompt_begin Reheat DoneRecv: //action:prompt_button ContinueRecv: //action:prompt_showRecv: echo:busy: paused for userRecv:  T:239.25 /240.00 B:60.00 /60.00 @:27 B@:33Send: M876 S0Recv: M876 Responding PROMPT_FILAMENT_RUNOUT_CONTINUERecv:  T:239.00 /240.00 B:60.00 /60.00 @:29 B@:33 W:?Recv:  T:238.00 /240.00 B:60.00 /60.00 @:42 B@:33 W:?Recv: echo:busy: processingRecv:  T:238.00 /240.00 B:60.00 /60.00 @:40 B@:33Recv:  T:237.69 /240.00 B:60.00 /60.00 @:43 B@:33 W:?Recv:  T:237.00 /240.00 B:60.00 /60.00 @:50 B@:33 W:?Recv: echo:busy: processingRecv:  T:236.85 /240.00 B:60.00 /60.00 @:51 B@:33Recv:  T:236.20 /240.00 B:60.00 /60.00 @:59 B@:33 W:?Recv:  T:235.55 /240.00 B:60.00 /60.00 @:66 B@:33 W:?Recv: echo:busy: processingRecv:  T:235.40 /240.00 B:60.00 /60.00 @:67 B@:33Recv:  T:235.40 /240.00 B:60.00 /60.00 @:64 B@:33 W:?Recv:  T:234.60 /240.00 B:60.00 /60.00 @:74 B@:33 W:?Recv: echo:busy: processingRecv:  T:234.60 /240.00 B:60.00 /60.00 @:73 B@:33Recv:  T:234.45 /240.00 B:60.00 /60.00 @:73 B@:33 W:?Recv:  T:233.80 /240.00 B:60.00 /60.00 @:82 B@:33 W:?Recv: echo:busy: processingRecv:  T:233.80 /240.00 B:60.00 /60.00 @:80 B@:33Recv:  T:233.80 /240.00 B:60.00 /60.00 @:79 B@:33 W:?Recv:  T:233.80 /240.00 B:60.00 /60.00 @:78 B@:33 W:?Recv: echo:busy: processingRecv:  T:233.80 /240.00 B:60.00 /60.00 @:78 B@:33Recv:  T:233.80 /240.00 B:60.00 /60.00 @:78 B@:33 W:?Recv:  T:233.80 /240.00 B:60.00 /60.00 @:79 B@:33 W:?Recv: echo:busy: processingRecv:  T:233.80 /240.00 B:60.00 /60.00 @:79 B@:33Recv:  T:233.80 /240.00 B:60.01 /60.00 @:80 B@:30 W:?Recv:  T:233.80 /240.00 B:60.00 /60.00 @:81 B@:33 W:?Recv: echo:busy: processingRecv:  T:233.80 /240.00 B:60.00 /60.00 @:82 B@:33Recv:  T:234.30 /240.00 B:60.00 /60.00 @:74 B@:33 W:?Recv:  T:234.60 /240.00 B:60.00 /60.00 @:73 B@:33 W:?Recv: echo:busy: processingRecv:  T:234.60 /240.00 B:60.01 /60.00 @:74 B@:29Recv:  T:234.85 /240.00 B:60.02 /60.00 @:71 B@:26 W:?Recv:  T:235.40 /240.00 B:60.01 /60.00 @:66 B@:29 W:?Recv: echo:busy: processingRecv:  T:235.40 /240.00 B:60.00 /60.00 @:68 B@:33Recv:  T:236.10 /240.00 B:60.00 /60.00 @:59 B@:33 W:?Recv:  T:236.20 /240.00 B:60.00 /60.00 @:62 B@:32 W:?Recv: echo:busy: processingRecv:  T:236.25 /240.00 B:60.01 /60.00 @:63 B@:29Recv:  T:237.00 /240.00 B:60.01 /60.00 @:53 B@:29 W:?Recv:  T:237.00 /240.00 B:60.01 /60.00 @:58 B@:29 W:?Recv: echo:busy: processingRecv:  T:237.50 /240.00 B:60.00 /60.00 @:51 B@:32Recv:  T:238.00 /240.00 B:60.00 /60.00 @:46 B@:32 W:?Recv:  T:238.19 /240.00 B:60.00 /60.00 @:48 B@:32 W:?Recv: echo:busy: processingRecv:  T:238.75 /240.00 B:60.00 /60.00 @:40 B@:32Recv:  T:239.00 /240.00 B:60.00 /60.00 @:40 B@:31 W:?Recv:  T:239.00 /240.00 B:60.00 /60.00 @:45 B@:31 W:?Recv: echo:busy: processingRecv:  T:239.06 /240.00 B:60.00 /60.00 @:45 B@:31Recv:  T:240.00 /240.00 B:60.00 /60.00 @:32 B@:31 W:9Recv:  T:240.00 /240.00 B:60.00 /60.00 @:37 B@:31 W:8Recv: echo:busy: processingRecv:  T:240.00 /240.00 B:60.01 /60.00 @:39 B@:28Recv:  T:240.19 /240.00 B:60.00 /60.00 @:38 B@:31 W:7Recv:  T:240.94 /240.00 B:60.00 /60.00 @:29 B@:31 W:6Recv: echo:busy: processingRecv:  T:241.00 /240.00 B:60.00 /60.00 @:29 B@:31Recv:  T:241.00 /240.00 B:60.00 /60.00 @:31 B@:31 W:5Recv:  T:241.00 /240.00 B:60.00 /60.00 @:34 B@:31 W:4Recv: echo:busy: processingRecv:  T:241.00 /240.00 B:60.00 /60.00 @:35 B@:31Recv:  T:241.00 /240.00 B:60.00 /60.00 @:36 B@:31 W:3Recv:  T:241.00 /240.00 B:60.00 /60.00 @:37 B@:31 W:2Recv: echo:busy: processingRecv:  T:241.00 /240.00 B:60.00 /60.00 @:37 B@:31Recv:  T:241.00 /240.00 B:60.00 /60.00 @:38 B@:31 W:1Recv:  T:241.00 /240.00 B:60.00 /60.00 @:38 B@:31 W:0Recv: echo:busy: processingRecv:  T:241.00 /240.00 B:60.00 /60.00 @:39 B@:31Recv: echo:busy: processingRecv:  T:241.00 /240.00 B:60.00 /60.00 @:39 B@:31Recv: echo:busy: processingRecv:  T:240.94 /240.00 B:60.00 /60.00 @:40 B@:31Recv: echo:busy: processingRecv:  T:240.00 /240.00 B:60.00 /60.00 @:54 B@:31Recv: echo:busy: processingRecv:  T:240.00 /240.00 B:60.00 /60.00 @:50 B@:31Recv: echo:busy: processingRecv:  T:240.00 /240.00 B:60.00 /60. [... cut ...] ecv: ok T:240.00 /240.00 B:60.00 /60.00 @:46 B@:31Send: N117 G1 X103.252 Y106.331 E4.59472*109Recv: okSend: N118 G1 X102.718 Y105.105 E4.70539*107Recv: okRecv: okSend: N119 G1 X102.345 Y104.055 E4.79760*100Send: N120 G1 X102.094 Y103.182 E4.87277*111Recv: okSend: N121 G1 X101.795 Y101.878 E4.98348*102Recv: okSend: N122 G1 X101.623 Y100.778 E5.07559*102Recv: okSend: N123 G1 X101.519 Y99.670 E5.16767*90

 [image]
[octoprint.log](https://github.com/MarlinFirmware/Marlin/files/4211880/octoprint.log)

stevereno30, 2020-03-11:
Not solved, but opening a bug report instead since my issue is one of the feature not working, not of a feature not existing.

closed 2020-03-11 by stevereno30