MarlinFirmware/Marlin #16853: SKR PRO V1.1 with TMC2209

fiveofds, 2020-02-13:
I have an SKR PRO V1.1 with TMC2209s using Configurations from Bigtreetec I can get sensorless homing working IF it is configured as a cartesian printer but then configuring for my delta printer:

// Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR 1 //MPD
#define Y_HOME_DIR 1 //MPD
#define Z_HOME_DIR 1 //MPD

When this is switched from -1 to 1 sensorless homing stops working
Any advice?

boelle, 2020-06-20:
@fiveofds do you still have issues?

fiveofds, 2020-06-21:
Have not tried in a while now, but I will find the time and see what happens.MikeSent from my T-Mobile 4G LTE Device

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/MarlinFirmware/Marlin/issues/16853#issuecomment-647044549",
"url": "https://github.com/MarlinFirmware/Marlin/issues/16853#issuecomment-647044549",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle