MarlinFirmware/Marlin #16854: Adjusting EEPROM z home offset + makes nozzle move down, firmware opposite

DroneMang, 2020-02-13:
Hi,

In firmware, if the probe position

#define NOZZLE_TO_PROBE_OFFSET { 0, -36, -0.56 }

is made more negative the nozzle will move down, the first layer will be thinner, etc. This makes sense.

But if I edit the EEPROM value, I must make the z probe offset more positive to get the first layer thinner, get the nozzle to move down, etc.

To me it seems the sign of the adjustment (+-) within EEPROM should have the same effect as within firmware.  So if I have -0.56 set in firmware and I get a first layer thickness of 0.25 and I want it to be 0.20, I should be able to adjust/set the z probe offset to -0.05 in EEPROM, or make it that much smaller within firmware and recompile, to get the nozzle that much closer to the bed.  

I have verified this behavior with a micrometer and measured the first layer printed. 

Perhaps there is a convention or documentation note that explains this that I have not seen.  

Thank you!

thinkyhead, 2020-02-15:
There are a lot of POVs to take on the effects of the probe Z offset. But, the point is to make it accurate so you no longer have to worry about it. I'm not sure how better to describe it than to simply explain how to calibrate it…

The Z position should be 0 when the tip of the nozzle is just touching the bed.

So, first you do a `G28` homing using the probe, then use the LCD or host to move the nozzle down slowly until it just touches the bed.

- If the Z position is positive when the nozzle is placed on the bed, then the probe Z offset is too far in the negative direction. Adjust the probe Z offset towards zero (to move Z=0 up to the level of the bed).
- If the Z position is negative when the nozzle is placed on the bed, then the probe Z offset is too far in the positive direction. Adjust the probe Z offset away from zero (to move Z=0 closer to the level of the bed).

In both cases you simply add the Z value shown on the screen to your Z probe offset to correct it.

So: if you have a probe Z offset of -2.0 the probe should trigger when the nozzle is at 2mm from the bed. But if you do a G28 and then put the nozzle on the bed and Z is 1.0, just add that 1.0 to your probe Z offset and it will be perfect.

In the near future these steps will be done as part of the "setup" menu so we don't have to remember which way is up or down.

thinkyhead, 2020-02-15:
NOTE that the z home offset is not the same thing at all as the Z probe offset. If this is at all problematic, you should enable `NO_WORKSPACE_OFFSETS` to remove all home offsets and workspaces from the firmware.

closed 2020-02-15 by thinkyhead

DroneMang, 2020-02-15:
Interesting way to calibrate it for setting up. I will pick that up for the
future that is a good method.

So what do you do when you are really fine-tuning your first layer height?
Are you going into your slicer and adjusting it there or are you uploading
a new firmware every time? Ideally in the slicer I would set the first
layer height at 0.2 if that is the layers I am printing with for the model
and do the tweaks in firmware. For example I got my first layer height at
0.25 measured with a micrometer but I wanted to be 0.2. So how would you go
about adjusting things to get it to that thickness?  What is the intended
use of the home offset options in EEPROM? Thanks.

thinkyhead, 2020-02-16:
> So what do you do when you are really fine-tuning your first layer height?

`BABYSTEP_ZPROBE_OFFSET` followed by `M500`.

Since the bed expands as it heats, it's important to let the printer sit for a period of time after heating up before attempting any calibration. You'll get the most consistent results if you wait for heating before doing any `G28` or `G29` that involves a probe.