MarlinFirmware/Marlin #16855: [BUG] Sensorless Homing seems broken for TMC2130 SPI

ProfEngr, 2020-02-14:
MKS Gen L board w/ TMC2130 drivers in SPI w/ diag pins of X and Y wired to the respective min endstops. Had to enable software SPI to keep the single ribbon to the Ender 3 LCD. I've adjusted M914 for X and Y to the point that +/-1 step will cause the axis to either not move or ram the home position requiring a power off. The Z axis is working as expected w/ a BLTouch v2.x. Stepper motions when moved via terminal are correct.

When I tried compiling the 2.0.3 release a week ago it would crash both X and Y when issued G28. When issued G28 X or G28 Y the steppers would just 'twitch' and the coordinate would be set to 0. No value of M914 would change this behavior; with the 2.0.3-bugfix at least M914 has an effect. G28 Z behaves as expected.

Everything works fine on 1.1.9-bugfix from mid last year (not sure the exact build date). 

I had the 1.1.9 files open beside the 2.0.3 files so I could change the setting values to match. Everything is being compiled with VSCode. These are the files I modified in both 1.1.9 and 2.0.3 bugfixes; 2.0.3 versions.

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4202114/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4202115/Configuration.txt)
[pins_MKS_GEN_L.txt](https://github.com/MarlinFirmware/Marlin/files/4202138/pins_MKS_GEN_L.txt)
[pins_RAMPS.txt](https://github.com/MarlinFirmware/Marlin/files/4202139/pins_RAMPS.txt)

Balzac40, 2020-02-14:
I use sensorless homing, but no STEALTHCHOP, X_STALL_SENSITIVITY=8 and also #define X_CURRENT_HOME at value equal to X_CURRENT / 2 (half).

ProfEngr, 2020-02-14:
> 
> 
> I use sensorless homing, but no `STEALTHCHOP`, `X_STALL_SENSITIVITY` = 8 and also `X_CURRENT_HOME` at value equal to `X_CURRENT` / 2 (half).

I'll see if that helps, but I thought I saw a comment in the config saying homing always uses spreadcycle.

Balzac40, 2020-02-14:
I told you as you have allowed stealthchop in your config files!

ProfEngr, 2020-02-14:
So this paragraph in configuration_adv.h doesn't matter? Next time I'll try disabling it anyway since the notes say SPI sensing is "beta."

   * Use StallGuard2 to home / probe X, Y, Z.
   *
   * TMC2130, TMC2160, TMC2209, TMC2660, TMC5130, and TMC5160 only
   * Connect the stepper driver's DIAG1 pin to the X/Y endstop pin.
   * X, Y, and Z homing will always be done in spreadCycle mode.

thinkyhead, 2020-02-15:
You're correct that Marlin now sets TMC drivers to spreadCycle during homing.

It can never hurt to issue `M502`, `M500` and do a reboot to make sure your settings are fresh.

Beyond that, AFAIK there are no super serious issues with TMC drivers at this time, but we're keeping an eye on how to best troubleshoot them when things like this crop up.

fab4fun, 2020-02-17:
I can confirm the issue (also TMC2130). 
If I set:
`#define X_STALL_SENSITIVITY  3`
it will detect the end, but rather harshly.  Setting to 2 just bounces off the end stops endlessly.  Setting to 4 won't let it even get moving. 

default I'm using is 
```
#if AXIS_IS_TMC(X)
   #define X_CURRENT       800        // (mA) RMS current. Multiply by 1.414 for peak current.
   #define X_CURRENT_HOME  X_CURRENT  // (mA) RMS current for sensorless homing
```
I will try X_CURRENT/2 when I get home

seandop, 2020-02-20:
I can also confirm the issue (TMC2130 and sensorless homing for X-Y, as well) but not using SPI. Sometimes (but not always) when calling G28 X, G28 Y, or simply G28, the drivers will "twitch" for the X and/or Y motor and the coordinate will be set to 0, regardless of where the nozzle actually is on the buildplate. Other times, homing works as expected. Z-axis homing (which is not sensorless -- I use a BLtouch) works fine. I'd never seen this strange behavior on the 1.x builds, and I didn't have it on the early 2.0-bugfix builds (around 5/2019). However, it did start appearing after I upgraded from a 5/2019 build to a newer (2/2020) 2.0-bugfix a few weeks ago.

Interestingly, I've found that if the "twitch" does occur, calling G28 X, G28 Y, or G28 a second time always results in correct homing behavior. This problem is so random, that I've actually modified my start G code in my slicer to manually home X twice, followed by Y twice, and then Z always works just fine (not setup for sensorless homing). The firmware really seems to be working well, so I'm content to use this as a bandaid fix for now...

Camaro1, 2020-03-12:
I'm having the same issue, as well as some others here: #16420

amaschas, 2020-03-31:
I think I'm having a similar problem. I have stallguard activated, but the motors never actually stall. I connected a meter to the diag pins and the voltage never changes, which indicates to me that stallguard isn't triggering on a hardware level.

LuSEllDen, 2020-05-28:
Hello. I have exact the same problem with latest build of Marlin 2.0.5.3. It works great on 1.1.9. Now I must be always ready to press reset button. I tried many settings with current and bump sensitivity but this behavior seems very random :-(

boelle, 2020-06-21:
@ProfEngr still an issue?

LuSEllDen, 2020-06-21:
Yes, it is still an issue. I don't find a working solution yet. Printer still damaging a belt due to this issue. Sensorless homing works successfully only about 25% of attempts.

thinkyhead, 2020-06-21:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-06-21 by thinkyhead

ProfEngr, 2020-06-21:
I'm haven't been running Marlin on my printer since March.

> @ProfEngr still an issue?

I can't easily say at this point. I haven't had Marlin on my printer since March.

fab4fun, 2020-06-21:
I suspect the issue lies somewhere in the translation of the 'sensitivity' calibration to a driver current threshold in the trinamic drivers.  It seems to be off by an order of magnitude, such that a single integer value change is either too sensitive and false triggers all the time, or not sensitive enough and will only occasionally trigger (and bang against the end).  The 'real' robust value is somewhere in-between.

seandop, 2020-06-21:
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this [...]

At least four of us have confined the issue. Kinda seems like a slap in the face to just close it out and tell us to seek support for "questions" elsewhere...

amaschas, 2020-06-21:
I can confirm this is still happening. I went several rounds with people in the Discord trying to solve it and no dice. I've since reverted to mechanical end stops.

macasero, 2020-07-11:
I hava had the same problem this evening on a friends CoreXY printer.

I figured it out the homing with at lower current solved it, so I left  it like this (X a Y with the same value since it´s a coreXY and uses both steppers to move either X and Y)

`#define X_CURRENT       1000
#define X_CURRENT_HOME  X_CURRENT/2 `

seandop, 2020-07-11:
@macasero Interesting. I've never needed to adjust that before... any idea why that would fix it?

C4ptAwesome, 2020-08-24:
@macasero this also fixed it for my. I'm also using a CoreXY. But I'm only able to M914 without decimals like .5, also now G28 X and G28 Y home correct sensorless, but if I G28 after homing the print area is reduced to a fraction of the space. SKR 1.4 Turbo, TMC2130, 2.0.x_bugfix.