MarlinFirmware/Marlin #16858: [question][samd51] TMC with serial, min baud

GMagician, 2020-02-14:
I'm going to implement SoftwareSerial for samd51 boards.
To save more timers as possible (I may need them for H/W pwm) I'm going to move temperature to rtc timer (max 32KHz).

I'm going to investigate if such timer may also be used for software uart but due to its lower frequency I may be able to get 9600 baud max (with 3, and not more, samples)

better to move to other solution? Or may I pursue this path?

GMagician, 2020-02-14:
My fault...RTC clock source is not 32KHz, may be choosen

GMagician, 2020-02-15:
Yeah on datasheet is reported multi clock source but this means 32K or 1K, so question is still valid

thinkyhead, 2020-02-15:
If the Temperature ISR is on a SAMD51 timer that is inconvenient for Software Serial, then that should definitely be rectified first. I don't think there are a huge number of SAMD51 users right now, so this is a good time to …

### "Take it apart!"
[image]

GMagician, 2020-02-15:
Well, that's not what I meant to say...forgive me,  I try to write differently.

Now temperature isr is on a timer and that's good. Only problem is timer may also be used by H/W pwm. Less free timer I have, less H/W pwm may be used.

What I'm doing now is moving temperature isr from a timer to RTC (so I free up 2 timers)
RTC has  32.768KHz frequency, so even if I can't get a precise 1ms scheduling I think will be not an issue.

SoftwareSerial is not implemented in samd51 core so what I'm going to do is....implement it for Marlin.
With softser I need anorther timer so I was wondering if not to use the same RTC for it but since its limited frequency I can't raise up baud.

So here is my question:
How important is baud for tmc communication? may be in the future a limit 9600?

thinkyhead, 2020-02-15:
> How important is baud for tmc communication? may be in the future a limit 9600?

Not super important. Lower speeds should be fine for STEP/DIR/EN setups that just use UART for config and diagnostics.

closed 2020-02-15 by GMagician

sjasonsmith, 2020-02-15:
I believe 9600 is the minimum specified by TMC. If you are modeling SoftwareSerial after the version used for LPC1768/STM32, lower speeds are better for the system. All those boards currently default to 19200 baud in Marlin.

If you are modeling it after the AVR version, then low baud rates are very bad. The AVR version has to disable interrupts for the duration of each byte, which would be about one millisecond at 9600 baud.

GMagician, 2020-02-15:
I'm basing my work on lpc version, but such version seem wrong in some points:

1) it doesn't manage inversion correctly, only pullup/down but doesn't reverse read/written bits
2) it starts sending bits too early (changing tx_bit_cnt even before setting speed)

my 9600 limit is due to, if using RTC timer, its 32KHz max. With 3 samples per bit no more than 9600 can be achieved. To be faster I have to use another timer.