MarlinFirmware/Marlin #16860: [BUG] BOARD_MKS_BASE_16 + FILAMENT_RUNOUT_SENSOR won't compile

yet-another-average-joe, 2020-02-14:
### Bug Description

Marlin 2.0.x bugfix 2.0.4 / 71539bc

BOARD_MKS_BASE_16 + FILAMENT_RUNOUT_SENSOR won't compile : 

Gives this error :

Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:695:6: error: #error "FILAMENT_RUNOUT_SENSOR requires FIL_RUNOUT_PIN."
     #error "FILAMENT_RUNOUT_SENSOR requires FIL_RUNOUT_PIN."

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4205955/Marlin.zip)

### Steps to Reproduce

define MOTHERBOARD = BOARD_MKS_BASE_16

**Expected behavior:** [What you expect to happen]

the preprocessor using the default pin for filament runout sensor (digital 4)

**Actual behavior:** [What actually happens]

the preprocessor throws an error

#### Additional Information

workaround : define MOTHERBOARD = BOARD_MKS_BASE_15

(1.5 always worked fine for MKS Base 1.6)

OR

in pins_MKS_BASE_16.h, commenting out :

```
//#ifndef FIL_RUNOUT_PIN
//  #define FIL_RUNOUT_PIN   -1
//#endif

```

It seems FIL_RUNOUT_PIN  is redefined as -1 somewhere, and upsets sanity check

Can't figure where's the bug : 

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-02-15:
> `#error "FILAMENT_RUNOUT_SENSOR requires FIL_RUNOUT_PIN."`

This behavior was changed in #16806 by @DanielMazurkiewicz & @thinkyhead. There isn't a `FIL_RUNOUT_PIN` defined in your configs, nor the `pins_MKS_BASE_16.h` file, so it's disabled.

The MKS Base 1.5 pins file doesn't undefine the pin automatically, so it's set to 4 (and allows for a successful compile):
https://github.com/MarlinFirmware/Marlin/blob/71539bcbeb68284663d0627207ffd5a9a377e920/Marlin/src/pins/ramps/pins_RAMPS.h#L252-L255

Maybe @DanielMazurkiewicz can explain why the change was needed on the 1.6 pins file & not the 1.5/other BASE boards.

yet-another-average-joe, 2020-02-15:
Thanks for the explanations. But I don't understand the reasons why 1.6 does not" behave" as other RAMPS...

A while ago, I had FIL_RUNOUT_PIN defined in Configuration.h. At this time, I did use Z+ for filament runout sensor. I then stopped doing this, decided to use the default pin D4, and removed the FIL_RUNOUT_PIN definition from Configuration.h : Marlin and me were happy and I completely forgot about it !

Maybe, in Configuration.h comment, it could be good to add :

`For MKS Base 1.6 boards you MUST define FIL_RUNOUT_PIN, FIL_RUNOUT2_PIN, etc.`

thinkyhead, 2020-02-15:
> Maybe @DanielMazurkiewicz can explain why the change was needed on the 1.6 pins file & not the 1.5/other BASE boards.

Here is the original proposed change: d9ce0f12ffeeb1896dec4e9e2003ce791a0775dd

DanielMazurkiewicz, 2020-02-17:
@thisiskeithb @yet-another-average-joe 

> Maybe @DanielMazurkiewicz can explain why the change was needed on the 1.6 pins file & not the 1.5/other BASE boards.

Sorry for late answer, but the answer is simple - the board does not define pin FILAMENT_RUNOUT_SENSOR #16783 . This pin has to be remapped from one of other pins defined on board. Aldo wasn't fully aware of issues with Marlin code refactoring when made that PR. But if there were no issues with that I would really keep clean board pin definitions in one file and any mods/redefinitions in separate file in main directory, similar to configuration.h. That would keep code clean easier to understand and modify and would beautifully fall into HAL idea.

closed 2020-06-20 by boelle