MarlinFirmware/Marlin #16861: [BUG] (short description)

Vladimir-VS, 2020-02-14:
### Bug Description

Enabling ENDSTOPS_ALWAYS_ON_DEFAULT prevents printing when nozzle is above the  z-min endstop (e.g. when inductive probe or optical endstop is used)
### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4205900/Configuration.zip)

### Steps to Reproduce

1. enable (or override with M120)
#define ENDSTOPS_ALWAYS_ON_DEFAULT
2. Auto home the  printer
3. Move the nozzle down  -- the printer moves until triggering endstop and stops. Moving further down is impossible regardless any other settings.

**Expected behavior:** [What you expect to happen]
When ENDSTOPS_ALWAYS_ON_DEFAULT is enabled, the printer should stop on any endstop trigger but
If z move is explicitly allowed below the endstop (e.g. by mesh leveling, z-probe offset or manual z home offset) the z-triggering should not (at least immediately) stop the printer.

**Actual behavior:** [What actually happens]
When ENDSTOPS_ALWAYS_ON_DEFAULT is enabled, the printer stops on any endstop trigger including z-min regardless any other settings.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thinkyhead, 2020-02-15:
I assume you are actually referring to `SD_ABORT_ON_ENDSTOP_HIT` … ? Have you found this feature useful? I always keep it turned off. Sounds like it should ignore the Z endstop generally.

Vladimir-VS, 2020-02-15:
Not really. I do mean the ENDSTOPS_ALWAYS_ON_DEFAULT. When enabled, the printer stops on z-min endstop. Thus, if the nozzle is higher than z-min, the printing is not possible as the nozzle can not lower to the bed.

thinkyhead, 2020-02-15:
Oh, I see. Well, the printer is doing what it is supposed to do when endstops are enabled, which is to stop the axes on an endstop hit, and outside of homing with G28 or probing with G29 this behavior will simply throw off your axes. It is not in any way useful as a safety feature by itself, and is really only useful with `SD_ABORT_ON_ENDSTOP_HIT`. So I would recommend you simply disable `ENDSTOPS_ALWAYS_ON_DEFAULT` since it is getting in your way and it doesn't sound like it's something you want or need.

Vladimir-VS, 2020-02-15:
thanks, thinkyhead.
Actually, I do want to use this feature as a safety -- when not homed the feature should stop the printer going beyond the hardware limits.
Besides, I must set the software  limits outside the physical ones and I do prefer to have the preventing feature enabled.
It is unfortunate you do see it as a "as intended" behavior -- it is quite an outdated thinking that the endstop always indicates the physical limit of the movement. By setting the offset from the endstop (e.g. via leveling) I explicitly allow the movement outside the z-min. Enabling the enstops in the absolute logic simply makes no sense anymore.

Practically speaking, is there a way (via code adjustment?) to switch-off the enstops effect of z-min only?

tablatronix, 2020-04-23:
This could be split into distinct checks for different axis. But would require some refactoring since this is a global flag it seems. I would love crash protection in x and y but allow z for mesh leveling.probing

```C++
  // Enable endstops
  enable_globally(
    #if ENABLED(ENDSTOPS_ALWAYS_ON_DEFAULT)
      true
    #else
      false
    #endif
  );
```

AnHardt, 2020-04-23:
For this reason a probe is not an endstop and has to be deployed.

tablatronix, 2020-04-23:
Yeah but manual mesh leveling uses ends top as a probe doesn’t it via end stop probe pin or whatever

AnHardt, 2020-04-24:
`#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` used to imply "There is no Z-min-endstop - a probe has replaced it."
Since there was recently worked on that may be broken.

tablatronix, 2020-04-24:
I think its a work around or kludge to make manual mesh leveling work, fakes a probe etc.

Vladimir-VS, 2020-05-06:
AnHardt,
I am not sure I understand the  suggested work-around.
Could you, please, describe your option in some more details?

Thank you,
Vladimir

Vladimir-VS, 2020-05-11:
> If used together with ENDSTOPS_ALWAYS_ON_DEFAULT a z-min-endstop has to trigger below first-layer-height. The probe then has to trigger higher than that.

AnHardt,
this approach is only feasible if the z-homing point has the lowest height of the bed. This is practically not possible.

closed 2020-05-11 by Vladimir-VS