MarlinFirmware/Marlin #16862: [BUG] Problems with filament runout detection after manually swapping filaments at start of print

MohanMaker, 2020-02-14:
### Bug Description

I am experiencing a bug with filament runout detection when switching out filaments during heating, homing, and auto-leveling before a print begins. The printer remembers that the filament sensor has been open (no filament) at some point during heating/homing/leveling and then triggers a filament runout error (when it actually starts printing/extruding) even if I've re-inserted filament.

### My Configurations

I am running Marlin 2.0.3 with an SKR E3 DIP on Ender 3 Pro. My filament runout sensor is a re-used z-endstop.
[Marlin 2.0.3 Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4207210/Marlin.2.0.3.Configurations.zip)

### Steps to Reproduce

1. Switch on the printer and begin a print from the SD card. (Some filament is already loaded into the printer through the runout sensor).
2. While the printer is heating up and homing/auto-leveling, manually pull out the filament currently in the printer (so the filament runout sensor reads "open"). Note: this will not trigger a filament runout error.
3. Put in the new filament through the filament runout sensor before the print begins (printer is still homing/auto-leveling/heating)
4. Immediately after the printer starts doing it first extrusions/movements at the start of the print, the printer "remembers" that the filament sensor was previously open (when I pulled out the filament in step 2), and a filament runout error is triggered, even though there is filament currently the machine.
5. To get the printer to keep printing, I either have to follow the steps in the filament runout process (it retracts the filament and then has me reload) or I have to reboot the printer.

**Expected behavior:**
The printer should not trigger a filament runout error when it has filament in it at the start of the print (even if I've swapped out filament during the heating/homing/leveling process).

**Actual behavior:**
At the start of a print, the filament runout error is triggered even though filament has been reloaded into the machine.

#### Additional Information

Video demonstrating the problem: [https://youtu.be/2gwhPKvpyD8](url)

thinkyhead, 2020-02-15:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

MohanMaker, 2020-02-15:
Just tested the bugfix-2.0.x branch from 2/15/20 using the exact same steps outlined above. The problems still persists, no major differences from the 2.0.3 release I was using. Used the same configuration settings as 2.0.3 attached above, just transferred the settings into the new bugfix configuration files. I've attached the new files anyways in case they're useful. Thanks!

[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/4208712/Marlin-bugfix-2.0.x.zip)

thinkyhead, 2020-02-16:
Can you confirm with `M119` that the filament runout state changes when you insert and remove filament? And, if needed, you may also enable `PINS_DEBUGGING` and use `M43` to get a pins report, just to makes sure there are no conflicts.

MohanMaker, 2020-02-17:
@thinkyhead, I just tested both of your suggestions! M119 reports "filament: triggered" when I insert filament, "filament: open" when I remove the filament and "filament: triggered" when I insert filament again.
[image]

I also enabled PINS_DEBUGGING and then ran M43, and the pins report seems fine (I've attached it below in case you want to review). I also ran M43 P34 W (to watch/monitor the filament runout pin) and it successfully gave feedback when I inserted/removed the filament.)
[image]

Let me know if you have other debugging ideas/things you want me to try!

tfgcampos, 2020-05-14:
I can confirm this happens in 2.0.5.3. It's just like the triggered state is stored in a buffer.

It's easily reproducible if you manually pull and reinsert the filament after hitting start to print (during heating process, for example).

Yura80, 2020-05-23:
Happens to me as well with SKR 1.4 board on 2.0.5.3

boelle, 2020-06-21:
@MohanMaker still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

MohanMaker, 2020-06-25:
Yes, I still have this issue on my Ender-3. I have been meaning to look at how to fix the code, but haven't gotten to it yet. I will work on it as soon as I can (but am not too experienced with programming), and if anyone else has input or a solution please let us know!

MohanMaker, 2020-07-13:
@thinkyhead, just tested the bugfix-2.0.x branch on my Ender-3 and the problem still stands.

thinkyhead, 2020-07-13:
> Switch on the printer and begin a print from the SD card

This is the expected behavior. It is a feature, not a bug. Change your filament before you start an SD print.

closed 2020-07-13 by thinkyhead