MarlinFirmware/Marlin #16863: [BUG] LCD not responding on Z offset adjustment during printing

davidcgu, 2020-02-15:
### Bug Description

With latest release 2.0.3 my LCD gets unresponsive when trying to adjust Z offset, I tried printing or without with same result.

I was running exactly same config with 2.0.1 with no problem at all.

### My Configurations

I have enabled options as follows:

#define BABYSTEPPING
#define DOUBLECLICK_FOR_Z_BABYSTEPPING 
#define BABYSTEP_ZPROBE_OFFSET 
#define MINIMUM_STEPPER_PULSE 2
#define SQUARE_WAVE_STEPPING

### Steps to Reproduce

Whenever trying to change Z offset value on LCD rotating  the encoder.

**Expected behavior:** [What you expect to happen]

**Actual behavior:** LCD seems unresponsive but only on this option.

#### Additional Information

Config files attached
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4207495/Marlin.zip)

MangaValk, 2020-02-15:
here kinda the same issue using latest commit, lcd works but z isnt moving to the offset anymore.

thinkyhead, 2020-02-15:
Duplicate of #16617

closed 2020-02-15 by thinkyhead