MarlinFirmware/Marlin #16866: [BUG] Rounding problem editing numbers on LCD

drifkind, 2020-02-15:
### Bug Description

When editing some parameters on the LCD, the number loaded into the editor is wrong. (Close, bug not close enough.)

### My Configurations

Ender-3 configuration from MarlinConfigurations.

[marlin-config.zip](https://github.com/MarlinFirmware/Marlin/files/4207824/marlin-config.zip)

### Steps to Reproduce

1. Select Main --> Temperature --> Fan Speed.
2. Set fan speed to (e.g.) 10%.
3. Press knob to re-enter editor. Editor shows 9% instead of 10%.
4. Repeat ad lib.

Also happens some other places, though not quite as easy to find. For example:

1. Select Main --> Configuration --> Advanced --> Temperature --> Fact.
2. Set factor to 0.30. (Or 0.60 or 0.90...)
3. Press knob; editor shows 0.29.

### Suggested Fix

```
--- a/Marlin/src/lcd/menu/menu.h
+++ b/Marlin/src/lcd/menu/menu.h
@@ -236,5 +236,5 @@ class TMenuEditItem : MenuEditItemBase {
     typedef typename NAME::type_t type_t;
     static inline float unscale(const float value)    { return value * (1.0f / NAME::scale);  }
-    static inline float scale(const float value)      { return value * NAME::scale;           }
+    static inline float scale(const float value)      { return roundf(value * NAME::scale);   }
     static void load(void *ptr, const int32_t value)  { *((type_t*)ptr) = unscale(value);     }
     static const char* to_string(const int32_t value) { return NAME::strfunc(unscale(value)); }
```

thinkyhead, 2020-02-15:
Thanks for pointing out the issues. The fan percent issue has been repaired by correcting the rounding. The other items where you see 0.29 instead of 0.30 are due to the issue the way floating point numbers work. Some calculations result in float values that cannot be represented perfectly so you end up seeing a value like 0.2999999. Although the values themselves cannot be rounded to any precision we want, the issue can been addressed by concealing the rounding error on the display, so that's what I have done.

closed 2020-02-15 by thinkyhead

drifkind, 2020-02-15:
Ought to mind my own business but...is it possible you overthought that?

thinkyhead, 2020-02-16:
We get rounding issues a lot in code world, so we already had `FIXFLOAT(x)` at the ready.

commit efdaf940 (2020-02-15): Fix byte-to-percent display

Fixes #16866
commit 5584da38 (2020-02-15): Conceal float rounding errors on display

Fix #16866