MarlinFirmware/Marlin #16869: [BUG] SKR PRO Runout sensor

MikeEitel, 2020-02-15:
### Bug Description
I try to configure filament runout sensor for bigtree skr pro v1. 1
Enabling the sensor gave me impression it is not yet implemented for that board. 
I tried same config successfully for atmega, mks, due, re-arm. All accept 

### My Configurations

DIY Delta with 3 extruders.  Version 2.0.3

### Steps to Reproduce
Same config without activated runout is ok to compile. 
Just include in pinxxxx file a #define filament_runout_sensor pin does not work neither.

thinkyhead, 2020-02-16:
The filament sensor is simply a switch. Use `M119` to test it and make sure it's properly configured. Enable it with `M412 S` if needed.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-02-16 by thinkyhead

MikeEitel, 2020-02-16:
Thx thinkyhead
I know the way and usage of filament runout. 
BUT
If i try to implement the function in config.h it generates a compilation error. 
My impression is that for the skr pro 1.1 the declaration of such pin is missing. 
For all the boards i mentioned this flag worked flawless. 

I am not a real coder but use vsc for different platforms. But the search for such implementation is beyond my payroll.

thinkyhead, 2020-02-16:
Yes, if the pins file doesn't define the pin then you will have to define it. There is no standard pin for it, AFAIK.

MikeEitel, 2020-02-16:
I tried to do that in the according pin file (#define runout_pin...) but my promlem is that the "validation macros" stop the compiling. I was also not able to use the X-min pin for that as in my other boards acceted. 
What do you propose me as best way to make augment the board with such pin.
P. S. Your fast in respone. Wow. THX for your efforts.

AnHardt, 2020-02-16:
Please cite the exact error.
Likely you have to set the pin previously used by the same pin number to -1 . Else the checks moan about using the same pin twice.

MikeEitel, 2020-02-16:
THX AnHardt

I'm getting
In file included from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32\HAL_SPI.cpp:25:
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:693:6: error: #error "FILAMENT_RUNOUT_SENSOR requires FIL_RUNOUT_PIN."
     #error "FILAMENT_RUNOUT_SENSOR requires FIL_RUNOUT_PIN."

Attached the files I change in ZIP

-
[MeTrium_SKR_Pro.zip](https://github.com/MarlinFirmware/Marlin/files/4209787/MeTrium_SKR_Pro.zip)

MikeEitel, 2020-02-16:
THX AnHardt
You gave me the right direction.

In Configuratio.h:   I had to (re-)activate the Y-pin

// @section homing
................
//#define USE_XMIN_PLUG
**#define USE_YMIN_PLUG  // use as filamant runout sensor**
#define USE_ZMIN_PLUG // a Z probe
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG 
#define USE_ZMAX_PLUG

and in pins_BTT_SKR_PRO_V1_1.h: I had to make the additional declaration:

//
// Get a filamant runout sensor                       // MeTrium generation of FILAMENT_RUNOUT_SENSOR pin
//
#ifdef FILAMENT_RUNOUT_SENSOR
  #define  FIL_RUNOUT_PIN  PE12  // Y_MIN_PIN          PE12
#endif

THX both of you  !
Good job, we all have big benefits from your work
THX again

thinkyhead, 2020-02-16:
So if you put this in your config it doesn't work?
```cpp
#define FIL_RUNOUT_PIN PE12
```

MikeEitel, 2020-02-17:
I had to (re-)enable in config.h the USE_YMIN_PLUG first.
For me that makes sens as otherwise there is no "description" how to configure the hw of the pin PE12. 
That is probably the reason why my tries to use one of the pins from unused extention 1 and 2. And to include such definitions for the mcu in marlin is beyond my payroll. Thx