MarlinFirmware/Marlin #16870: [BUG] Double ADC read frequency (#16864) breaks temperature readings

thisiskeithb, 2020-02-16:
### Bug Description

After rebasing my config to 64b96f390867700834641b01bbb37a40867a7bef/"Double ADC read frequency (#16864)", my temperature readings are 0 & -14. If I rebase on top of the prior commit 
(52a126808917f480e0ee1f8bcdf8fb6eeaacc4ba/"[cron] Bump distribution date (2020-02-16)") readings come back.
### My Configurations

[btt002-mk3s-bmg.zip](https://github.com/MarlinFirmware/Marlin/files/4209283/btt002-mk3s-bmg.zip)

### Steps to Reproduce

1. Build config on top of 52a126808917f480e0ee1f8bcdf8fb6eeaacc4ba & observe correct temps: [image]
2. Build config on top of 64b96f390867700834641b01bbb37a40867a7bef & observe the incorrect temps: [image]

**Expected behavior:** Temperature readings to work correctly.

**Actual behavior:** Temperature readings do not work correctly.

thisiskeithb, 2020-02-16:
Something similar occurs after 64b96f3 with the SKR 1.3/`LPC1768`, but correct temps are displayed pretty quickly after boot: https://youtu.be/72AW736kWrU

With the prior commit (52a1268), the correct temps are displayed correctly after boot without a delay: https://youtu.be/PRMVtg238pE

thinkyhead, 2020-02-16:
I may be missing a flag, or the ADC reading is just acting up due to altered timing. I've reverted the change until I can give it more testing and refinement.

closed 2020-02-16 by thinkyhead