MarlinFirmware/Marlin #16871: skr1.4 marlin2.0.3 dual Z sensorless homing not triggr

sacy80, 2020-02-16:
### Bug Description

dual Z sensorless homing not trigger Bigtreetech TMC 2209 and TMC2130

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[203_config.zip](https://github.com/MarlinFirmware/Marlin/files/4209290/203_config.zip)

### Steps to Reproduce

G28 Z

Z1 and Z2 will not register stall when it hit an endBlock

**Expected behavior:** [What you expect to happen]
Z1 and Z2 will stop when it hit each endblock

**Actual behavior:** [What actually happens]
Z1 and Z2 continue moving

#### Additional Information
has try changing Z_STALL_SENSITIVITY from 255 to 100 to 68 to 4 .. still same ressult Z will not trigger.

Try swap X and Y driver that sensorless tested working , over to Z1 and Z2.. with the same result, Z will not trigger

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thinkyhead, 2020-02-16:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

sacy80, 2020-02-16:
ok . let me try bugfix-2 :) will get back

sacy80, 2020-02-16:
Tested with recent bigfix2... but same result.  
same X and Y able to trigger.. but Z1 and Z2 not able to trigger

thinkyhead, 2020-02-17:
Marlin trusts the drivers to tell it when they trigger a hit. If the firmware isn't responding, it's because the driver signals are not being produced. So the DIAG pin is not connected, the endstop pin definition is incorrect, or the driver is not in the right mode.

sacy80, 2020-02-17:
humm. let me double check the connection.. will get back again :)

zigou83, 2020-02-22:
Same issue here on single Z. Diag well connected to the pin (when sending Vcc to the pin it stop the motor), motor and driver switched to test but same result. Sensorless work on other axis. (S6+2209)

sacy80, 2020-02-24:
im downloading marlin 2.0.4 to test out

colemorgan, 2020-04-05:
Having the same issue. Any update?

superkefir, 2020-04-07:
Having the same issue with tmc5160. X and Y axis works fine. Double Z axis not working.

superkefir, 2020-04-11:
Fixed. Added jumpers to Z and Z2 pins on plate.

sacy80, 2020-04-11:
havent got the time to test the new 2.0.5.3..

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle