MarlinFirmware/Marlin #16873: Auto Bed Levelling not working

TheSingularity-42, 2020-02-16:
### Bug Descri
I have an Anet A8 printer running Marlin 2.0 with a DC 6-36V 100mA LJ8A3-2-Z/BX inductive sensor. When I activate the auto home or auto bed levelling it does not work. The printer head goes through the same procedure as homing with a switch
Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4209387/Configuration.zip)

Expected Behaviour: Printer uses the probe to level the bed.
Actual Behaviour: Auto homes with same procedure as before I changed my Marlin code.

thinkyhead, 2020-02-16:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-02-16 by thinkyhead

mrv96, 2020-03-08:
@TheSingularity-42 I have the same problem. Did you solved it? If yes, how?