MarlinFirmware/Marlin #16875: [Q] Temperature oscillating after M109 and M600

SCiunczyk, 2020-02-16:
I have problem with hotend temperature oscillation. Asked on RepRap forum https://reprap.org/forum/read.php?415,867456, but no one can help.

Here is description:

I have problem with Marlin 2.0.3 on my Prusa clone.

I have Precision Piezzo sensor - so to have good Z zero reading, I have to preheat the nozzle first.

If the nozzle is on exact temperature and I start printing (I'm using Octoprint) the printer perform all commands from initial Slic3r G-Code until line:

`M109 S265 ; set temperature and wait for it to be reached`

then temperature start oscillating until heater timeout is triggered.

You can see this on picture attached Oscillating01.jpg.

If I lower the hotend temperature by say 20-25°C then start printing - there is no problem. Temperature reach target level and there is no oscillating.

This same phenomenon I have with M600 command. Printer has filament run out detector, sometimes I want to change colour using M600 - it's very, difficult because very often I get oscillating problem see picture Oscillating02.jpg

The trick to to rescue the print when temperature is oscillating, is to somehow lower the hotend temperature by 15-20°C (strongly blowing or touching heated block with wrench). If I'm lucky I can have print resumed if not, I just have temperature run-out error.

I was trying PID tuning - same situation. But I thing that problem is somewhere else - As you can see on picture attached Oscillating01.jpg hotend target temp is reached almost perfectly (in PID meaning). Maybe the D factor should be a little stronger - but lower D - means less chances to oscillate.

Any idea where to start looking for help?

best regards
Slawek
[image]
[image]

lrpirlet, 2020-02-16:
I think this is NOT a bug, so it should NOT be here… The entry on the RepRap forum is NOT that old… You should have read the rules to enter something here… 
Common answer will be some like:

> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on this page.) For best results getting help with configuration and troubleshooting, please use the following resources:
> •	MarlinFW.org Marlin Documentation
> •	RepRap.org Marlin Forum
> •	Tom's 3D Forums
> •	Facebook Group "Marlin Firmware"
> •	Facebook Group "Marlin Firmware for 3D Printers"
> •	Marlin Configuration on YouTube
> •	Marlin Discord server. Join link: https://discord.gg/n5NJ59y
> After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

However, let me GUESS that the PID cannot be executed because when cooled, by some filament inserted  you get a BANG-BANG behaviour… Notice that the oscillation has a very small constant amplitude… Verify that your `PID_FUNCTIONAL_RANGE` is at least 10 so that the system has time enough to perform the correction… 

NO the induced temp at the cartridge by the heat process and the detected temp is NOT synchronous but need time… Not enough time kills the PID process and results in BANG-BANG behaviour.

thinkyhead, 2020-02-16:
If you find that you get different PID behavior with different situations, you may be able to obtain good PID values for those situations and then apply them during printing with `M301` to deal with materials changing, new temperature ranges, etc.

SCiunczyk, 2020-02-17:
Thanks, for the suggestions.
The PID_FUNCTIONAL_RANGE is equal 10 now so I will try increase. 

The please note that there is no filament insertion in first example. The filament is all the time loaded. I preheat the Hotend and Bed to target temperature then when is stable I just start printing and after all standard Slic3r initial commands on M109 printer start the oscillation. 

Honestly it not look as BANG-BANG to me. On my bed I have 230V 600W heater connected via SSR and then shape of temperature fluctuation (obviously BANG-BANG) is saw teeth style. The oscillation of hotend looks very sinusoidal IMHO.

The hotend is original E3D V6 with 40W heater on Titan AERO direct extruder.

When I start printing on cold hotend (or with temperature  20-25°C lower than target) the M109 is increasing temperature "from the bottom" and there is no oscillation. 

What I'm doing now is just commenting out the M109 command from G-Code before loading to Octoprint. Unfortunately Slic3r has this startup code somehow hardcoded or I can't find how to change it permanently.

SCiunczyk, 2020-02-18:
First test with PID_FUNCTIONAL_RANGE=20 was OK. There was two bumps up/down on M109 but lower than 20°C so PID process was not killed. Still wonder what could be source of the temperature bumps. First down then up, and after that PID took controll.

lrpirlet, 2020-02-18:
OK, I would advise to tune your PID with your machine….
Yes use M303 (https://marlinfw.org/docs/gcode/M303.html) **but DO NOT issue U**… yet issue the correct target temperature...
Do refer to this article: https://reprap.org/wiki/PID_Tuning, do read this paragraph titled _Modifying Marlin Autotune parameters_ … 
As you can see, the value generated by M303 is a good approximation, yet still an approximation...
Now you want to understand the effect of each of the 3 parameters... goto https://en.wikipedia.org/wiki/PID_controller and watch the animation under paragraph named _Manual tuning_ 
direct link to animation is https://en.wikipedia.org/wiki/PID_controller#/media/File:PID_Compensation_Animated.gif
No, I am

SCiunczyk, 2020-02-18:
Hi @lrpirlet. Probably you didn't notice in my first post "I was trying PID tuning - same situation." 

To make it funny by accident I graduate from Warsaw Technology University - speciality Automatic Control, so I have a little knowledge about PID stuff.... ;)

If we put the "M109 problem" aside, the regulation of temperature, response for changes - send manually from Octoprint Console or from Printer LCD Panel is very good, quick, with little dying oscillation when target is reached. Then temperature is stable and very very flat. Reactions to external temperature disruption is also model. For example changing nozzle (cold wrench) drops temperature by 15°C but PID algorithm recover target nice and quickly. This allows me to think that I don't need to adjust the PID parameters.

Also if I start printing without preheating and firmware reach M109 with low hotend temperature there are no oscillations. ONLY when temperature of hotend is at target level and is stable - M109 leads to oscillations. Probably this is rare situation because by definition M109 is used as command setting new target.

I will try to investigate what's going on in M109 source code. Maybe I will find something interesting?

SCiunczyk, 2020-02-19:
I did simple test.

- I sent from terminal: M104 S265, 
then after 5 minutes when temperature was stable and 265°C
- I sent from terminal: M109 S265

only two commands with this same target temperature. The result shows temperature fluctuation just after M109 S265. Can anybody explain why?

Here is the output from OctoPrint terminal and temperature diagram:

```
Recv:  T:33.41 /0.00 B:46.61 /0.00 @:0 B@:0
Send: M104 S265
Recv: ok
Recv:  T:33.43 /265.00 B:46.60 /0.00 @:127 B@:0
Recv:  T:34.38 /265.00 B:46.60 /0.00 @:127 B@:0
Recv:  T:37.33 /265.00 B:46.58 /0.00 @:127 B@:0
Recv:  T:42.08 /265.00 B:46.58 /0.00 @:127 B@:0
Recv:  T:47.25 /265.00 B:46.56 /0.00 @:127 B@:0
Recv:  T:53.01 /265.00 B:46.35 /0.00 @:127 B@:0
Recv:  T:59.28 /265.00 B:46.32 /0.00 @:127 B@:0
Recv:  T:65.24 /265.00 B:46.07 /0.00 @:127 B@:0
Recv:  T:72.01 /265.00 B:45.81 /0.00 @:127 B@:0
Recv:  T:78.15 /265.00 B:45.79 /0.00 @:127 B@:0
Recv:  T:84.24 /265.00 B:45.61 /0.00 @:127 B@:0
Recv:  T:90.28 /265.00 B:45.33 /0.00 @:127 B@:0
Recv:  T:96.64 /265.00 B:45.49 /0.00 @:127 B@:0
Recv:  T:102.51 /265.00 B:45.54 /0.00 @:127 B@:0
Recv:  T:108.31 /265.00 B:45.26 /0.00 @:127 B@:0
Recv:  T:114.01 /265.00 B:45.26 /0.00 @:127 B@:0
Recv:  T:119.49 /265.00 B:46.40 /0.00 @:127 B@:0
Recv:  T:125.52 /265.00 B:46.40 /0.00 @:127 B@:0
Recv:  T:130.74 /265.00 B:46.35 /0.00 @:127 B@:0
Recv:  T:136.12 /265.00 B:46.32 /0.00 @:127 B@:0
Recv:  T:141.26 /265.00 B:46.32 /0.00 @:127 B@:0
Recv:  T:146.58 /265.00 B:46.32 /0.00 @:127 B@:0
Recv:  T:152.07 /265.00 B:46.32 /0.00 @:127 B@:0
Recv:  T:157.15 /265.00 B:46.28 /0.00 @:127 B@:0
Recv:  T:162.02 /265.00 B:46.23 /0.00 @:127 B@:0
Recv:  T:166.89 /265.00 B:46.12 /0.00 @:127 B@:0
Recv:  T:171.57 /265.00 B:46.09 /0.00 @:127 B@:0
Recv:  T:176.81 /265.00 B:46.05 /0.00 @:127 B@:0
Recv:  T:181.36 /265.00 B:45.87 /0.00 @:127 B@:0
Recv:  T:186.25 /265.00 B:45.82 /0.00 @:127 B@:0
Recv:  T:190.56 /265.00 B:46.04 /0.00 @:127 B@:0
Recv:  T:195.03 /265.00 B:45.97 /0.00 @:127 B@:0
Recv:  T:199.31 /265.00 B:45.81 /0.00 @:127 B@:0
Recv:  T:204.06 /265.00 B:45.79 /0.00 @:127 B@:0
Recv:  T:208.21 /265.00 B:45.64 /0.00 @:127 B@:0
Recv:  T:212.50 /265.00 B:45.81 /0.00 @:127 B@:0
Recv:  T:216.67 /265.00 B:45.81 /0.00 @:127 B@:0
Recv:  T:221.00 /265.00 B:45.77 /0.00 @:127 B@:0
Recv:  T:225.00 /265.00 B:45.74 /0.00 @:127 B@:0
Recv:  T:228.56 /265.00 B:45.53 /0.00 @:127 B@:0
Recv:  T:232.50 /265.00 B:45.26 /0.00 @:127 B@:0
Recv:  T:236.25 /265.00 B:44.93 /0.00 @:127 B@:0
Recv:  T:240.00 /265.00 B:44.47 /0.00 @:127 B@:0
Recv:  T:243.65 /265.00 B:44.71 /0.00 @:127 B@:0
Recv:  T:247.71 /265.00 B:44.54 /0.00 @:127 B@:0
Recv:  T:251.46 /265.00 B:44.43 /0.00 @:127 B@:0
Recv:  T:254.90 /265.00 B:44.41 /0.00 @:127 B@:0
Recv:  T:257.71 /265.00 B:44.39 /0.00 @:48 B@:0
Recv:  T:260.00 /265.00 B:44.25 /0.00 @:18 B@:0
Recv:  T:262.66 /265.00 B:44.36 /0.00 @:0 B@:0
Recv:  T:264.22 /265.00 B:45.00 /0.00 @:0 B@:0
Recv:  T:264.22 /265.00 B:45.00 /0.00 @:0 B@:0
Recv:  T:262.97 /265.00 B:44.98 /0.00 @:47 B@:0
Recv:  T:262.19 /265.00 B:44.87 /0.00 @:68 B@:0
Recv:  T:260.63 /265.00 B:44.71 /0.00 @:105 B@:0
Recv:  T:260.00 /265.00 B:44.65 /0.00 @:112 B@:0
[... 104 lines cut ...]
Recv: ok
Send: M113 S2
Recv: ok
Recv:  T:257.40 /265.00 B:38.67 /0.00 @:127 B@:0
Recv:  T:258.65 /265.00 B:38.65 /0.00 @:127 B@:0
Recv:  T:259.48 /265.00 B:38.52 /0.00 @:111 B@:0
Recv:  T:262.03 /265.00 B:38.35 /0.00 @:51 B@:0
Recv:  T:264.22 /265.00 B:38.67 /0.00 @:17 B@:0
Recv:  T:266.41 /265.00 B:38.63 /0.00 @:0 B@:0
Recv:  T:267.19 /265.00 B:38.56 /0.00 @:0 B@:0
Recv:  T:267.34 /265.00 B:38.54 /0.00 @:3 B@:0
Recv:  T:266.25 /265.00 B:38.46 /0.00 @:39 B@:0
Recv:  T:265.16 /265.00 B:38.44 /0.00 @:61 B@:0
Recv:  T:263.91 /265.00 B:38.33 /0.00 @:86 B@:0
Recv:  T:262.97 /265.00 B:38.33 /0.00 @:96 B@:0
Recv:  T:262.81 /265.00 B:38.33 /0.00 @:92 B@:0
Recv:  T:264.38 /265.00 B:38.33 /0.00 @:48 B@:0
Recv:  T:265.00 /265.00 B:38.33 /0.00 @:39 B@:0
Recv:  T:265.63 /265.00 B:38.33 /0.00 @:31 B@:0
Recv:  T:266.88 /265.00 B:38.33 /0.00 @:9 B@:0
Recv:  T:266.09 /265.00 B:38.27 /0.00 @:36 B@:0
Recv:  T:266.41 /265.00 B:38.27 /0.00 @:30 B@:0

```
[image]

lrpirlet, 2020-02-21:
OK, this looks interesting… just like if M109 would heat before it checks temperature….

My advice here would be to verify in the code… However I have NOT ENOUGH information (nor time presently)…

What COULD be interesting would be to perform M104 S265 followed by M109 S260 (YES go down) and verify if it still heats above 265°c... 

This would indicate that the code assumes M109 Sxxx to HEAT first then verify. Note that this could be expected since M109 S265 means wait for temperature while M109 R265 means wait for temp to go up or down (see https://marlinfw.org/docs/gcode/M109.html) …

What would be even more interesting would be to perform M104 S265 followed by M109 R265 and see what happens...

1. Needed to do is  

- try with bugfix-2.0.x (this MAY have been fixed)

- give us your FULL configuration.h and configuration_adv.h so that we can refer to YOUR hardware (for example, other factors may be needed)

2. Best would be you to

- find the problem or feature

- Propose a fix or a new feature via a PR so that @thinkyhead (or someone else with the privileges) may merge it

SCiunczyk, 2020-02-21:
Sure, as my printer finish current print (3-4 hours) I'll perform tests and will share the results with more tech data.

SCiunczyk, 2020-02-21:
Hi again, I did perform test recommended by @lrpirlet . Here are results. 

Fresh Marlin version bugfix-2.0.x with very last UBL problem patch. To be sure, I checked also how firmware behave with standard PID settings and after PID AutoTune - same results (I mean my M109 problem)

[image]
Test one (5°C delta between target and M109):

[image]
Test two (1°C delta between target and M109)

[image]
Test three (20°C delta between target and M109)

[image]
Test four (5°C delta between target and M109 with R instead of S)

I have three modified files in my config:
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4237371/Configurations.zip)

I checked the  M104_M109.cpp file and at first glance it seems to me that the problem is somewhere deeper. I don't have much experience in firmware debugging directly in the printer. But I'll look again the weekend.

closed 2020-06-20 by boelle