MarlinFirmware/Marlin #16878: [BUG] missing "Click to resume" and no clicks register with PRINTER_EVENT_LEDS

Moyster, 2020-02-16:
### Bug Description

When a print is done, the status led turns green, x/y homes, LCD doesn't show "click to resume" nor registers clicks (led doesn't turn off/white, menus can't be confirmed, naviguating sd works)

### My Configurations

[cfg.zip](https://github.com/MarlinFirmware/Marlin/files/4210545/cfg.zip)

### Steps to Reproduce
When a print from SD is over, needs PRINTER_EVENT_LEDS defined.

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]
should display a "click to resume" screen

**Actual behavior:** [What actually happens]
doesn't display a "click to resume" screen

#### Additional Information
-

thinkyhead, 2020-02-17:
Although it doesn't say "Click to resume…" it should still be waiting for a click. Are you able to get it un-stuck by sending `M108` from the host?

thinkyhead, 2020-02-17:
For testing purposes put `M111 S3` at the front of a small G-code file that does nothing but a few moves, and run that from SD. This will give extra logging and maybe show where the firmware is getting stuck.

Moyster, 2020-02-17:
succesful print : (bugfix tree before +/- coolstep)  

> echo:M105
> echo:G1 F2100 E793.54679
> 		 
> echo:M140 S0
> echo:M107
> echo:M104 S0
> echo:M140 S0
> echo:G92 E1
> echo:G1 E-1 F300
> echo:G28 X0 Y0
> echo:M84
> echo:enqueueing "M31"
> Done printing file
> echo:M82
> echo:M104 S0
> echo:M105
> echo:M31
> echo:Print time: 11m 40s
> echo:M105

today's bugfix :  

> echo:M105
> echo:G1 F2100 E793.54679
> echo:M105
> echo:M140 S0
> echo:M107
> echo:M104 S0
> echo:M140 S0
> echo:G92 E1
> echo:G1 E-1 F300
> echo:G28 X0 Y0
> echo:M84
> echo:enqueueing "M31"
> Done printing file
> >>> m108
> SENDING:M108 

M108 didn't change a thing, lcd menu is still bugged and impossible to confirm any actions (auto-home, etc).

Moyster, 2020-02-21:
https://github.com/MarlinFirmware/Marlin/commit/19991e269b07f77e2ef4134da406bb1a6afe9d25 didn't fix it.  
in M0_M1.cpp:  
`    else if (!parser.seenval('Q')) {`
code after this never runs, so `LCD_MESSAGEPGM(MSG_USERWAIT);` never displays a confirm screen.

ellensp, 2020-02-21:
M0_M1 is broken see https://github.com/MarlinFirmware/Marlin/issues/16908

Moyster, 2020-02-22:
Closing as https://github.com/MarlinFirmware/Marlin/issues/16908 looks to be the same origin.

closed 2020-02-22 by Moyster