MarlinFirmware/Marlin #16879: [BUG] Probing broken for probes with Z-offset

AndersSahlman, 2020-02-16:
### Bug Description
If a probe with a negative z offset is used the current.z position seem to be calculated using the probe value in wrong direction

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4210585/Configuration.zip)

### Steps to Reproduce

1. Start with Z probe offset 0
2. Enable debug info , M111 S32
3. Run delta calibration , G33
4. The probe fails check the debug output
5. Reboot printer
6. Set Z probe offset to the proper value (negative to the Z where it failed) in my case M851 Z-9.22
7. M111 S32
8. G33
9. The probe still failes , check the output.

Running delta configuration with probe offset set to zero gives a probe fail with Z height 9.21 and that is same as my normal probe offset:
Recv: SLOW Probe fail!
Recv:   current_position= X0.00 Y0.00 Z9.21 : <<< run_z_probe
Recv:   current_position= X0.00 Y0.00 Z9.21 : Probe::set_deployed

If I set my probe offset (with M851 Z-9.21) the probe is still failing and now Marlin thinks the Z height has raised with the probe offset so it still fails
Recv: SLOW Probe fail!
Recv:   current_position= X0.00 Y0.00 Z18.43 : <<< run_z_probe
Recv:   current_position= X0.00 Y0.00 Z18.43 : Probe::set_deployed

**Expected behavior:** 
Delta calibration should at least in second case be able to run as probe offset is correct and the Z position should then be 0.  But as delta configuration is in part calculating the delta height it should be able to run even if the delta height is a bit wrong and even if the probe offset has not yet been set.

**Actual behavior:**  
Delta calibration always thinks the probe triggered too high

#### Additional Information

This could maybe have been wrong before but after *Better probe fault handling* #16811 probing now fails everytime unless you raise the Z_CLEARANCE_MULTI_PROBE value to much larger than you would like to.

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

AndersSahlman, 2020-02-16:
Removing line 620 in probe.cpp makes the probe not fail, however also removes the purpose of that change...

AndersSahlman, 2020-02-16:
Unfortunatly removing that line and calibrating with proper probe offset the calibration and bed level works but it prints as far below the bed as the probe offset is.... so in my case it wanted to print 9.2mm below the bed, not too good...

thinkyhead, 2020-02-16:
What kind of probe is sensitive enough to trigger at almost a full cm from the surface?

AndersSahlman, 2020-02-17:
It is not sensitivity but probe placement that is the reason. Most delta probes are mounted under the nozzle, this for example triggers about 2 cm below the nozzle so it needs a probe offset of over 20 mm
https://www.ebay.com/itm/Auto-Level-Sensor-Heatbed-Leveling-Probe-Anycubic-Kossel-3D-Printer-/183730893548

pablopeu, 2020-02-17:
Had the same problem on an Anycubic Kossel Delta plus, the second touch of delta calibration fails the calibration procedure. Offset is 16.17mm below the nozzle and its attached to the moving head by a magnet, its the provided version 2 probe (wired NC).

AndersSahlman, 2020-02-17:
This *can* be a rebasing problem on my side. I have cloned a new fresh copy and I think the problem is not present.

closed 2020-02-17 by AndersSahlman