MarlinFirmware/Marlin #16880: [BUG] Bulge on linear advance speed change

Liger0, 2020-02-16:
Bug Description

When using linear advance on my bowden system, no matter what value, even between 0 and 5, there is a big drop of plastic at the point where the speed changes, even tho the 3 segments are changed correctly (my correct k value is around 0.5)

### My Configurations

[cura.zip](https://github.com/MarlinFirmware/Marlin/files/4210929/cura.zip)
Cr10S, skr 1.3, TMC 2209
Scurve acceleration, Standard jerk

**Expected behavior:** [What you expect to happen]

Linear transition between slow and fast speed

**Actual behavior:** [What actually happens]

Big drop left at the speed change

thinkyhead, 2020-02-16:
In your slicer it will help to tell it not to subdivide curves too finely. Too many small segments lead to planner starvation because the G-code input cannot feed the queue fast enough.

Liger0, 2020-02-17:
I am directly referring to the official linear advance test. 
There is not even a minimal difference in the bulge between 0 and 3.

Liger0, 2020-02-18:
Tried with scurve off: No difference.

fatalbranch, 2020-03-01:
> there is a big drop of plastic at the point where the speed changes

> Tried with scurve off: No difference.

I think I have the same issue.
While calibrating my linear advance settings, I also noticed a bulge at the speed change.
I used the official linear advance test found here: https://marlinfw.org/tools/lin_advance/k-factor.html
While trying around I also tried very low acceleration settings and noticed something strange: 
At the point where the print head is supposed to accelerate, it actually decelerates first before speeding up. This can be clearly seen and heard. I don't think this behavior is correct. Maybe you want to try it out for yourself? These are the speed settings I used:
```
; Settings Speed:
; Slow Printing Speed = 900 mm/min
; Fast Printing Speed = 3600 mm/min
; Movement Speed = 7200 mm/min
; Retract Speed = 2400 mm/min
; Printing Acceleration = 40 mm/s^2
```

thinkyhead, 2020-03-02:
In order to track down spurious issues, after first testing the latest `bugfix-2.0.x` code it can help to go back in time to find a version of Marlin that doesn't exhibit the issue. Through a process of bisection you can quickly locate the day when the issue first arises. We can then examine the code changes from that time period and see which changes are the most likely problem.

VanessaE, 2020-03-09:
I can confirm this also - when moving from the first slow segment into the middle fast segment, it not only decelerates, it also briefly moves the extruder *backward* just before the normal surge forward.  It's been a long time since I last tried to set up LA, so I couldn't even begin to guess when it last worked.

Liger0, 2020-03-09:
It never worked correctly to me.

fatalbranch, 2020-03-11:
My i3 plus 3D printer which shows the strange behavior is running a Marlin firmware version 1.1.4. (You can find the the firmware project here  https://github.com/Silverquark/i3PlusPlus)

If the bug was never fixed in between, this would probably mean it has been around since at least mid 2017.

thinkyhead, 2020-03-11:
> briefly moves the extruder backward

Yeah, Linear Advance is allowed to do that.

It's been a while since @Sebastianv650 has been around, but I'm sure if LA was seriously broken he would have harshly criticized us by now!

VanessaE, 2020-03-12:
> > briefly moves the extruder backward
>
> Yeah, Linear Advance is allowed to do that.

Sure, if it's moving from a fast segment to a slow segment in the calibration pattern, E moving backward slightly makes perfect sense, since it has to bleed-off some pressure just before the slow segment starts.  On the transitions from slow to fast on each line, one would expect only a forward E surge, to boost the pressure just before the fast segment starts.

That's how it worked when LA v1.5 was introduced (and I seem to recall that that's more or less how it worked in the old implementation that preceded it).

But now, a slow-to-fast transition is getting both a backward move and a forward surge, one after the other, which cancel each other out.  Thus the start of the fast segment never fills-out.  I did not take note of the behavior of fast-to-slow transitions, but I'll hazard a guess that an inverse pair of moves happens there, since the fast-to-slow bulges never smooth out.

Since @Sebastianv650 hasn't stopped by to yell at anyone, I can only assume he has moved on to some other project.

ManuelMcLure, 2020-03-12:
The last I remember hearing from @Sebastianv650 was https://github.com/MarlinFirmware/Marlin/issues/10272#issuecomment-534905221 where it looked like he had moved on to other things.

XDA-Bam, 2020-04-14:
@Liger0 Can you upload a picture of the bulge? I think I have a similar problem, but I'm not sure because the description is a bit vage. This is an image ob the bulge on my pattern:
[image]

XDA-Bam, 2020-04-14:
Good. What extruder do you use? Bowden, direct drive or flexdrive? I've got the problem on a stock Ender 3 Pro, which uses a Bowden.

qwewer0, 2020-04-14:
I had this really noticeably on stock bowden Ender 3, and still can see it - even if very slightly - with direct drive now.

Liger0, 2020-04-14:
I use bowden.

NoxIX, 2020-04-21:
I have this issue as well, the bulge exists all the way up to 1.8K or so. Running marlin 1.1.9 on an Ender 3.

viper93458, 2020-04-21:
Same issue here with SKR pro 1.1, 2208s, Bowden and latest bug fix from today. It’s been an issue for a VERY long time even back to the AVR days in my past setup. This has been seen across my i3 variants with both Bowden and direct drive. I have just lived with it and assumed I was just not calibrating something properly. Which is still a likely possibility. :)

-William

thinkyhead, 2020-04-21:
> …it looked like he had moved on to other things.

I guess that means we'll have to actually understand his code now! I was hoping to maybe get him to throw us a hint in a short comment. Has anyone looked at Prusa Firmware to see if their algorithm is significantly different?

_**Anyways…**_

Just so everyone is on the same page here:

- S-curve acceleration is not deemed to be compatible with Linear Advance. You can experiment with it, so it's not completely blocked, but Linear Advance expects to be working alongside of Linear Acceleration. So, please test with S-Curve Acceleration disabled and compare those results.

NoxIX, 2020-04-21:
>S-curve acceleration is not deemed to be compatible with Linear Advance. You can experiment with it, so it's not completely blocked, but Linear Advance expects to be working alongside of Linear Acceleration. So, please test with S-Curve Acceleration disabled and compare those results.

Oh really? I've only heard it being used along with S-curve, I'll go test this rq.

NoxIX, 2020-04-22:
@thinkyhead 

So I tested it with S-curve disabled, if there was a difference, it was marginal. There is still a bulge at the slow down transition. This, eventually goes away but then your slower speeds are much thinner, if there at all. 

I tested V1 last night and found that the opposite problem exists, where the fast portion is not fed fast enough.

XDA-Bam, 2020-04-25:
Just to see what happens, I tested it with different accelerations.

**80 mm/s²** 🔽 
[image]

**400 mm/s²** 🔽 
[image]

**2000 mm/s²** 🔽 
[image]

All other settings are identical (speed 20/80/20 mm/s, segment lengths and extrusion settings default, retraction 3 mm). As you can see, there is no bulge at 80 mm/s². At 400 mm/s², it's already very visible. This is a typical result for my printer. At 2000 mm/s², it's only slightly bigger than at 400.

Looking at where the bulge is placed shows, that most of it (maybe 80%) is directly after the speed change 20 mm before the end of the line. That is, after the deceleration phase, which I would have thought to be somewhat more critical.
[image]

NoxIX, 2020-04-26:
@XDA-Bam Your 2k results are exactly what I'm seeing. So it seems for LA to work worth a damn, you have to slow your printer to a crawl; which seems to defeat the purpose.

qwewer0, 2020-04-26:
@XDA-Bam Is S-Curve Acceleration on?

NoxIX, 2020-04-26:
> @XDA-Bam Is S-Curve Acceleration on?

From my tests, it won't matter, the results stay the same.

XDA-Bam, 2020-04-26:
@qwewer0 No S-curve on those tests. Also, using either Jerk or JD does not make a difference, as far as I can tell.

XDA-Bam, 2020-04-26:
My understanding is, that the planner never moves faster than the nominal speed of a block. In case of the LA test pattern, that would mean the speed profile should look roughly like this:
[image]

So the deceleration happens right up to the 60 mm line, at which the second slow segment begins. For anybody accustomed to Marlin movement code: **Is this correct?**

Pizzahd88, 2020-04-27:
I am seeing exactly the same result with bulge in the deceleration zone.
I have tried to switch all kind of features off (including s_curve, JD), but it does not seem to matter - the blob stays present.
Is it possible that the flow decrease happens too late in relation to the deceleration? It looks like the deceleration happens and while it being decreased the flow decrease follows - but a bit too late.

jll544, 2020-05-21:
I'm also seeing this bulge on the LA calibration pattern regardless of K-value.

> @thinkyhead Has anyone looked at Prusa Firmware to see if their algorithm is significantly different?

Looking at Prusa's [stepper.cpp history](https://github.com/prusa3d/Prusa-Firmware/commits/MK3/Firmware/stepper.cpp), indeed they have put significant work into linear advance since merging LA v1.5 one year ago. They've made enough changes that their [April 8, 2020 commit](https://github.com/prusa3d/Prusa-Firmware/commit/919386c957eaf35ff0cca442c9e3000ccf36f2ae) mentions "[...] removing the last vestiges of the original implementation [...]"

sbasmanov, 2020-05-26:
Same issue with my E3Pro, all stock, branch 2.0.x. Printing test pattern (from marlinfw site) with K=0..2 and 0.1 step produces good results (same width) on acceleration side (left) starting at K=1. But on entire range from 0 to 2 there is bulge on slow part on right. I think this issue is mostly with Bowden setups. When accelerating, it adds pressure to keep line width, but it is not retracting fast enough (or not enough volume) on slow down (because of Bowden). In my understanding, this happen because of same amount of filament is being pushed on acceleration and retracted on slow down. Probably, Bowden needs two separate K - one for acceleration, another for slowdown. May be this will work better.

XDA-Bam, 2020-05-26:
> I think this issue is mostly with Bowden setups. When accelerating, it adds pressure to keep line width, but it is not retracting fast enough (or not enough volume) on slow down (because of Bowden). [...]

From what I'm seeing, the slowdown is not the problem. If you measure it, you will see that the blob occurs _after_ the slowdown, when printing at constant speed in the slow last segment. At least that's what I observe on my prints.

I also did a couple of experiments some weeks ago, by altering `stepper.cpp` and `planner.cpp`:
- Force LA to reduce the advance steps 2 mm earlier: Line thins during deceleration, but blob on short segment is still present.
- Force LA to reduce advance steps twice as fast:  Line thins during decelration, but blob on short segment is still present.
- Increase or decrease `K` only for the deceleration phase: Various effects on line thickness, blob still present.

I'm not sure if the problem is sort of a delay, or if it rather is a hysteresis. I'm now looking into altering LA in the slow last segment, where the print head is essentially just cruising.

XDA-Bam, 2020-05-26:
To diagnose the problem further, I changed a couple of lines in `stepper.cpp` to slow down the increase in advance steps. [Lines 2273 - 2278](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/stepper.cpp#L2273) by default looks like this (comment omitted):
```
else if (step_events_completed < decelerate_after && LA_current_adv_steps < LA_max_adv_steps) {
  LA_steps++;
  LA_current_adv_steps++;
  interval = LA_isr_rate;
}
```
For a 2x slowdown, I changed it to
```
else if (step_events_completed < decelerate_after && LA_current_adv_steps < LA_max_adv_steps) {
  if (step_events_completed % 2 == 0) {
    LA_steps++;
    LA_current_adv_steps++;
  }
  interval = LA_isr_rate;
}
```
So we're only increasing `LA_steps` on every second step.

The resulting LA test pattern with the same GCode I used before looks like this:
[image]

Now there's two things to note, here:
1. Green ellipse 🟢 : All blobs after the slowdown are gone. The lines are not perfectly even, but damn close. Also, note that this is the coarse initial test pattern, and further tuning might result in even cleaner lines.
2. Dashed red ellipse 🔴 : As a side effect of the slower advance step increase, the faster segments now suffer in the beginning. This is expected and could be rectified by _only_ applying the slower advance increase after a decelration phase.

**Conclusions:**
- 80-90% of the blob is located in the slow segment (-> after a deceleration phase, not during that phase). Therefore, any fix should affect mainly the beginning of slow segments which follow on a deceleration.
- It looks like LA is increasing the advance steps too fast for segments which are preceeded by faster segments (-> after a deceleration).
- The actual cause may be a bug in the code I don't see, or it may be a mechanical oddity of bowden setups. However, direct drive users reported the same problem within this issue.

sbasmanov, 2020-05-26:
I'm not familiar with how LA works (yet), but from Your explanation I see that it's not that easy as I thought :) Anyway, for me it seems like this algorithm missing something from real world physics and this is why we have extra filament where we don't expect it. I just checked my test (from 0 to 2 with 0.1 step) and yes, all blobs appear right at the beginning of slow segment. Attached pic, sorry for poor quality.
[image]

> It looks like LA is increasing the advance steps too fast for segments which are preceeded by faster segments (-> after a deceleration).

So, may be this is where second K for deceleration is needed? Seems acceleration and deceleration need different advance for some reason. And I think for different setups, it will be different. So, fixed difference won't work.

sbasmanov, 2020-05-26:
Hmm, may be I understand wrong, but this part works when accelerating?

`
else if (step_events_completed < decelerate_after && LA_current_adv_steps < LA_max_adv_steps) {
  if (step_events_completed % 2 == 0) {
    LA_steps++;
    LA_current_adv_steps++;
  }
  interval = LA_isr_rate;
}
`

Shouldn't we modify this part instead? And may be decrease steps -=2 ? So we decrease advance twice faster?

`
if (step_events_completed > decelerate_after && LA_current_adv_steps > LA_final_adv_steps) {
        LA_steps--;
        LA_current_adv_steps--;
        interval = LA_isr_rate;
      }
`

XDA-Bam, 2020-05-26:
> 
> 
> Hmm, may be I understand wrong, but this part works when accelerating?
> 
> `else if (step_events_completed < decelerate_after && LA_current_adv_steps < LA_max_adv_steps) { if (step_events_completed % 2 == 0) { LA_steps++; LA_current_adv_steps++; } interval = LA_isr_rate; }`

Partly wrong: It is responsible for the acceleration **and** cruising phase. And the latter is the one we need to modify.

> Shouldn't we modify this part instead? And may be decrease steps -=2 ? So we decrease advance twice faster?
> 
> `if (step_events_completed > decelerate_after && LA_current_adv_steps > LA_final_adv_steps) { LA_steps--; LA_current_adv_steps--; interval = LA_isr_rate; }`

No. I tested and mendtioned this in [my previous post](https://github.com/MarlinFirmware/Marlin/issues/16880#issuecomment-634019249), altough I may not have stated it that clearly. It does not eliminate the blob.

Here is the result for a doubled `LA_steps--` & `LA_current_adv_steps--`:
[image]

It helps a bit (maybe 20% less volume to the blob), but it also creates a lack of filament right before the blob, at the end of the fast segment. The main problem is the beginning of the slow segment - so the beginning of a cruising phase.

NoxIX, 2020-05-27:
> * I originally suspected this to be a problem exclusive to bowden setups, but direct drive users reported the same bulges in this issue.

I will say, I converted my enders to direct drive to test LA, and while there may still be a bit of bulge in the test, (i've almost completely foregone using that test as a testing method) when running test towers comparing bowden to to direct drive, the direct drive towers act as one would assume. Corners bulge and Z seam bulge, then they smooth out into "sharp" corners and a, relatively, flat Z seam, and as the K value increases the corners and Z seam suck in. 

When running the same test on the bowden setup, the corners and Z seam do not change, *however*, artifacts will begin to appear. I.E., under-extrusion due to high K values. Please note, 0 corner and Z seam change was noted from 0 to K 1.65 on bowden. 

Apologies I can't help out with the firmware side of things, but hopefully that info is somewhat helpful.

sbasmanov, 2020-05-27:
XDA-Bam, I was sure that LA affect only acceleration and deceleration parts of segment. Now I see that it completely handles pressure advance on entire segment. So, now I reading source to get understanding how things work. But I have strong feeling, that after deceleration, at end of segment, there is some extra pressure left in nozzle. And this volume is being over extruded on slow part, in addition to normal amount. May be logging values of LA_* variables when segment is finished will shed some light? This is what I found: https://marlinfw.org/docs/features/lin_advance.html

> During the block runtime, the pressure will be advanced until the precalculated needed extra step amount is reached or the deceleration is started. During deceleration we reduce the advance step amount until we reach the amount for final speed or the block comes to an end. This checks are necessary as Marlin uses an approximation for acceleration calculation as mentioned above. Therefore, for example, we might not reach the final pressure at end of move perfectly but that’s not important as this error will not cumulate.

So, yes, it is not cumulating, but may be producing this bulb?

XDA-Bam, 2020-05-27:
> XDA-Bam, I was sure that LA affect only acceleration and deceleration parts of segment. Now I see that it completely handles pressure advance on entire segment. So, now I reading source to get [...]

Yes, there is a small amount of pressure left. You can see from [my tests](https://github.com/MarlinFirmware/Marlin/issues/16880#issuecomment-634200137), that if this pressure is reduced to near zero (either by reducing advance steps 2 mm earlier, or by reducing them twice as fast), the bulge is reduced a little bit. Maybe by 20%, maybe it's even 30%. But it is still there.

> And this volume is being over extruded on slow part, in addition to normal amount.

According to my tests, this is definietly not what's happening. Look at the results again - the bulge is still there, even if all pressure is bled from the nozzle before the slow segment begins. I am fairly certain that the problem is the increase of advance steps in the beginning of the slow segment. And reducing this increase in the slow segment has [eliminated the bulge](https://github.com/MarlinFirmware/Marlin/issues/16880#issuecomment-634059794). If that doesn't prove where the problem originates, I don't know what will.

But feel free to test yourself and hopefully you will come to the same conclusion. Four eyes do see more than two. 

> May be logging values of LA_* variables when segment is finished will shed some light? 

That will certainly be helpful.

>This is what I found: https://marlinfw.org/docs/features/lin_advance.html
>...
> So, yes, it is not cumulating, but may be producing this bulb?
> 
> At Your last test 0.6 line looks very good. And under extrusion at end of fast line happen because decreasing pressure advance two times faster than it should be for this part. It corresponds to 2xK for deceleration.

Yes, that's why the underextrusion happens. And at that point, right at the 60 mm mark, there will be very little pressure left in the nozzle - we're printing with zero advance at high speeds. And yet, the bulge is still there in the small segment!

> I feel that decrease speed of LA_steps should be slightly faster than increase, so when segment complete, there will be zero or slightly negative pressure in nozzle.

I agree somewhat: We will probably need to _also_ increase the reduction of advance steps while decelerating to totally fix our issue. But you can try - my tests show, that we won't get rid of the bulge this way. It helps, but only a small amount. We will also need to slow down the increase of advance steps directly after the deceleration, during the cruising phase.

sbasmanov, 2020-05-27:
Hmm... Here is our line:
G1 X85 Y37.5 E0.7982 F1200 ; print line
G1 X125 Y37.5 E1.5965 F4800 ; print line
G1 X145 Y37.5 E0.7982 F1200 ; print line
First (slow) segment starts from zero, and last segment starts from some non-zero speed? And LA doesn't take this into account and calculating advance steps like if we start from zero?

XDA-Bam, 2020-05-27:
Not sure if LA assumes that we start from zero speed. But `step_events_completed < decelerate_after` (in [this line](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/stepper.cpp#L2273)) is certainly true at the 60 mm mark (beginning of second slow segment). And `LA_current_adv_steps < LA_max_adv_steps` seems to be true, too. Otherwise, my fix/hack with the slower step increase wouldn't work.

I assumed, that `LA_current_adv_steps = LA_max_adv_steps` at the beginning of the second slow segment: The final speed at the end of the fast segment is exactly the same as the speed at the beginning of the slow one. Consequently, the number of advance steps should be the same, right? 🤔

Of yourse, that would only be true, if the current advance steps are carried over from one segment to the next. If they are forgotton each time the segment changes, `LA_current_adv_steps = 0` would be true at the beginning of the last slow one. Which is it?

sbasmanov, 2020-05-27:
Hmm.. that's good question. In my understanding, there should be no advance at all at beginning of second slow segment, as we are already at cruise speed and extruder feeds proper amount of filament. Here is how I see this:
[image]
First graph show how it really goes, and second how LA interprets it and pushing some extra filament on beginning of second slow segment like if it was starting from zero. I don't know (yet) if this is what happen, but this my suggestion.
So, seems You're right - there should be no advance at beginning of second slow segment.

XDA-Bam, 2020-05-27:
> 
> Hmm.. that's good question. In my understanding, there should be no advance at all at beginning of second slow segment, as we are already at cruise speed and extruder feeds proper amount of filament. 

Not 100% sure, but yeah. Looks like there's a problem.

> First graph show how it really goes, and second how LA interprets it and pushing some extra filament on beginning of second slow segment like if it was starting from zero. I don't know (yet) if this is what happen, but this my suggestion.

That's a nice way to visualize it.

sbasmanov, 2020-05-27:
So, this is what I see in planner:
`
            #if ENABLED(LIN_ADVANCE)
              if (block->use_advance_lead) {
                const float comp = block->e_D_ratio * extruder_advance_K[active_extruder] * settings.axis_steps_per_mm[E_AXIS];
                block->max_adv_steps = current_nominal_speed * comp;
                block->final_adv_steps = next_entry_speed * comp;
              }
            #endif
`

and in stepper:
`
      // Initialize the trapezoid generator from the current block.
      #if ENABLED(LIN_ADVANCE)
        #if DISABLED(MIXING_EXTRUDER) && E_STEPPERS > 1
          // If the now active extruder wasn't in use during the last move, its pressure is most likely gone.
          if (stepper_extruder != last_moved_extruder) LA_current_adv_steps = 0;
        #endif

        if ((LA_use_advance_lead = current_block->use_advance_lead)) {
          LA_final_adv_steps = current_block->final_adv_steps;
          LA_max_adv_steps = current_block->max_adv_steps;
          initiateLA(); // Start the ISR
          LA_isr_rate = current_block->advance_speed;
        }
        else LA_isr_rate = LA_ADV_NEVER;
      #endif
`
So, final_adv_steps of fast segment must be same as max_adv_steps of last slow segment. In this case no extra E steps will be executed. But how we can check if this is true?

sbasmanov, 2020-05-27:
Is there any place at stepper/planner where we can pull LA_current_adv_steps from previous block?

AnHardt, 2020-05-27:
> s there any place at stepper/planner where we can pull LA_current_adv_steps from previous block?

No.
`[LA_current_adv_steps](https://github.com/MarlinFirmware/Marlin/blob/8dbbcfd382356eb5d16b8bfa75c713bbc1de1631/Marlin/src/module/stepper.h#L327)` is a privat from stepper class.
Maybe you can save it to an new variable when leaving/deleting a block.

https://user-images.githubusercontent.com/10542322/83009295-344b4580-a01f-11ea-8d7a-ce9927cebe9d.png
The second graph is looking as if the third block is not properly connected to the second. This will (for example) happen when the second block is already stepped when the third is planned (planner buffer not full enough).

sbasmanov, 2020-05-27:
> is a privat from stepper class.
Maybe you can save it to an new variable when leaving/deleting a block.

Memory scope is always confuse me:( So, is it keeps its value since initialized at startup? Or being reset to zero each time? And if reset - when this happen? I can't find any assignment in code except declaration.

As to second graph - it is my vision of how advance is calculated. This blob looks like more E steps was executed at beginning of last segment as if it was starting from zero speed (with acceleration). So, it's not reflecting real motion :)

AnHardt, 2020-05-27:
```cpp
      #if ENABLED(LIN_ADVANCE)
        #if DISABLED(MIXING_EXTRUDER) && E_STEPPERS > 1
          // If the now active extruder wasn't in use during the last move, its pressure is most likely gone.
          if (stepper_extruder != last_moved_extruder) LA_current_adv_steps = 0;
        #endif
```
When setting up a block `LA_current_adv_steps` should have the value left over from the previous.

sbasmanov, 2020-05-27:
> When setting up a new block LA_current_adv_steps should have the value left over from the previous.

Yes, this what I thought. But here (https://marlinfw.org/docs/features/lin_advance.html) it says:

> Therefore, for example, we might not reach the final pressure at end of move perfectly but that’s not important as this error will not cumulate.

That means to me that `LA_current_adv_steps` is being reset every block. Or am I getting it wrong? There is no other place in code where it sets value of `LA_current_adv_steps`. So, how do I find out value of `LA_current_adv_steps` at block setup? If I properly understand how things work, at beginning of last slow segment `LA_current_adv_steps` must equal `LA_max_adv_steps` so no extra E steps executed. But not sure if this is true.

XDA-Bam, 2020-05-27:
>Yes, that's what I thought. But here (https://marlinfw.org/docs/features/lin_advance.html) it says:

Don't go by what the documentation says it should do. Go by what the code is actually doing 😉 

> 
> When setting up a new block `LA_current_adv_steps` should have the value left over from the previous.

In that case, it's probably best to check whether this is actually the case when printing the LA test pattern. So, as @sbasmanov suggested, we need to debug it while running. I'm still waiting for a cable to be able to set this up on my printer, so I can't help with that right now.

AnHardt, 2020-05-27:
My guess is that the error is not accumulating because the error is transferred to the new block via `LA_current_adv_steps`. In the new block that number of additional steps is done to reach the new value. (At least, i guess, that is the plan.)

sbasmanov, 2020-05-27:
I'm still waiting for skr board, as mine (original) behave very strange - after two-three times I update firmware it may stop booting at all, so I have to connect Pi, erase, flash bootloader and only after it will work again. So, I don't want to touch it while it works :)
But I think we can try to simulate this situation if we can figure out at block setup if we already at cruise speed and there is no need to accelerate. If this is possible, we can put 'LA_current_adv_steps=LA_max_adv_steps' right after setup, after line 2152: https://github.com/MarlinFirmware/Marlin/blob/6c994002af1ef13004490b5e15501e6cdc70dd88/Marlin/src/module/stepper.cpp#L2152

sbasmanov, 2020-05-27:
> In the new block that number of additional steps is done to reach the new value. (At least, i guess, that is the plan.)

That looks like real. So, probably in case when no acceleration is needed and head already at cruising speed, we don't want these extra steps to be executed. Am I right? That means 'LA_current_adv_steps==LA_max_adv_steps' must be true.

XDA-Bam, 2020-05-27:
> 
> My guess is that the error is not accumulating because the error is transferred to the new block via `LA_current_adv_steps`. In the new block that number of additional steps is done to reach the new value. (At least, i guess, that is the plan.)

I think it doesn't accumulate, because we increase `LA_current_adv_steps` until we reach `LA_max_adv_steps` in _each_ block. At that point, the error will be zero again. Maybe we're saying the same thing, not sure.

However, regardless of the error we start out with at the beginning of a block (which is only non-zero, if `LA_final_adv_steps` has not been reached), the code always increases `LA_current_adv_steps` until the desired value is reached. So, if `LA_final_adv_steps` has not been reached in the last block, `LA_current_adv_steps` will be too high in the current one - and we will also begin to increase it right away. In that case, our advance will be consistently too high from some point in the deceleration phase of the last block, until a couple of steps _after_ we reach `LA_max_adv_steps` in our current block. That's not good.
I assume that the opposite case - `LA_final_adv_steps` being undercut and `LA_current_adv_steps` being too low at the start of the next block - should never happen. At least I don't currently see how it could.

sbasmanov, 2020-05-27:
>So, if LA_final_adv_steps has not been reached in the last block, LA_current_adv_steps will be too high in the current one - and we will also begin to increase it right away.

It will not increase as it will increment only when `LA_current_adv_steps < LA_max_adv_steps` and `LA_max_adv_steps ` should be same as `LA_final_adv_steps ` from previous block.

Anyway, I think that value of `LA_current_adv_steps` is not root of problem. Remember, when You decreased it twice faster? It was definitely reaching `LA_final_adv_steps` before end of block. And that didn't affected blob significantly. That means, even if `LA_current_adv_steps ` reached `LA_final_adv_steps` and next block have same value of `LA_max_adv_steps` (so no extra E steps), there is still too much pressure in nozzle. And this is where I think that pressure should go lower than needed in next block. Here is what I mean:
[image]
May be we should adjust final_adv_steps in planner, so it is less than max_adv_steps of next block?
```
              if (block->use_advance_lead) {
                const float comp = block->e_D_ratio * extruder_advance_K[active_extruder] * settings.axis_steps_per_mm[E_AXIS];
                block->max_adv_steps = current_nominal_speed * comp;
                block->final_adv_steps = next_entry_speed * comp;
```
For test it could be:
```
if(next_entry_speed < current_nominal_speed) block->final_adv_steps = next_entry_speed * comp * 0.5;
else block->final_adv_steps = next_entry_speed * comp;
```

At least we will see how it will affect blob (and where).

XDA-Bam, 2020-05-27:
> It will not increase as it will increment only when LA_current_adv_steps < LA_max_adv_steps and LA_max_adv_steps should be same as LA_final_adv_steps from previous block.

True, in our case it should not increase it further. `LA_current_adv_steps` might still be too high (`>LA_max_adv_steps`), though. This was more of a general idea - it might not happen in our case, but it looks like a possible bug in LA.

> Anyway, I think that value of LA_current_adv_steps is not root of problem. Remember, when You decreased it twice faster? It was definitely reaching LA_final_adv_steps before end of block. And that didn't affected blob significantly.

True. But that only tells us that `LA_current_adv_steps = LA_final_adv_steps` is reached at the end of the fast block. It does not tell us what happens to `LA_current_adv_steps` at the beginning of the second slow block. I still think that our problem originates there.

> That means, even if LA_current_adv_steps reached LA_final_adv_steps and next block have same value of LA_max_adv_steps (so no extra E steps), there is still too much pressure in nozzle.

But my tests also show, that `LA_current_adv_steps` seems to be increased at the beginning of the last slow block, because halving the increase rate got rid of the blobs 🤔  It shouldn't happen - you're right. But it seems to me, that it does happen nonetheless. Maybe you should test it, too. Maybe there was a problem with my machine or test method.

sbasmanov, 2020-05-27:
>True. But that only tells us that LA_current_adv_steps = LA_final_adv_steps is reached at the end of the fast block. It does not tell us what happens to LA_current_adv_steps at the beginning of the second slow block. I still think that our problem originates there.

If I understand properly, Stepper class is not initialized on start of each block. It means that `LA_current_adv_steps` keeps its value where it was at end of previous block. And look at planner, how it calculates values for current block:
```
                block->max_adv_steps = current_nominal_speed * comp;
                block->final_adv_steps = next_entry_speed * comp;
```
So, for fast block, `final_adv_steps` will be same as `max_adv_steps` for slow block, when it will come to this point at planner. So, here will be no additional E steps.

> But my tests also show, that LA_current_adv_steps seems to be increased at the beginning of the last slow block, because halving the increase rate got rid of the blobs 🤔 It shouldn't happen - you're [...]

Well, we don't know if it increasing or not until we log this values. Now we can only guess :) Here is what I think about this: when accelerating, we increase E speed in advance, that means more filament coming to extruder. To handle this, more heat is needed to melt increased amount of filament. This raises nozzle temperature to compensate this. When we slowing down, we lowering extrusion speed, but nozzle (and filament) can't cool down with same speed. So, extra filament still coming from nozzle even if we are at right (slower) extruder speed. What I'm trying to say, that K for advance when accelerating and K when decelerating is not same. We should reduce extrusion amount slightly more than we added in advance on acceleration. I think it will be small difference in K1,K2 on direct setup, but significant on Bowden.
And there is another caveat - isr interval is calculated at planner for entire block. If we want to try decreasing `LA_current_adv_steps ` with different speed (compared to how it was increased), we should adjust this interval as well, to let isr count down to `LA_final_adv_steps'. 

ps: I'll definitely do my tests as soon as I get my skr board. As I mentioned before, my creality board behaves very strange, so I don't want to brick it now, when I don't have replacement. So, if You have time for tests, try to change `final_adv_steps` value calculation as I proposed above?

XDA-Bam, 2020-05-27:
> So, for fast block, final_adv_steps will be same as max_adv_steps for slow block, when it will come to this point at planner. So, here will be no additional E steps.

I agree that this should be the case. That's what I would expect seeing the code. Also agree that we need to debug to see if it is as expected when actually printing.

>  What I'm trying to say, that K for advance when accelerating and K when decelerating is not same. We should reduce extrusion amount slightly more than we added in advance on acceleration. I think it will be small difference in K1,K2 on direct setup, but significant on Bowden.

Mostly agree. Not sure if this will "cure" the blobs, but we will see.

> So, if You have time for tests, try to change final_adv_steps value calculation as I proposed above?

Tested it a couple of weeks ago. I forced `K` +-0.4 and +-0.5 (more or less randomly chosen) for the deceleration phase only. Both changes seemed to make the blobs worse and had unwanted side effects (slow lines super thin or super thick). I'm not totally happy with the tests though. Might repeat them in the coming days.

The 'LA_steps-=2;'-stuff is also something I tested and which resulted in [this image](https://github.com/MarlinFirmware/Marlin/issues/16880#issuecomment-634200137). So, a little better but not solved.

Looking forward to your new board and my damn cable 😀

sbasmanov, 2020-05-27:
>The 'LA_steps-=2;'-stuff is also something I tested and which resulted in this image. So, a little better but not solved.

Seems twice faster is too much. We need to decrease LA_steps with something like x0.8 of normal speed or fire isr more frequently. But this will have effect only with lowering of LA_final_adv_steps for fast segment, so at end we remove little more pressure than needed. And because LA_current_adv_steps will be low as well, it will start incrementing at beginning of slow segment to get to proper level.
Anyway, my board arrive in two weeks (I hope), so there is enough time to think :)

XDA-Bam, 2020-05-29:
Got the cable and did some initial testing. This is what `LA_current_adv_steps` looks like for each call of `Stepper::advance_isr` during printing a pretty typical test pattern (20/40/20 mm segment length, 20 mm/s slow, 80 mm/s fast, accel. 750 mm/s², K=0 - 2.0 in 0.2 steps):
[image]

The two long plateaus at the beginning should be the nozzle purging / priming. Notice that there are 11 calibration lines (K=0 to 2.0), but only 10 peaks follow - I guess K=0 does not produce any changes to `LA_current_adv_steps`, which kind of makes sense. Other than the fact that the lowest `LA_current_adv_steps` reached inbetween each of the calibration lines increases continously, it looks good to me.

Now, investigating one peak in more detail. Let's pick the second one after the plateaus, which should be K=0.4:
[image]

I have to say - this is more boring than expected 😄 Looks to me as it should. No unexpected increases, no bumps or drops, all good.

The value of the low plateaus before and after the peak, corresponding to the slow segments, are both "29". Therefore,  `LA_current_adv_steps` seems to reach `LA_final_adv_steps` just fine. It's also pretty clear that it reaches `LA_max_adv_steps` for the first couple of peaks and probably doesn't for the later ones - there is not visible plateau at the top of those.

sbasmanov, 2020-05-29:
Oh, such useful information. Many thanks. I expected something similar :) While I still digging back & forth in code to get understanding in details, here is what I found so far:
[image]
So, `pulse_phase_isr` is just altering `LA_steps` and not doing actual pulses. Next, `advance_isr` is called and this is where it fires all pending pulses. In case we're accelerating/decelerating it additionally alters `LA_steps` adding/removing extra pulses. While cruising, `LA_steps` altered only at `pulse_phase_isr`.
Ok, another thing. There is `advance_speed` that is being calculated using K (and lot of other stuff). And according to this speed, `advance_isr` frequency is set. And of course, this speed is absolutely same for acceleration and deceleration. And this is where I think problem comes from: on acceleration we fed a lot of filament in Bowden tube and now, when we lowering speed, this filament straightens up which is resulting in that bulge. I'm still thinking we could try setting `final_adv_steps` to x0.8 (with `if(next_entry_speed < current_nominal_speed)` ) value and see if `LA_current_adv_steps` will count down to it.  https://github.com/MarlinFirmware/Marlin/issues/16880#issuecomment-634761609

ps: what cable, board and software you're using to get values from running board? I definitely need same :)

sbasmanov, 2020-05-29:
>It's also pretty clear that it reaches LA_max_adv_steps for the first couple of peaks and probably doesn't for the later ones - there is no visible plateau at the top of those.

Yes, with higher K it starts going down without reaching cruise value. With high enough K it starts retracting (LA_steps becomes negative).

[101 more replies]

closed 2020-08-13 by github-actions[bot]