MarlinFirmware/Marlin #16883: [BUG] Print end and send M84. Not turn off all motor (Extruder is on)

DJDOUBLEG, 2020-02-17:
I have SKR pro v1.1 with Marlin 2.0.3, I downloaded it from here. (https://github.com/MarlinFirmware/Marlin/archive/2.0.x.zip)

My end code last code  is:
M84 ; Disable stepper motors
It turned off X Y and Z except E0.
And (#define USE_CONTROLLER_FAN) still runing the case fan.
If i go to motion menu and disable all steppers then same issue. 
Teminal command (Same issue):
Send: M84
Recv: ok

Thank you!

### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

theWaldschrat, 2020-02-20:
I can confirm the same behavior on BTT SKR Mini E3.
Neither M84 nor disabling steppers via menu works for the extruder.

DJDOUBLEG, 2020-02-20:
:(

theWaldschrat <notifications@github.com> ezt írta (időpont: 2020. febr.
20., Cs 20:49):

Ravenor, 2020-02-20:
I have the same problem on BTT SKR V1.4 turbo

theWaldschrat, 2020-02-21:
Duplicate of #16753

Looks like the same has been discussed in a different issue. The problem has been found and fixed in 2.0.4 as far as I can see.
I cannot test it in the moment because my printer is running a long job, but I will later.

miniskipper, 2020-02-26:
Have built master yesterday and can confirm the bug is fixed. Board: skr mini e3 v1.2

closed 2020-02-27 by thinkyhead