MarlinFirmware/Marlin #16884: [BUG] Serial Connection inconsistent

kurtisane, 2020-02-17:
Hi guys,
I have a SKR PRO v1.1 and I have huge Problems with my Serial Connection to my board. Mid print it starts to stutter and eventually it just stops since there is no serial connection possible anymore. First I thought it must be a OctoPrint Issue therefore I tried printing with the TFT35 v2.0 with a SD Card but I have the same result. 
I lowered the Baudrate to 115200 and tried again - Same Result. 

Any advice on how to Debug this ? I'm not sure if this is a Marlin Issue or a Setup thing ( I did not change anything on my setup except the marlin firmware). Right now I'm on the Marlin 2.0.x branch from 16.02.2020.

Regards !

skipfie, 2020-02-25:
Are you flashing it with USB enabled?

BastR, 2020-04-28:
I print with octoprint on a SKR PRO and I don't have this problem. On the other hand, I think a lot of people have problems with the TFT35, look on the GitHub Bigtreetech about the TFT35

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle