MarlinFirmware/Marlin #16886: [BUG] Probe not deployed on G28 when using Z_MIN_PROBE_PIN

szopen111, 2020-02-17:
### Bug Description

The probe doesn't deployed on `G28` when the end-stop signal (white wire) is connected to the SCK pin (ICSP header) and the pin is set using `#define Z_MIN_PROBE_PIN 7`.

_(a workaround for this bug is at the end of the post)_

`M401` and `M402` are working. `M119` correctly reports probe end-stop triggered.

Hardware: BL-Touch-like probe connected to Creality Ender 3 Pro with stock Creality Melzi Board (pins_MELZI_CREALITY.h).

#### Rationale
The original end-stop is still connected to the original connector. It was intended as a fail-safe when testing the new probe - set to trigger just below the probe triggers to prevent the hot-end ramming into the bed if something fails.

I don't want to use the original end-stop socket because of the capacitor. I intend to use it later to connect a filament run-out sensor.

### My Configurations

```
#define USE_ZMIN_PLUG // original end-stop connected
#define Z_MIN_ENDSTOP_INVERTING false

#define Z_MIN_PROBE_ENDSTOP_INVERTING true
// #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_PIN 7

#define BLTOUCH
```

### Steps to Reproduce

Connect the probe and change the config to match the above.

**Expected behavior:** The probe deploys before homing Z-AXIS

**Actual behavior:** The probe doesn't deploy

#### Additional Information

I think this behavior may be by design, but it was not clear to me.

The root cause is the following lines in `Conditionals_LCD.h`:
```
#define HAS_CUSTOM_PROBE_PIN  DISABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
#define HOMING_Z_WITH_PROBE   (Z_HOME_DIR < 0 && !HAS_CUSTOM_PROBE_PIN)
```
It seems that homing with z probe deployed is possible only if "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN" is set.

#### Suggested changes
 * **if this is by design**, please add a more verbose comment for Z_MIN_PROBE_PIN, to indicate it disables the probe when homing and uses the Z_STOP_PIN. An additional suggestion to re-define the Z_STOP_PIN to get the desired behavior can save other users some time. 
 * add `FORCE_PROBE_DEPLOY_WHEN_Z_HOMING` to deploy the probe despite HAS_CUSTOM_PROBE_PIN is set

#### Solution
To make my setup working I had to change the `pins_MELZI_CREALITY.h` file:
```
#include "pins_MELZI.h"

// Change the end-stop pin to SKC pin in the ICSP header (Arduino pin 7)
#undef Z_STOP_PIN
#define Z_STOP_PIN 7
```
and use the following config:
```
#define USE_ZMIN_PLUG
#define Z_MIN_ENDSTOP_INVERTING true
#define Z_MIN_PROBE_ENDSTOP_INVERTING false
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

#define BLTOUCH
```

boelle, 2020-06-20:
@hamster65 is this still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle