MarlinFirmware/Marlin #16887: [BUG] EEPROM Datasize error when Extruders = 0 (fix included)

fab4fun, 2020-02-17:
### Bug Description

Setting extruders = 0,  while appearing to not be officially supported, works well except for an EEPROM datasize error when trying to write. 

### My Configurations

MPCNC machine with dual X and Y drivers (X2, Y2) and Z axis  
X2 and Y2 use E0 and E1 in pins.h
TMC2130 with sensorless homing enabled on both X, X2, Y, Y2
Ramps Board (MKS GEN L V2.0)

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4214730/Configurations.zip)

This bit of code is holding a place for at least 1 setting, even if extruders set to 0:
      //
      // Linear Advance
      //
      {
        float extruder_advance_K[_MAX(EXTRUDERS, 1)];
        _FIELD_TEST(planner_extruder_advance_K);
        EEPROM_READ(extruder_advance_K);
        #if ENABLED(LIN_ADVANCE)
          if (!validating)
            COPY(planner.extruder_advance_K, extruder_advance_K);
        #endif
      }

Expect EEPROM hold place for at least 1 extruder setting 
object declaration needs to match

Patch required:

-(326)  float planner_extruder_advance_K[EXTRUDERS];          // M900 K  planner.extruder_advance_K
+(326)   float planner_extruder_advance_K[_MAX(EXTRUDERS, 1)];          // M900 K  planner.extruder_advance_K

fab4fun, 2020-03-11:
any updates?  I thought this would be easy for someone to integrate

thinkyhead, 2020-03-11:
Thanks! The issues are coming in so fast that they get lost if we don't read 50 a day…. Not that we mind. I'll merge the fix shortly.

thinkyhead, 2020-03-11:
Hmm… Looks like it was merged on Feb 21.

closed 2020-03-11 by thinkyhead

fab4fun, 2020-03-11:
Thanks for checking...  the updates are also coming in very fast so I missed the merge!