MarlinFirmware/Marlin #16888: [BUG] (short description)

magicmellon, 2020-02-17:
Done

### Bug Description

I am making a testing rig. It is simply a stepper motor (nema 17) connected to a controller (ESP32 Wroom 32D board) through a stepper driver (TMC2130).

As I was unable to "simply" remove the x y and z axis from my marlin config, so they and their respective endstops have been set to use the pin -1.

I am still in testing phase of the project so have disabled thermal protection etc (as i currently dont have that circuit to connect one up). The motor wiring has been done on a breadboard, and is a very close copy to a RAMPS board. It has been tested good using the TMC2130 library and a simple sketch, showing SPI working and movement working.

I originally started this project in marlin 2.0.1 but have just transfered my pins, and basic config (board and thermal protection off) to the bugfix branch and was still able to reproduce.

**The actual bug**
1) motor only rotates in one direction
2) after a number of moves the esp32 keeps enabled low, but there is no holding current and the motor freewheels. This also sometimes happens a move or so after a negative move.

Also to note:
I get a brown-out and watchdog reset on boot 
(unsure if related, but worth noting)

I have used an oscilloscope and found that once the system enters this freewheel state, enable is low but no signal comes through step to the driver.

Secondly, the TMCDebug changes between before and after entering this freewheel mode:

Before:

`		E
Enabled		false
Set current	800
RMS current	795
MAX current	1121
Run current	25/31
Hold current	12/31
CS actual	12/31
PWM scale	2
vsense		1=.18
stealthChop	true
msteps		16
tstep		max
PWM thresh.
[mm/s]
OT prewarn	false
off time	3
blank time	24
hysteresis
 -end		-1
 -start		1
Stallguard thrs	0
DRVSTATUS	E
sg_result	0
stallguard
fsactive
stst
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:
		E	0x80:0C:00:00
Testing E connection... OK`

After:

`		E
Enabled		false
Set current	800
RMS current	1436
MAX current	2025
Run current	25/31
Hold current	12/31
CS actual	0/31
PWM scale	0
vsense		0=.325
stealthChop	false
msteps		256
tstep		max
pwm
threshold
[mm/s]
OT prewarn	false
OT prewarn has
been triggered	false
off time	0
blank time	16
hysteresis
-end		-3
-start		1
Stallguard thrs	0
DRVSTATUS	E
stallguard
sg_result	0
fsactive
stst		*
olb		*
ola		*
s2gb
s2ga
otpw
ot
Driver registers:
		E	0xE0:00:00:00
Testing E connection... OK
`

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
Attached these from the bugfix build which I have produced the results on + the pins file as im doing weird stuff with that too.

[MarlinBugReportInfo.zip](https://github.com/MarlinFirmware/Marlin/files/4214972/MarlinBugReportInfo.zip)

### Steps to Reproduce

1. connect printer to pronterface
2.G1 E10
3.G1 E-10
4.E1 E10

**Expected behavior:** [What you expect to happen]
rotation in one direction, then the other, then back to the original

**Actual behavior:** [What actually happens]
The motor only rotates in one direction.
Freewheeling occurs, sometimes after 3, sometimes after 4, sometimes a few more moves after 3, sometimes but rarely never. In this state no signals are going to the motor, enable is staying low, but no holding current to the coils.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

No Additional information provided at the moment, but can at request.

vivian-ng, 2020-02-18:
On brownout and reset, is there a core dump of the stack? If so, please use the [tool here](https://github.com/me-no-dev/EspExceptionDecoder) to decode the stack, and post the results. This will help to provide more information about where the error occurred.

On my printers (Ender-3 and CoreXY) running on [MRR ESPE](https://github.com/maplerainresearch/MRR_ESPE) (ESP32 board with I2S stepper stream), I am using TMC2130s and TMC2209s on X and Y axes with no issues. I have not tested on extruder yet. If you can wait, I will find time over the weekend to test out on my CoreXY build (which is currently running on TMC2130s).

(The stack decode tool only works in Arduino IDE for the moment, unfortunately. Also, you will need a copy of the .elf file that was compiled when building the firmware for that particular build.)

Also, for gcode, maybe try:
G90 ; set absolute positioning
G92 E0 ; set E to 0 position
G1 E10 ; extrude 10mm
G1 E-10 ; retract 10mm

magicmellon, 2020-02-19:
Hey @vivian-ng im not getting a core dump - just brownouts then moves on with the start up sequence. I have been using relative positioning as well as absolute but I havent tried to explicitly call out absolute positioning, nor have i "homed" the e-axis. will give this a try later on

magicmellon, 2020-02-20:
Have just tried your process to no avail.

to elaborate on the brown-out part. Here is the output from marlin on boot:

> echo: External Reset
 Brown out Reset
 Watchdog Reset
Marlin Deathrig Marlin 2.0.1
echo: Last Updated: 2019-12-24 | Author: https://www.reddit.com/user/ascii_dickbutt
echo:Compiled: Feb 20 2020
echo: Free Memory: 304912  PlannerBufferBytes: 1280
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm (mm)
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo: M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:Stepper driver current:
echo:  M906 T0 E800
echo:Driver stepping mode:
echo:  M569 S1 T0 E
echo:Filament runout sensor:
echo:  M412 S1
Testing E connection... OK
Printer is now online.

vivian-ng, 2020-02-20:
@magicmellon 
I tested with the following:
Test 1:
Board: MRR ESPE
Stepper drivers: X: TMC2130 SPI; Y: TMC2130 SPI, Z: A4988; E0: TMC2130 SPI
CS pins: X: 21; Y: 22; E0: 4
With this setup, I did a test print of a test cube on a coreXY printer. I managed to print without any issues (I stopped the print about 1/4 of the way in because it was printing with no problem).

Test 2:
Board: MRR ESPA
Stepper drivers: X: TMC2130 SPI; Y: TMC2130 SPI, Z: A4988; E0: TMC2130 SPI
CS pins: X: 21; Y: 22; E0: 0
This setup was not physically connected to any printer. All I did was connect E0 to a stepper motor. With this setup, I was able to send `G92`, then a series of `G1 E10` and `G1 E0` without any problems. The motor turned in the right direction each time. Or at least, the motors changed direction when it should. No brownout.

(Pins file for the above two boards, MRR ESPE and MRR ESPA, can be found in the pins/ESP32 folder of Marlin.)

A few suggestions:
1. Brownout without stack dump suggests a hardware problem. If there is a firmware problem, there will usually be a dump of the stack. One common cause of brownouts is power. Make sure the power supply you are using is capable of feeding a stable power supply of sufficient current to the ESP32.
2. Download a fresh copy of Marlin bugfix-2.0.x and try again. Your configuration files seem to be from an older version of Marlin. Maybe you can try using a newer copy of the configuration files.
3. If you are not connected to a real thermistor and hotend, remember to comment out PREVENT_COLD_EXTRUSION so that you can turn the E0 motor at any temperature.
4. Your pins file suggests you are using pin 38 and 36 for serial (E0 serial). I think you have these pins because you intend to test TMC2208 or TMC2209 in the future. However, I am not sure if these are the best pins to use. Pin 36 is also defined as the E0 temperature pin in your pins file, if I recall, so there is also potential conflict there.
5. Check your wiring to make sure everything is being connected correctly. I had many experiences of trying to hunt down a "bug" until I realized that it was wrong wiring on my part.
6. Test with X (or Y or Z) axis. Even if you don't have endstops, it is okay, you can still use `G1 X10` and `G1 X0` to move the X motor.

Without more information about your exact setup (pictures, schematics, wiring diagrams, hardware, etc.) this is the best I can offer. Personally, I think it is more likely a hardware/wiring issue than a Marlin firmware issue.

magicmellon, 2020-02-25:
Appreciate the help Vivian. I have decided that the circuit (what im currently testing) is good, and its an issue with the bread board or my marlin config. This conclusion has been made as my set up works fine on test movement code using the tmc2130 driver library.

Thanks for the heads up on the pins conflict, missed that one.

I am going to investigate the brownout further, but for now i will close the issue and thank you for your help and time.

closed 2020-02-25 by magicmellon