MarlinFirmware/Marlin #16889: [BUG] Babystepping does not work on SKR 1.3/LPC1768

thisiskeithb, 2020-02-17:
### Bug Description

After a recent hotend rebuild on an SKR 1.3/`LPC1768` printer, I had to redo my BLTouch/probe offsets & I noticed that babystepping was no longer working. I tried enabling `INTEGRATED_BABYSTEPPING` as well as disabling software endstops completely (already disabled on Z) with no success.

Reverting to `2.0.3` with the same config restores babystepping capability.

After this print is done, I will backtrack through these commits and see if one in particular breaks babystepping & I'll test on some other boards/MCUs:

- EXPERIMENTAL integrated BABYSTEPPING (64a81f887a3b9fd471484dba784e11bdba88ffbb)
- Clean up stepper and babystep (0b984519c30f453fdb5b4efb8008a07c69688824)
- Use prior babystep delay method (189c101793c72780fdff2b8cd693d4a53040a0e8)
- Unify step pulse timing of ISR / babystep (2836834d7ef2043ccd80b8a76560c21f2eb8fe74)

### My Configurations

[broken-babystepping.zip](https://github.com/MarlinFirmware/Marlin/files/4215457/broken-babystepping.zip)

### Steps to Reproduce

1. Compile Marlin based on 75e3aa1 with the above config
2. Observe that babystepping has no effect
3. Use the same config on top of Marlin `2.0.3` & observe that babytepping works (change `BOARD_BTT_SKR_V1_3` to `BOARD_BIGTREE_SKR_V1_3` since it was renamed)

**Expected behavior:** Babystepping to work

**Actual behavior:** Babystepping does not work

thisiskeithb, 2020-02-17:
After testing the related commits above, "Clean up stepper and babystep" (#16857)/https://github.com/MarlinFirmware/Marlin/commit/0b984519c30f453fdb5b4efb8008a07c69688824 is what broke babystepping on the SKR 1.3/`LPC1768`.

thisiskeithb, 2020-02-18:
I just rebased my MK3/BTT002 (`STM32F407VET6`)/TMC2209 config on https://github.com/MarlinFirmware/Marlin/commit/75e3aa12c5e963345c533d8f191aba207bb4d2c1 and babystepping does not work either, so this is likely not just SKR 1.3/`LPC1768` related.

rado79, 2020-02-18:
Confirmed. SKR1.3 TMC2208/9

rado79, 2020-02-18:
https://github.com/MarlinFirmware/Marlin/issues/16617#issuecomment-587855566

Scope666, 2020-02-20:
Also confirmed babystep not working with/without INTEGRATED_BABYSTEPPING. SKR1.3 TMC2209

hamster65, 2020-02-21:
My Arduino Mega with Ramps setup is also affected. Z Babystepping is not possible, nothing happens.

thisiskeithb, 2020-02-21:
Since `2.0.4` was released today, I tried the latest `bugfix-2.0.x` (5d16497), but babystepping is still broken.

Scope666, 2020-02-21:
Shouldn't 2.0.4 be held back until Live Z is fixed?   That's kind of a big deal.  Pretty much can't use a probe without live Z changes.

thisiskeithb, 2020-02-21:
> Shouldn't 2.0.4 be held back until Live Z is fixed? That's kind of a big deal. Pretty much can't use a probe without live Z changes.

It'll get fixed soon, I'm sure. Scott's been busy improving [Marlin Auto Build](https://marketplace.visualstudio.com/items?itemName=MarlinFirmware.auto-build) this week.

thinkyhead, 2020-02-22:
Thanks for keeping on this. I have been otherwise engaged. The issue should be patched now. Please confirm that it also works for you. Note that there is a known issue with babystepping on SKR boards with 2209 steppers where the stepper drivers can turn off. We're still experimenting to find a decent solution to that problem.

closed 2020-02-22 by thinkyhead

thisiskeithb, 2020-02-22:
> Thanks for keeping on this. I have been otherwise engaged. The issue should be patched now. Please confirm that it also works for you. Note that there is a known issue with babystepping on SKR [...]

@thinkyhead: I can't get it to compile:

```prolog
Marlin/src/module/stepper.cpp:2590:73: error: macro "BABYSTEP_AXIS" passed 4 arguments, but takes just 3
 2590 |           BABYSTEP_AXIS(Z, BABYSTEP_INVERT_Z, direction, (CORESIGN(1)<0));
      |                                                                         ^
Marlin/src/module/stepper.cpp:2512: note: macro "BABYSTEP_AXIS" defined here
 2512 |     #define BABYSTEP_AXIS(AXIS, INV, DIR) do{           \
      | 
Compiling .pio/build/LPC1768/src/src/sd/SdFile.cpp.o
Compiling .pio/build/LPC1768/src/src/sd/SdVolume.cpp.o
Compiling .pio/build/LPC1768/src/src/sd/cardreader.cpp.o
Compiling .pio/build/LPC1768/src/src/sd/usb_flashdrive/Sd2Card_FlashDrive.cpp.o
Compiling .pio/build/LPC1768/src/src/sd/usb_flashdrive/lib-uhs2/Usb.cpp.o
Compiling .pio/build/LPC1768/src/src/sd/usb_flashdrive/lib-uhs2/masstorage.cpp.o
Compiling .pio/build/LPC1768/src/src/sd/usb_flashdrive/lib-uhs2/message.cpp.o
Compiling .pio/build/LPC1768/src/src/sd/usb_flashdrive/lib-uhs2/parsetools.cpp.o
Compiling .pio/build/LPC1768/src/src/sd/usb_flashdrive/lib-uhs2/usbhost.cpp.o
Compiling .pio/build/LPC1768/lib0d2/Servo/Servo.cpp.o
Compiling .pio/build/LPC1768/libaba/LiquidCrystal/LiquidCrystal.cpp.o
Compiling .pio/build/LPC1768/lib839/U8glib-HAL_ID1932/U8glib.cpp.o
Compiling .pio/build/LPC1768/lib839/U8glib-HAL_ID1932/U8glibPrint.cpp.o
Compiling .pio/build/LPC1768/lib839/U8glib-HAL_ID1932/clib/chessengine.c.o
Compiling .pio/build/LPC1768/lib839/U8glib-HAL_ID1932/clib/u8g_bitmap.c.o
Compiling .pio/build/LPC1768/lib839/U8glib-HAL_ID1932/clib/u8g_circle.c.o
Marlin/src/module/stepper.cpp: In static member function 'static void Stepper::do_babystep(AxisEnum, bool)':
Marlin/src/module/stepper.cpp:2590:11: error: 'BABYSTEP_AXIS' was not declared in this scope
 2590 |           BABYSTEP_AXIS(Z, BABYSTEP_INVERT_Z, direction, (CORESIGN(1)<0));
      |           ^~~~~~~~~~~~~
```