MarlinFirmware/Marlin #16892: [BUG] build: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value

sl1pkn07, 2020-02-18:
### Bug Description

wen build:

~~~
Compiling .pio/build/LPC1768/src/src/sd/SdBaseFile.cpp.o
Marlin/src/sd/SdBaseFile.cpp: In member function 'bool SdBaseFile::open(SdBaseFile*, const uint8_t*, uint8_t)':
Marlin/src/sd/SdBaseFile.cpp:652:17: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
  652 |       dateTime_(&p->creationDate, &p->creationTime);
      |                 ^~~~~~~~~~~~~~~~
Marlin/src/sd/SdBaseFile.cpp:652:17: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
Marlin/src/sd/SdBaseFile.cpp:652:35: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
  652 |       dateTime_(&p->creationDate, &p->creationTime);
      |                                   ^~~~~~~~~~~~~~~~
Marlin/src/sd/SdBaseFile.cpp:652:35: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
Marlin/src/sd/SdBaseFile.cpp: In member function 'bool SdBaseFile::sync()':
Marlin/src/sd/SdBaseFile.cpp:1437:17: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
 1437 |       dateTime_(&d->lastWriteDate, &d->lastWriteTime);
      |                 ^~~~~~~~~~~~~~~~~
Marlin/src/sd/SdBaseFile.cpp:1437:17: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
Marlin/src/sd/SdBaseFile.cpp:1437:36: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
 1437 |       dateTime_(&d->lastWriteDate, &d->lastWriteTime);
      |                                    ^~~~~~~~~~~~~~~~~
Marlin/src/sd/SdBaseFile.cpp:1437:36: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
~~~

### My Configurations

https://github.com/sl1pkn07/Marlin/blob/sL1_Changes_ReARM/Marlin/Configuration.h
https://github.com/sl1pkn07/Marlin/blob/sL1_Changes_ReARM/Marlin/Configuration_adv.h

### Steps to Reproduce

1.: download sources
2.: configure marlin 
3.: run `pio run`

**Expected behavior:** [What you expect to happen]

build without warnings

**Actual behavior:** [What actually happens]

see description

#### Additional Information

a boy in a telegram chat he says have problems with read the SD card. maybe this warning is about the problem. for me works in my ReARM+Ramps

ellensp, 2020-02-18:
Warning confirmed.

Moyster, 2020-02-18:
Looks like a compiler update "false positive" as the sd code looks okay, these warnings can be squashed by adding them to the ignore list :  
in `makefile:700` replace : 
`LIBWARN =  -w -Wno-packed-bitfield-compat`
```
LIBWARN = \
    -w -Wno-packed-bitfield-compat \
    -w -Wno-address-of-packed-member
```

sjasonsmith, 2020-02-18:
Squashing this warning seems like a bad idea to me. I’ve seen unaligned struct member cause Marlin crashes in the past. Maybe the SD code is ok (I’m not in a position to look at it right now), but in other cases this warning is very valuable.

Moyster, 2020-02-18:
maybe something like this then :  
```
@@ -647,9 +647,11 @@
     memcpy(p->name, dname, 11);

     // set timestamps
+    u_int16_t creaDate = p->creationDate;
+    u_int16_t creaTime = p->creationTime;
     if (dateTime_) {
       // call user date/time function
-      dateTime_(&p->creationDate, &p->creationTime);
+      dateTime_(&creaDate, &creaTime);
     }
     else {
       // use default date/time
@@ -1433,8 +1435,10 @@
     d->firstClusterHigh = firstCluster_ >> 16;

     // set modify time if user supplied a callback date/time function
+    uint16_t lastWDate = d->lastWriteDate;
+    uint16_t lastWTime = d->lastWriteTime;
     if (dateTime_) {
-      dateTime_(&d->lastWriteDate, &d->lastWriteTime);
+      dateTime_(&lastWDate, &lastWTime);
       d->lastAccessDate = d->lastWriteDate;
     }
     // clear directory dirty
```
Not sure if that's the proper way to do but that's how other projects fixed similar warnings, advice ?

rado79, 2020-02-18:
Warning confirmed. There are "&" marks yellow underlined - compiled in VSCode, since i got  bugfix 9b4192cdd0755786a8d1892dfdffe3b875385681. There was also a VSCode update In the same period.

joe-ko, 2020-02-21:
here too with skr 1.4. Marlin Freeze sometimes since short time.
is there a solution ?

Hesi-Re, 2020-02-22:
I also get this waning!

thinkyhead, 2020-02-23:
Add this line to the top of `SdBaseFile.cpp`:
```cpp
#pragma GCC diagnostic ignored "-Waddress-of-packed-member"
```

closed 2020-02-23 by thinkyhead

thinkyhead, 2020-02-23:
The NXP platform updated GCC…
- Old: `toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)`
- New: `toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)`

Now patched in 1d431fe8d27b3b5c16333793c71408fe0156308b