MarlinFirmware/Marlin #16893: [BUG] "Printing..." never ends if the print was done over USB.

VanessaE, 2020-02-18:
### Bug Description

Prints done over USB never "finish" from Marlin's perspective.  That is to say, the object completes just fine, all the way through the last commands in my slicer's "end" g-code.  But, Marlin still thinks a print is going, i.e. it says "Printing..." on the display, instead of "xxxx ready", and the menus' contents are rearranged accordingly.

Here's the end g-code my prints normally use:

```gcode
G92 E0
G91
G1 E-2 F4800
G90
G92 E0
G1 X0 Y200 F15000
G1 Z272 F12000
M104 S0
M140 S0
M107
M84
```

This happens on two different printers, running `bugfix-2.0.x` at commits 0ebd2f44d4fc88948d941f3b16a08c027208dd6c and b26f7fca3fb085a502195319e04101b38d61dba1.

I am not sure when it broke, except that it was reasonably recent (say, within the past few months).

While writing this issue, I kept trying to find some way to "clear" this confused state, and found that clicking my custom "all off" button twice in a row (with a brief pause in between) seems to work.  That button executes these commands:

```gcode
M104 S0
M140 S0
M107
M84 XYE
```

But aside from the `XYE` (which my i3 needs, as this button needs to work for both machines), those commands are also present in the above-pasted "end" g-code and clearly should not need to be executed twice.

Normally, I'd just ignore this, but aside from it triggering my OCD :grin: it interferes with doing some things through the LCD menus, as some options are hidden from the user if Marlin thinks a print is in progress, primarily motion/jog controls.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4218761/Configuration.zip)

### Steps to Reproduce

1. Print something over USB -- anything at all.
2. Pause the print from the host (I'm assuming Pronterface here)
3. Wait.
4. ???
5. Profit! :grin:

**Expected behavior:**

Marlin should switch back to its "ready" state when the host stops sending data (other than e.g. `M105` or whatever else might be used to monitor the printer's state), or at least when it has been told to turn off its heaters and fans.

**Actual behavior:**

Marlin just stays in its "Printing..." state, even hours after a print ends and the entire machine has cooled back down to room temperature.

GMagician, 2020-02-18:
Maybe you need to use bugfix, see #16849

VanessaE, 2020-02-18:
b26f7fca3fb085a502195319e04101b38d61dba1 is only a week old... *looks* damn.  I missed that merge by 2 days?

closed 2020-02-18 by VanessaE