MarlinFirmware/Marlin #16894: [BUG] Marlin fails to send OK with empty planner buffers until buffer is empty

CRCinAU, 2020-02-18:
### Bug Description
When printing slowly (ie Luke's flow calibration test cube), there are times when OctoPrint is sending data to the printer, but the printer doesn't respond with OK until the planner buffer becomes empty. Once this happens, data flows correctly again - until the next failure to reply.

It seems that in this mode, maybe marlin should send back BUSY if it won't accept more data, but as the temperature autoreport is sent every 2 seconds, the pause between OK's was approx 60 seconds.

### My Configurations
* SKR Mini E3 v1.2
TX_BUFFER_SIZE 256
RX_BUFFER_SIZE 1024
BUFSIZE 64
BLOCK_BUFFER_SIZE 128

### Octoprint log:
```
Recv: ok N1644 P58 B63
Send: N1645 G1 X108.5 Y127.3 E480.88219*89
Recv: ok N1645 P57 B63
Send: N1646 G0 F830.5 X107.7 Y127.3*93
Recv: ok N1646 P56 B63
Send: N1647 M73 P88 R4*21
Recv: ok N1647 P56 B63
Send: N1648 G0 F300 X107.7 Y127.3 Z14*63
Recv: ok N1648 P55 B63
Send: N1649 G0 F5400 X108.1 Y126.9*113
Recv: ok N1649 P54 B63
Send: N1650 M117 L=68/74 ETC 3m44s*26
Recv: ok N1650 P54 B63
Send: N1651 G1 F2400 E478.88219*49
Recv: ok N1651 P53 B63
Send: N1652 G1 F2400 E480.88219*53
Recv: ok N1652 P52 B63
Send: N1653 G1 F922.8 X126.9 Y126.9 E481.57001*20
Recv: ok N1653 P51 B63
Send: N1654 G1 X126.9 Y108.1 E482.25783*85
Recv: ok N1654 P50 B63
Send: N1655 G1 X108.1 Y108.1 E482.94565*80
Recv: ok N1655 P49 B63
Send: N1656 G1 X108.1 Y126.1 E483.6042*101
Recv: ok N1656 P48 B63
Send: N1657 G0 F830.5 X108.1 Y126.9*95
Recv: ok N1657 P47 B63
Send: N1658 G0 F5400 X107.7 Y127.3*115
Recv: ok N1658 P46 B63
Send: N1659 G1 F922.8 X107.7 Y107.7 E484.32128*18
Recv: ok N1659 P45 B63
Send: N1660 G1 X127.3 Y107.7 E485.03837*83
Recv: ok N1660 P44 B63
Send: N1661 G1 X127.3 Y127.3 E485.75546*94
Recv: ok N1661 P43 B63
Send: N1662 G1 X108.5 Y127.3 E486.44328*89
Recv: ok N1662 P42 B63
Send: N1663 G0 F830.5 X107.7 Y127.3*90
Recv: ok N1663 P41 B63
Send: N1664 M73 P88 R4*20
Recv: ok N1664 P41 B63
Send: N1665 G0 F300 X107.7 Y127.3 Z14.2*44
Recv: ok N1665 P40 B63
Send: N1666 G0 F5400 X108.1 Y126.9*124
Recv: ok N1666 P39 B63
Send: N1667 M117 L=69/74 ETC 3m34s*24
Recv: ok N1667 P39 B63
Send: N1668 G1 F2400 E484.44328*59
Recv: ok N1668 P38 B63
Send: N1669 G1 F2400 E486.44328*56
Recv:  T:249.92 /250.00 B:99.93 /100.00 @:89 B@:73
Recv:  T:249.76 /250.00 B:100.00 /100.00 @:93 B@:69
Recv:  T:250.20 /250.00 B:99.96 /100.00 @:82 B@:72
Recv:  T:250.00 /250.00 B:100.04 /100.00 @:87 B@:67
Recv:  T:250.00 /250.00 B:99.99 /100.00 @:87 B@:70
Recv:  T:250.30 /250.00 B:99.98 /100.00 @:80 B@:71
[... 248 lines cut ...]
Recv:  T:191.18 /0.00 B:93.93 /0.00 @:0 B@:0
Recv:  T:188.92 /0.00 B:93.79 /0.00 @:0 B@:0
Recv:  T:185.79 /0.00 B:93.58 /0.00 @:0 B@:0
Recv:  T:183.55 /0.00 B:93.51 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:181.13 /0.00 B:93.51 /0.00 @:0 B@:0
Recv: ok N1763 P127 B62
Send: N1764 M82*45
Recv: ok N1764 P127 B63
Send: N1765 M104 S0*80
Recv: ok N1765 P127 B63
Send: N1766 M73 P95 R3*28
Recv: ok N1766 P127 B63
Send: N1767 M117 L=74/74 ETC*31
Recv: ok N1767 P127 B63
Send: N1768 M400*31
Recv: ok N1768 P127 B63
Changing monitoring state from "Finishing" to "Operational"
Send: M73 P100 R0
Recv: ok P127 B63
Recv:  T:178.49 /0.00 B:93.51 /0.00 @:0 B@:0
Recv:  T:176.48 /0.00 B:93.51 /0.00 @:0 B@:0
Recv:  T:174.09 /0.00 B:93.36 /0.00 @:0 B@:0
Recv:  T:171.87 /0.00 B:92.64 /0.00 @:0 B@:0
```

thinkyhead, 2020-02-21:
I have been looking into how this could occur. I notice that the host keepalive messages and the ok messages both got suspended, but the temperature auto-report did continue. This is particularly strange, because the temperature auto-report and the host keepalive both emanate from calls to `idle()`. Therefore, whatever was suspending and calling `idle()` could not have been a g-code handler, because g-codes set the busy status.

It looks like `printingHasFinished` is the true culprit. It gets triggered as soon as an SD card job file has finished reading, and if you have a very large buffer this is long before the buffer gets emptied. `printingHasFinished` does a `planner.synchronize` which will call `idle` in a loop without setting the busy status.

So what we really want to do is have a two-stage job completion, where we put off closing the file and completing the print job until the g-code buffer has first been emptied. It means re-writing a bit of the print completion code, but I believe I know what needs to be done.

CRCinAU, 2020-02-21:
Here's an interesting question - as this was via USB, would it matter about SD card stuff? Or is that how it does buffering?

thinkyhead, 2020-02-21:
Ah, right. For a print over USB there will be no such behavior. So, I will have to look deeper!

CRCinAU, 2020-02-29:
For what its worth, I see this with a smaller buffer as well:

```
Send: N143539 G1 X139.489 Y167.214 E2635.07366*100
Recv: ok N143539 P0 B31
Send: N143540 G1 X139.489 Y167.592 E2635.08749*101
Recv: ok N143540 P0 B31
Send: N143541 G1 X139.489 Y167.857 E2635.09718*101
Recv: ok N143541 P0 B31
Send: N143542 G1 X135.276 Y179.671 E2635.55607*111
Recv: ok N143542 P0 B30
Send: N143543 G1 X134.963 Y179.852 E2635.5693*87
Recv:  T:250.20 /250.00 B:100.02 /100.00 @:67 B@:59
Recv: ok N143543 P30 B31
Send: N143544 G1 X134.666 Y180.023 E2635.58184*107
Recv: ok N143544 P29 B31
Send: N143545 G1 X134.503 Y180.118 E2635.58874*101
Recv: ok N143545 P28 B31
Send: N143546 G1 X134.206 Y180.29 E2635.6013*101
Recv: ok N143546 P27 B31
```

You can see the P buffer only start to fill again once its empty after the temp autoreport.

CRCinAU, 2020-02-29:
Ok - I think I might have discovered something.....

I have a TFT35 v2 connected to the onboard TFT connector. I'm pretty sure this just acts as a serial port - and it sends commands to get status etc while the print is going on. If I unplug this screen (therefore we only get the USB serial data), then things don't pause after a T: line on USB.

I believe this may be a result of commands also coming in via the TFT port on from the screen.

Does this assist in the thought process of what's causing it?

thinkyhead, 2020-02-29:
You can try using `M111 S3` to have Marlin echo back everything it gets from the host, including the serial TFT. But, it might only echo back to the TFT that sent it. If this is the case, that echo can be tweaked to go to all serial out.

CRCinAU, 2020-02-29:
I believe the commands (not 100% sure - but believe!) is to retrieve the position, as it updates the X Y Z on the screen, as well as the command to get the current temps (as it doesn't do an auto-report).

M114 (get current position) seems to be called: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/605a921564f6df46638bcd79f20d16c2146740ad/TFT/src/User/Menu/StatusScreen.c#L295

M105 to get temperatures: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/92ee173ebb86442b77d8e5eb738f1d1309790fd0/TFT/src/User/Menu/Heat.c#L276

CRCinAU, 2020-02-29:
It's interesting that since I unplugged the TFT, the blotches that I was seeing on my print (2h left out of 5h) seem to have disappeared - which means they're caused by the small pause when the P buffer empties.

I was racking my brain to figure out what was causing this - hence the first try with a P buffer size of 128... However now we know the cause, what do we do to fix it? :)

thinkyhead, 2020-02-29:
That's it, then! The screen firmware is not compatible with Marlin. The `M114` command necessarily invokes a `planner.synchronize()` because the current position needs to correspond to the stepper positions, and they will be out of sync if the command isn't synchronized.

If the LCD firmware can be altered, then we could do something for it. The other alternative is to alter `M114` so that it "fakes" the stepper positions by calculating them (kinematics included) from the cartesian current position and displaying the calculated rather than true current stepper values.

CRCinAU, 2020-02-29:
Hmmm... Many thoughts.... I wonder if its correct for M114 to just report the position as per where it is right now - without syncing the planner - ie where it would be *after* everything in the P buffer has been executed?

If the P buffer is empty, it wouldn't matter - but if the P buffer contains things, then what M114 reports could only be correct AFTER the buffer has been emptied. Should it return sooner, but with the position as per when the command is executed - and not after the P buffer is empty?

Hmmmm

Failing that, I assume currently there is no way to subscribe to position updates on a timer - like there is for temperature reporting...

thinkyhead, 2020-02-29:
- The high level current position is based on the last G-code queued to the planner.
- The planner current position (in steps units) is based on the last queued planner block.
  - Workspaces, kinematics, skew correction are applied before and during this stage.
- The stepper current positions are the true physical stepper positions (in steps).

These are never in sync when the machine is in motion.

thinkyhead, 2020-02-29:
The simplest thing to do is to add a custom command (or extra flag for `M114`) that tells it to skip `synchronize` and only report the simple XYZ (optional E) position, for something like an LCD.

CRCinAU, 2020-02-29:
I think that might be the better option - as it doesn't make sense to *always* skip the sync.

As the S parameter is usually used for values, would it be better to use `M114 D1` = Enable dirty report?

This way the position can be reported for random updates, but it doesn't have to be accurate, and a dirty report of where it kinda is will probably be good enough.

I can then build a firmware that replaces all `M114` commands with `M114 D1` in the firmware to see if this fixes the problem in this - and in the TFT issues page I just linked.

thinkyhead, 2020-02-29:
- The `D` parameter is currently in use in association with the `M114_DETAIL` feature.
- The `E` parameter reports `E` separately (or alone) for the benefit of BTT, since they submitted that change.

CRCinAU, 2020-02-29:
Yeah - I believe the E parameter is for their 'smart filament sensor' thing... Does that also call `planner.synchronize()`? If so, that would cause the same issue...

thinkyhead, 2020-02-29:
Here's the context for `synchronize`: 34ff0c3

guruathwal, 2020-02-29:
The TFT sends the `M114` command when it is on `Status Menu` or when it is on `Move Menu` to display the correct nozzle position, setting the TFT to any other menu will stop the `M114` command.

The `synchronize` part can be avoided if Marlin is able to respond to `M114` with the approximate position details from G-code it is processing rather than recalculating during printing.
another way can be to send a confirmation message from marlin to all connected serial devices that a print has started which can be helpful to stop the sending commands from idle hosts during prints and display a `Print in progress` with printing progress/status.

About a month ago I open a PR https://github.com/MarlinFirmware/Marlin/pull/16682 to auto report the Z position while printing from Onboard sd-card to avoid sending `M114` from TFT for the update but it was closed with the reason being that it was getting mixed up and  M114 command is good enough. After that, I dropped the idea of displaying the Z height on the TFT when printing through Onboard sd-card as I did not want to poll M114 every few seconds to update the position.
Although I did not dig deep enough to check if https://github.com/MarlinFirmware/Marlin/pull/16682 would have triggered the `synchronize` part.

CRCinAU, 2020-02-29:
Yeah - it causes big quality issues when the P buffer becomes empty - hence it certainly needs a non-blocking mode. I chose `U` to indicate that it is reported in the unsync'ed state.

This way, the TFT can change to call `M114 U` and get a 'good enough' position without having to wait for sync - therefore causing the buffer to empty and a blob to form.

guruathwal, 2020-02-29:
@CRCinAU the `U` option will be helpful but the host will also need to know when to send `M114` and when to send 'M114 U'. There should also be some status message from Marlin to inform the hosts that a print is in progress through another host.
Alternatively, it can be possible to report the unsynced position on `M114' when the `Print Timer` is running and report the synced position when idle.

CRCinAU, 2020-03-01:
@thinkyhead Theres an idea.... If the print timer is running, should it skip the `planner.synchronize()` instead of adding a new parameter to do this?

hfog, 2020-03-01:
The problem is that if the print is running from another source such as Octoprint, the TFT may have no idea that a print is running.

CRCinAU, 2020-03-01:
Yeah - I've just tested that proposed patch - but it seems that there is still a call to `planner.synchronize()` somewhere... It makes me wonder if the `print_job_timer` is scoped to each serial interface instead of global...

It builds ok, but I still see the buffer empty when the TFT sends an M114.

I can see @thinkyhead is working on stuff at the moment - which look like much more drastic changes that for the moment I won't pretend to understand - but I'm guessing will be much more complete than my feeble attempt :grin:

guruathwal, 2020-03-01:
just saw #17032 by @thinkyhead addressing the M114 synchronization issue.

thinkyhead, 2020-03-01:
> Theres an idea.... If the print timer is running, should it skip the `planner.synchronize()` instead of adding a new parameter to do this?

Well, what you more or less end up with there is that it will only `synchronize` when it doesn't much matter. So I just took the approach of removing the `synchronize` entirely, but made `M114` still return the same values that it would if you waited. The position returned during movement will be ahead of the actual position.

- If debugging you would tend to use `M114 D` and for that it will still `synchronize`.
- To get really deep, `M114 R` can be added to return the position according to the steppers right now.

guruathwal, 2020-03-02:
@thinkyhead 
> * The `E` parameter reports `E` separately (or alone) for the benefit of BTT, since they submitted that change.

The `E` parameter should also be changed to report the length of filament in `MM` instead of the total step count because the TFT firmware converts the steps in `MM` based on the extruder's `steps per MM` specified in the configuration of the TFT firmware. This will give an inaccurate reading when someone uses precompiled TFT firmware with an extruder with different `steps per MM`

CRCinAU, 2020-03-02:
Ah - I figured out what was wrong with my build.... the scripts I use run a `git reset --hard` - so it was removing the patch after compiling the first binary :)

However, instead of testing my patch again, I've rolled a number of firmwares with @thinkyhead changes in the PR and will give it a test when I get to sit down and do a print again.

I haven't done any special changes to any Configuration.h or Configuration_adv.h files - so if I understand the code correctly, an M114 will return without the sync now by default.

CRCinAU, 2020-03-02:
@thinkyhead - I've done some test prints using your PR code on top of bugfix-2.0.x.

I can confirm that the P buffer does not decrease now when the TFT issues an M114 command.

+1 from me.

denis-itskovich, 2020-03-04:
@CRCinAU Were you eventually be able to solve this somehow?

CRCinAU, 2020-03-04:
> @CRCinAU Were you eventually be able to solve this somehow?

Yep - grab the latest bugfix-2.0.x. The default M114 behaviour has changed to resolve that problem.

closed 2020-03-04 by CRCinAU