MarlinFirmware/Marlin #16896: TMC Drivers Not Responding After Changing To Dual Independant X Carridge

MrLAtt, 2020-02-18:
Hi, I have just upgraded my printer to a dual independent X-carridge configuration, from a standard prusa i3 configuration.

My printer has TMC2209 drivers all round and I have been using sensorless homing for the past 2-3months without problem, however, since I have made the firmware changes to use dual independent x carridges sensorless homing is not working (on any axis) and when I try to check the TMC status, marlin returns "Unknown command: "M122"".

It would appear that marlin is now unaware, for whatever reason, that it has trinamic drivers. All of these things were working before I enabled Dual X carridges, and that was the only change to the configuration that I made.

[Config Files TMC Not working.zip](https://github.com/MarlinFirmware/Marlin/files/4220746/Config.Files.TMC.Not.working.zip)

I am also using a modified version of marlin 2.0 bugfix, that allows the onboard SDCard reader to work, I doubt this is the problem (I was using it fine before) But i can repost this there if thats better.

Please ask for any more info if it is needed

LinoBarreca, 2020-02-19:
The "unknown command" sounds like you are sending a wrong character to the terminal.
it might happen if you press "backspace" (for example) before entering the command.
To reproduce this just send backspace and return. you'll get an "unknown command" error.
The M122 command should be recognized, no matter what branch you are using.
Please try again by pressing just a return before issuing the M122

MrLAtt, 2020-02-19:
I just tried that and it's not it, I can't say for sure that I am doing the correct thing, but I'm typing it in exactly the same way I am typing 'M503', and that works fine.

Find attached a picture of my pronterface screen, I hope it can be of help
[image]

italocjs, 2020-02-20:
I'm using two TMC's on the Y axis and they work fine (i also have two mechanical endstops for skew correction). but i never tried on X.   something might be happening there.

Another thing, try using a older version (at least one release before) of the tmc library just to be sure its nothing related to it

MrLAtt, 2020-02-22:
I have just confirmed that my problem is nothing to do with the experimental marlin branch that I was using by switching over to the latest marlin bugfix branch, an I am still have the exact same problem. The printer responds to all commands, except those relating to the Trinamic drivers. 

Sensorless homing also still doesn't work, everything else, as far as I am aware, is working fine

I'm not sure how to use an older version of the trinamic library, I'm using Atom, and I don't see any obvious option for it.

sjasonsmith, 2020-02-23:
@MrLAtt, the configuration you posted does not define any drivers as TMC drivers. You need to uncomment the lines, not only change the driver name.

This is what you have right now, entirely commented out.
```
//#define X_DRIVER_TYPE  TMC2209
//#define Y_DRIVER_TYPE  TMC2209
//#define Z_DRIVER_TYPE  TMC2209
//#define X2_DRIVER_TYPE TMC2209
//#define E0_DRIVER_TYPE TMC2209
//#define E1_DRIVER_TYPE TMC2209
//#define E2_DRIVER_TYPE TMC2209
```

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle