MarlinFirmware/Marlin #16899: [BUG] Y_DUAL_STEPPER_DRIVERS / X_DUAL_STEPPER_DRIVERS steppers don't move in alignment

dpedu, 2020-02-19:
### Bug Description

I've enabled dual Y stepper drivers like:

```
#define Y_DUAL_STEPPER_DRIVERS
```

But when I send move commands (e.g. jog control from desktop control software) the steppers move different amounts. I am testing by connecting one driver at a time to a test motor and observing it. I guess it moves about 1.2 times further when the "extra" Y driver is tested. It seems faster too.

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4223751/config.zip)

### Steps to Reproduce

1. Compile with Y_DUAL_STEPPER_DRIVERS or X_DUAL_STEPPER_DRIVERS
2. Send move commands

**Expected behavior:** The steppers move the same distance 

**Actual behavior:** The steppers don't move the same distance 

#### Additional Information

* SKR 1.3 board
* TMC2208 steppers
* Did the `M502` then `M500` trick but I don't think that's relevant here?

TheNeskik, 2020-02-19:
In your configuration.h file, try to activate Y2 driver (line 679)

It should look like :
```
#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208
//#define X2_DRIVER_TYPE A4988
#define Y2_DRIVER_TYPE TMC2208
```

dpedu, 2020-02-19:
Much appreciated! That made it behave correctly. The step is obvious once you know it.

closed 2020-02-19 by dpedu