MarlinFirmware/Marlin #16901: [BUG] Creality Cr-10 and cr-10S bed wrong configs

Liger0, 2020-02-19:
### Bug Description
The creality cr-10 and cr-10s bed related settings are wrong

For the heating feature, these are the stock codes in the firmware:

 ```
 //120V 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
  //from FOPDT model - kp=.39 Tp=405 Tdead=66, Tc set to 79.2, aggressive factor of .15 (vs .1, 1, 10)
  #define DEFAULT_bedKp 10.00
  #define DEFAULT_bedKi .023
  #define DEFAULT_bedKd 305.4

  //120V 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
  //from pidautotune
  //#define DEFAULT_bedKp 97.1
  //#define DEFAULT_bedKi 1.41
  //#define DEFAULT_bedKd 1675.16

```

those are unrelated from the cr-10(S) and don't work.
If I use the one defined there without manual PID, I can't print over 60°C, which even takes an eternity to reach.
If I define the manual PID autotune, the PID autotuning feature doesn't work because it still uses the defined PID values, so there is no way around that and still fails the PID autotuning at 60°C.

I worked manually on them until being able to do a proper PID autotune at 80°C (mid value between 60°C for PLA and 100°C for ABS), and this is the PID value that results:

```
  //Stock cr10s bed and glass
  #define DEFAULT_bedKp 185.12
  #define DEFAULT_bedKi 36.99
  #define DEFAULT_bedKd 617.69
```

with this I can easily print even on ABS temp at 110°C.

Talking about mechanical limits, the bed area settings don't make much sense.

This setting is non-sense:

```
#define X_BED_SIZE 296 // Nozzle is at X4 when homed
#define Y_BED_SIZE 298 // Nozzle is at Y2 when homed
```
it should be, instead

```
#define X_BED_SIZE 300 // Nozzle is at X0 when homed
#define Y_BED_SIZE 300 // Nozzle is at Y0 when homed
```

again, this doesn't make sense
```
#define X_MAX_POS 319 // 15mm Extra space on the right end
#define Y_MAX_POS 306 // 4mm Extra space behind the bed
```

what is correct is, since the bed size is 5mm ahead:
```
#define X_MAX_POS 305 // 15mm Extra space on the right end
#define Y_MAX_POS 305 // 4mm Extra space behind the bed
```
same for this, which is wrong:
```
#define MANUAL_X_HOME_POS 4
#define MANUAL_Y_HOME_POS 2
```
while the correct values are

```
//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
```
because the bed is larger of 5mm on the negative side, so printing at the homing position is not on the edge and is totally fair.

Marlor, 2020-02-22:
PID for the Bed is disabled by default on cr-10(S):
`//#define PIDTEMPBED`
before changing pid settings you should argue if it should be active by default

I tried to explain (but did a bad job) why MANUAL_X/Y_HOME_POS is flat out wrong:
https://github.com/MarlinFirmware/Marlin/issues/16481

maybe you should open a new issue in https://github.com/MarlinFirmware/Configurations

Liger0, 2020-02-22:
Even if PID for bed is disabled by default, with it disabled the bed won't go above 60°C anyway.

houseofbugs, 2020-02-25:
> Even if PID for bed is disabled by default, with it disabled the bed won't go above 60°C anyway.

Thats a machine issue, not a firmware issue. This "issue" request should be closed.

Liger0, 2020-02-25:
This is a firmware issue as said 3 times already, not a machine issue. With the correct PID value, it will work.

thinkyhead, 2020-02-26:
> With the correct PID value, it will work.

`M303` is not optional.

closed 2020-02-26 by thinkyhead

Marlor, 2020-02-26:
@thinkyhead the bed dimensions are wrong for everyone.
HOME_POS shoud never be positive, this leads to the Carriage grinding into endstops.
Please look at https://github.com/MarlinFirmware/Marlin/issues/16481

thinkyhead, 2020-02-26:
I fixed that already.

Marlor, 2020-02-26:
So if i set `#define MANUAL_X_HOME_POS 10`

0 is now 10mm inside the bed and not 10 inside the endstop?

thinkyhead, 2020-02-26:
> So if i set #define MANUAL_X_HOME_POS 10

If your nozzle was at 10 when homed that would be good. But also set `X_MIN_POS` to 10!

Liger0, 2020-02-26:
The center is not on the center but on the right side, they are totally wrong by all meanings. The beds are all the same, and homing at x=10 is totally wrong.

Liger0, 2020-02-26:
> 
> 
> > With the correct PID value, it will work.
> 
> `M303` is not optional. The bed dimensions are pretty good for me. The nozzle is quite close to exact center when moved to 150, 150. Of course these are not meant to be perfect for everyone. Change them as you see fit for your personal setup.

I think I already said it multiple times, but M303 FAILS with stock configs when PID autotuning because it can't go over 60°C with stock PID configs and the heating process ends in a HALTS.